pbatard/libwdi #126: "This app can't run on your PC" on Windows 10 x64

fantoro, 2018-05-17:
[image]
I tried every version and it didn't work and I tried compatibility options, I don't know why it won't work.

pbatard, 2018-05-17:
Are you sure you aren't using an ARM64 based version of Windows?

This is exactly the message one would expect to see when trying to run an x86_32 or x86_64 executable on ARM64.

Please open a Windows command prompt and type the following:
`set processor`
Then copy/paste the output.

fantoro, 2018-05-17:
[image]

pbatard, 2018-05-17:
Okay then I need the exact URL of the executable you downloaded that resulted in your message.

fantoro, 2018-05-17:
http://zadig.akeo.ie/downloads/zadig-2.3.exe

pbatard, 2018-05-17:
Okay, in that case, the only logical explanation I see is you are using an x64 version of Windows where 32-bit compatibility is disabled.

You will need to install 32-bit compatibility (not sure how, since all I know is you have to get out of your way to disable 32-bit on Windows x64), as I have no plans to release 64-bit versions of Zadig.

Otherwise, you will have to compile Zadig yourself ([this Wiki page](https://github.com/pbatard/libwdi/wiki/Compiling-and-debugging-libwdi-or-Zadig) may help).

I strongly suspect that you will run into this issue over and over as you try to run 32-bit software on your system...

fantoro, 2018-05-17:
Wait, I think that it is enabled, otherwise, I wouldn't be able to launch some applications that I use like Steam (Steam is 32-bit afaik).

pbatard, 2018-05-17:
Well, I don't know how your system is set up, but what I can say is:
* You're the only person reporting this (which means it's not a problem from the exe, but from your environment)
* I only know of this message appearing when Windows detects that the executable you are trying to run is not compatible with the CPU architecture of your system

Logically then, there's not much I can do unless we identify what is so specific on your system, that prevents a standard 32-bit executable from running. And I'm afraid you're the only person who can figure that out...

mcuee, 2018-05-20:
To OP: are you running Windows 10 S mode which only allow apps from the Windows store?

mcuee, 2018-05-20:
Reference URL: https://www.drivereasy.com/knowledge/this-app-cant-run-on-your-pc-solved/
One possibility is that "Sideload apps" is disabled.

krillme, 2018-05-24:
I just made an account to say I am also getting this problem. Latest version of Windows 10 Pro, sideload apps on. I've tried with developer mode as well, I've tried disabling Windows SmartScreen and my anti-virus, I've tried many different versions of the program along with different compatibility modes, I've tried creating a new administrator account and running it, nothing. Same error every time, "This app can't run on your PC".

pbatard, 2018-05-24:
Can you please contact Microsoft about it?

Unfortunately, the message does not provide any kind of help as to why the application may not be compatible with your versions of Windows, and is it therefore up to Microsoft support to help with this issue, as without having Microsoft providing the exact reason why the message pops up, I literally cannot do anything about this.

pbatard, 2018-05-24:
Oh, and for the record, I am using the April 2018 Update on Windows 10 x64 too (and I am not seeing this issue, and neither seem the many other people who updated to the latest), so this is __definitely not__ a global issue with the application. There is something specific to your environment, that Microsoft needs to help clarify, that prevents the application from running.

krillme, 2018-05-25:
I've tried resetting my services to default values and safe mode, still nothing. I suppose contacting Microsoft support is the next step, but I don't think they'll be any help. They'll most likely just brush us off and say "oh it's a problem with the software not us bye".

Are there any dependencies required for Zadig we might be missing? Here is my Windows Features list, am I missing anything? 
[image]

Walnut356, 2018-05-25:
It doesn't seem to be a hardware issue, I have literally 0 parts in common with you

https://pcpartpicker.com/user/Anthony356/saved/QRwCmG

krillme, 2018-05-25:
Don't know if this is of any value, but I attempted to run the program through PowerShell and CMD elevated and got these errors. 
[image]

krillme, 2018-05-25:
Hey, sorry for the double post, but I think this is important enough to keep separate instead of editing it into my double post. I have absolutely no idea why this is the case, but downloading the program through Microsoft Edge and running it instead of saving it worked. Google Chrome, for some reason, seems to mess up the file during download. This is the strangest thing I've seen in a long time.

Walnut356, 2018-05-25:
this is the solution, works for me as well. Super weird.

pbatard, 2018-05-25:
Interesting...

I can't seem to replicate the issue with Chrome. If you still have the problematic file, can you please check its size? The only thing I can think of is that maybe Chrome got a partial download for one reason or another. `zadig-2.3.exe` should be 5,157,496 bytes. You can see the size in bytes if you right click on the file and select _Properties_.

krillme, 2018-05-25:
Extremely weird. Downloading it with Chrome gives me an 8,595,826 byte file. Downloading with Edge gives me the proper size. I've included the incorrectly sized file if you want to take a look at it.
[zadig-2.3.zip](https://github.com/pbatard/libwdi/files/2038977/zadig-2.3.zip)

pbatard, 2018-05-26:
Interesting. The file you attached seems to "loop" at `0x1A3B7D` and then again at `0x3476FA` (which is `0x1A3B7D` x `2`), and then, from byte `0x3476FA`, you have the proper full size `zadig-2.3.exe`.

So, it looks like, for whatever reason, Chrome replicated the first 1,719,165 bytes of the Zadig executable twice.

If you are consistently seeing this issue with Chrome, I would contact the Chrome developers, as I am not seeing the issue, and as far as I can tell, the browser does ask for the complete file size before it starts the download (so that it can display the download progress), and I certainly did not configure the web server to report different download sizes according to this and that condition.

In other words, if Chrome downloaded more data than the actual size of the file, and created an invalid executable as a result, it seems likely that this is an issue with Chrome, or possibly with a Chrome extension that you use. I would suggest that you try to disable the extension you use, especially the ones that have to do with downloading, as it is very possible that they could screw up the download size reported by the server. If not, then, as suggested, you may try to hear from the Chrome developers what they think could be the issue.

krillme, 2018-05-29:
Doing so stops the "This app can't run" dialog from appearing, but gives me this error instead.

[image]

eladkarako, 2018-05-31:
@krillme zadig might be <a href="https://en.wikipedia.org/wiki/UPX">UPX</a> compressed,
in which case resource editing is not a good idea until decompressed with <code>upx.exe -d zadig....</code>.
nevermind, try <code>zadig.exe</code> from here: https://github.com/eladkarako/mods/tree/master/HeimdallSuite/Drivers
already modified. run as admin!

krillme, 2018-05-31:
Same "side-by-side configuration" error, at least with Chrome. I haven't bothered testing with Edge as it works fine normally with Edge so I don't see the point in testing the other versions.

pbatard, 2018-06-11:
I'm going to close this issue now. Clearly, this is not a Zadig issue but a problem with how third party applications are interfering with the official binary download.

Also, for the record, the Zadig binary __already__ comes with a manifest that indicates Windows 10 support (see [here](https://github.com/pbatard/libwdi/blob/master/examples/common_controls_and_elevation.manifest#L30-L31)), since we need that to be able to detect the Windows 10 version.

closed 2018-06-11 by pbatard

atomicthumbs, 2018-06-20:
I am experiencing this problem whether Zadig is downloaded bare or unzipped from krillme's zip by 7-zip. Running Windows 10 64-bit, build 17134.

pbatard, 2018-06-20:
> I am experiencing this problem whether Zadig is downloaded bare

Have you tried a different browser than Chrome. If you are using Chrome, you should be aware that Chrome will __MESS UP THE DOWNLOAD__

> or unzipped from krillme's zip by 7-zip. 

krillme's zip __IS__ the messed up download. Of course that will not work!!!

Please confirm that the SHA-256 of the `zadig-2.3.exe` you downloaded is `E2695E5732A8B512DB00B5B7D5DB4854195E82D343633C245E50F26363CAFB2E`. If it's not that, then the download was corrupted by whatever browser you're been using and you'll need to find a browser that doesn't corrupt downloads.

atomicthumbs, 2018-06-21:
1. I'm a poor reader, apparently.
2. I downloaded it with Opera, which is Chromium-based. Firefox download works fine. Very odd. Sorry to bother you!