pbatard/libwdi #127: Should I replace the default usbprint.sys ?

abulka, 2018-05-22:
I am trying to install libusb on windows 10 so that I will have a 'backend' when I talk to my printer using pyusb.  When I run zadig my Epson Thermal Printer TM-TS20 is not listed, unless I list all devices, in which case it is listed with a driver already installed!  This makes sense because:

> When you plug in a usb printer into a computer running Windows ... the “plug and pray” subsystem loads and attaches the usbprint.sys bus driver to it, in essence taking over the device. https://blog.peter.skarpetis.com/archives/2005/04/07/getting-a-handle-on-usbprintsys/comment-page-2/

Zadig recommends not replacing such drivers.  So my question is how is libusb supposed to be installed when it necessarily involves replacing the default windows Usbprint.sys driver, which ironically, zadig recommends not to do?

diabolo38, 2018-05-22:
Note that you may install and use usbdk as libusb back end. You need latest lib and set the aproriate option before to  init lib and claim interface . this want require to uninstall initial drivers or install Amy new one. Somehow a new filter driver will be setup as needed. I have tested only once usdk when it was a static option on the DLL built. It did work nicely with some minor side effect.

pbatard, 2018-05-22:
Please understand that the Zadig notices are aimed at people who don't know what they are doing and clicking at random (which, judging by the number of _"Help, I replaced a system driver that I didn't want to replace, and now I need to revert it!"_ requests I receive is way too many). So for these people, Zadig will indicate that, __usually__ you don't want to replace a system driver.

Now, if you are absolutely sure that you have selected the right device, and you can live with that device being accessible only with libusb __and nothing else__, then you can ignore the notice (which is what it is: a notice, not an order).

Or, if you want to use libusb and continue to have Windows being able to use your printer, then you can look into using UsbDk.

As I believe I have answered your question, I will close your issue.

closed 2018-05-22 by pbatard

abulka, 2018-05-23:
Thanks for the explanation - sounds like UsbDk is indeed what I want, because I don't want to replace existing windows drivers etc. So I installed it from https://github.com/daynix/UsbDk/releases and running `'C:\Program Files\UsbDk Runtime Library\UsbDkController.exe' -n` successfully lists my device.

However it seems (from my reading) that I still need libusb-1.0 Windows library (libusb-1.0.dll) to programmatically use UsbDk ? - can Zadig help with that?  Running zadig without selecting a device seems to copy various drivers into ~/usb_driver directories, but I cannot see `libusb-1.0.dll` in that list.

diabolo38, 2018-05-23:
Yes you still need libusb DLL >= 1.0.22 . use set option api to enable usbdk usage (default is off)

pbatard, 2018-05-23:
> can Zadig help with that?

No. Zadig only installs drivers. It does not install libusb.

abulka, 2018-05-24:
Thanks @diabolo38 - presumably this would be something like:

    libusb_context * ctx = NULL;
    libusb_init(&ctx);
    libusb_set_option(ctx, LIBUSB_OPTION_USE_USBDK);

Do I also have to recompile libusb myself with the `--enable-usbdk` flag as [this post](https://sourceforge.net/p/libusb/mailman/libusb-devel/thread/CAL-1MmVd%2BHBWeuseP6Gd9aF6ORcZewZ7%3D8OYe6DpaA6%2BQaQgKg%40mail.gmail.com/) suggests?

diabolo38, 2018-05-24:
Yes it must be something like that
No recompile needed use 1.0.22 or newest. The post refer to usbdk used on earlier version