pbatard/libwdi #129: Can't create self signed cert, invalid algorithm (Windows 10 build 17134)

Forst, 2018-05-26:
Hello!

I am unable to install the WinUSB (or any other, in fact) driver for my RTL-SDR stick. My wild guess is that the latest Windows 10 update removed some obsolete ciphers/hashes/whatever, as it's complaining about an invalid algorithm:

```
libwdi:warning [CreateSelfSignedCert] could not create self signed certificate: [#80090008] Указан неправильный алгоритм.
libwdi:warning [wdi_prepare_driver] could not sign cat file
```

*(translates as "Invalid algorithm specified")*

Full log attached below.

-----

```
Zadig 2.3.701
Windows 10 64-bit (Build 17134)
ini file 'zadig.ini' not found - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
5 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_203A&PID_FFF9&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.17134.1
libwdi:debug [wdi_create_list] usbvideo USB device (0): USB\VID_203A&PID_FFF9&MI_00\6&3B2622F3&0&0000
libwdi:debug [wdi_create_list] Device description: 'FaceTime HD Camera (Built-in) (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Upper filter: RTL2832UBDA
libwdi:debug [wdi_create_list] Driver version: 64.1.802.2010
libwdi:debug [wdi_create_list] RTL2832UUSB USB device (3): USB\VID_0BDA&PID_2838&MI_00\6&38A180A5&0&0000
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_203A&PID_FFFC&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.17134.1
libwdi:debug [wdi_create_list] HidUsb USB device (5): USB\VID_203A&PID_FFFC&MI_00\6&100686D9&0&0000
libwdi:debug [wdi_create_list] Device description: 'Absolute Coordinate Interface (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 8664.1.818.2009
libwdi:debug [wdi_create_list] RTL2832U_IRHID USB device (7): USB\VID_0BDA&PID_2838&MI_01\6&38A180A5&0&0001
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_203A&PID_FFFC&REV_0100&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.17134.1
libwdi:debug [wdi_create_list] HidUsb USB device (9): USB\VID_203A&PID_FFFC&MI_01\6&100686D9&0&0001
libwdi:debug [wdi_create_list] Device description: 'Relative Coordinate Interface (Interface 1)'
Using inf name: Bulk-In_Interface_(Interface_0).inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\Foster Snowhill\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\Foster Snowhill\usb_driver\Bulk-In_Interface_(Interface_0).inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Foster Snowhill\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Foster Snowhill\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'bulk-in_interface_(interface_0).inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Foster Snowhill\usb_driver\bulk-in_interface_(interface_0).inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Foster Snowhill\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Foster Snowhill\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] successfully created file 'C:\Users\Foster Snowhill\usb_driver\Bulk-In_Interface_(Interface_0).cat'
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] acquired existing key container
libwdi:debug [CreateSelfSignedCert] generated new keypair
libwdi:warning [CreateSelfSignedCert] could not create self signed certificate: [#80090008] Указан неправильный алгоритм.
libwdi:warning [wdi_prepare_driver] could not sign cat file
[... 97 lines cut ...]
libwdi:debug [syslog]      sig:                          Key      = bulk-in_interface_(interface_0).inf
libwdi:debug [syslog]      sig:                          FilePath = C:\WINDOWS\System32\DriverStore\Temp\{3907f1fd-a0ec-324c-a171-72defa276b69}\bulk-in_interface_(interface_0).inf
libwdi:debug [syslog]      sig:                          Catalog  = C:\WINDOWS\System32\DriverStore\Temp\{3907f1fd-a0ec-324c-a171-72defa276b69}\Bulk-In_Interface_(Interface_0).cat
libwdi:debug [syslog] !    sig:                          Verifying file against specific (valid) catalog failed.
libwdi:debug [syslog] !    sig:                          Error 0x800b0100: No signature was present in the subject.
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE exit(0x800b0100)} 22:00:46.808
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE} 22:00:46.809
libwdi:debug [syslog]      sig:                          Key      = bulk-in_interface_(interface_0).inf
libwdi:debug [syslog]      sig:                          FilePath = C:\WINDOWS\System32\DriverStore\Temp\{3907f1fd-a0ec-324c-a171-72defa276b69}\bulk-in_interface_(interface_0).inf
libwdi:debug [syslog]      sig:                          Catalog  = C:\WINDOWS\System32\DriverStore\Temp\{3907f1fd-a0ec-324c-a171-72defa276b69}\Bulk-In_Interface_(Interface_0).cat
libwdi:debug [syslog] !    sig:                          Verifying file against specific Authenticode(tm) catalog failed.
libwdi:debug [syslog] !    sig:                          Error 0x800b0100: No signature was present in the subject.
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE exit(0x800b0100)} 22:00:46.811
libwdi:debug [syslog] !!!  sig:                     Driver package catalog file does not contain a signature, and Code Integrity is enforced.
libwdi:debug [syslog] !!!  sig:                     Driver package failed signature validation. Error = 0xE0000247
libwdi:debug [syslog]      sto:                {DRIVERSTORE IMPORT VALIDATE: exit(0xe0000247)} 22:00:46.812
libwdi:debug [syslog] !!!  sig:                Driver package failed signature verification. Error = 0xE0000247
libwdi:debug [syslog] !!!  sto:                Failed to import driver package into Driver Store. Error = 0xE0000247
libwdi:debug [syslog]      sto:           {Stage Driver Package: exit(0xe0000247)} 22:00:46.815
libwdi:debug [syslog]      sto:      {Setup Import Driver Package - exit (0xe0000247)} 22:00:46.821
libwdi:debug [process_message] switching timeout back to finite
libwdi:debug [installer process] operation cancelled by the user
libwdi:debug [process_message] installer process completed
Driver Installation: Cancelled by User
```

pbatard, 2018-05-26:
Thanks for the report.

Despite many people (including myself) using the latest fully patched Windows 10, you are the only person reporting this, so I logically have to suspect that this has to do more with something that is specific to your environment, as opposed to something that Microsoft may have broken for everybody.

For one thing, I certainly cannot replicate this issue.

From the error you are getting (`Invalid algorithm specified`), and from the code in `CreateSelfSignedCert()`, the only possibility I see is that, somehow, your Windows configuration disabled the use of SHA-256 RSA implementation (`szOID_RSA_SHA256RSA`), as this is the only way I see where you would get _Unknown/Invalid cryptographic algorithm_ as an error.

I know that, for instance, Local Group Policy settings can disable the use of some cryptographic algorithms, though that should be for obsolete or potentially vulnerable ones, which SHA-256 isn't.

What happens if you try on a different Windows 10 platform?

Forst, 2018-05-27:
I succeeded in reproducing this issue on a new installation of Windows 10. The problem is caused by [CryptoPro CSP](https://www.cryptopro.ru/en/node/584) – a crypto provider implementing support for Russian cryptographic algorithms. Trying it with CSP version 5.0.11057.

If my understanding is correct, Windows can have multiple crypto providers installed and can choose whichever one it needs. Zadig perhaps picks the first available one, which happens to be CryptoPro?

Forst, 2018-05-27:
Thank you very much for a thorough explanation!

It is indeed trying to reimplement RSA and ECDSA if you tell it to, however it's not the default option, that might explain why nobody encountered this issue before.

For the record, in case anyone else gets hit by this. What one has to do is to uncheck "Support for RSA/ECDSA cryptoproviders" in Add/remove programs for CryptoPro CSP, which removes relevant broken components and thus fixed the problem.