pbatard/libwdi #130: TDM64-GCC GCC's setupapi.h is old

mcuee, 2018-06-24:
This is not really a libwdi issue but the Wiki recommends using TDM64 as it seems to be the only popular multilib capable MinGW-w64 distribution. 

TDM64 has not been updated for a while and its setupapi.h file is missing SetupDiGetDevicePropertyW function. Workaround is to get a newer version from other distro like MSYS2.

```
$ make
make  all-recursive
make[1]: Entering directory `/c/work/libwdi/libwdi'
Making all in libwdi
make[2]: Entering directory `/c/work/libwdi/libwdi/libwdi'
  CCLD   embedder
  CC       installer_x86-installer.o
  CCLD     installer_x86.exe
  CC       installer_x64-installer.o
  CCLD     installer_x64.exe
  EMBED  C:\work\libwdi\wdk10-redist\wdk\redist\wdf\x86\WdfCoInstaller01011.dll (2018.06.21)
  EMBED  C:\work\libwdi\wdk10-redist\wdk\redist\wdf\x86\winusbcoinstaller2.dll (2018.06.21)
  EMBED  C:\work\libwdi\wdk10-redist\wdk\redist\wdf\x64\WdfCoInstaller01011.dll (2018.06.21)
  EMBED  C:\work\libwdi\wdk10-redist\wdk\redist\wdf\x64\winusbcoinstaller2.dll (2018.06.21)
  EMBED  C:\work\libwdi\libusb-win32-bin-1.2.6.0\bin\x86\libusb0_x86.dll (2018.06.21)
  EMBED  C:\work\libwdi\libusb-win32-bin-1.2.6.0\bin\x86\install-filter.exe (2018.06.21)
  EMBED  C:\work\libwdi\libusb-win32-bin-1.2.6.0\bin\x86\libusb0.sys (2018.06.21)
  EMBED  C:\work\libwdi\libusb-win32-bin-1.2.6.0\bin\amd64\libusb0.dll (2018.06.21)
  EMBED  C:\work\libwdi\libusb-win32-bin-1.2.6.0\bin\amd64\libusb0.sys (2018.06.21)
  EMBED  C:\work\libwdi\libusb-win32-bin-1.2.6.0\bin\amd64\install-filter.exe (2018.06.21)
  EMBED  C:\work\libwdi\libusb-win32-bin-1.2.6.0\installer_license.txt (2018.06.21)
  EMBED  C:\work\libwdi\libusbK-3.0.7.0-bin\bin\sys\x86\libusbK.sys (2018.06.21)
  EMBED  C:\work\libwdi\libusbK-3.0.7.0-bin\bin\dll\x86\libusbK.dll (2018.06.21)
  EMBED  C:\work\libwdi\libusbK-3.0.7.0-bin\bin\sys\amd64\libusbK.sys (2018.06.21)
  EMBED  C:\work\libwdi\libusbK-3.0.7.0-bin\bin\dll\amd64\libusbK.dll (2018.06.21)
  EMBED  c:\work\libwdi\libwdi\libwdi\installer_x86.exe (2018.06.24)
  EMBED  c:\work\libwdi\libwdi\libwdi\installer_x64.exe (2018.06.24)
  EMBED  c:\work\libwdi\libwdi\libwdi\winusb.inf.in (2018.04.07)
  EMBED  c:\work\libwdi\libwdi\libwdi\libusb0.inf.in (2018.04.07)
  EMBED  c:\work\libwdi\libwdi\libwdi\libusbk.inf.in (2018.04.07)
  EMBED  c:\work\libwdi\libwdi\libwdi\usbser.inf.in (2018.04.07)
  EMBED  c:\work\libwdi\libwdi\libwdi\winusb.cat.in (2018.04.07)
  EMBED  c:\work\libwdi\libwdi\libwdi\libusb0.cat.in (2018.04.07)
  EMBED  c:\work\libwdi\libwdi\libwdi\libusbk.cat.in (2018.04.07)
  EMBED  c:\work\libwdi\libwdi\libwdi\usbser.cat.in (2018.04.07)
make  all-am
make[3]: Entering directory `/c/work/libwdi/libwdi/libwdi'
  CC       libwdi_la-logging.lo
  CC       libwdi_la-tokenizer.lo
  CC       libwdi_la-vid_data.lo
  CC       libwdi_la-pki.lo
  CC       libwdi_la-libwdi_dlg.lo
  CC       libwdi_la-libwdi.lo
libwdi.c: In function 'wdi_create_list':
libwdi.c:846:8: error: implicit declaration of function 'SetupDiGetDevicePropertyW' [-Werror=implicit-function-declaration]
   if (!SetupDiGetDevicePropertyW(dev_info, &dev_info_data, &DEVPKEY_Device_BusReportedDeviceDesc,
        ^
cc1.exe: some warnings being treated as errors
make[3]: *** [libwdi_la-libwdi.lo] Error 1
make[3]: Leaving directory `/c/work/libwdi/libwdi/libwdi'
make[2]: *** [all] Error 2
make[2]: Leaving directory `/c/work/libwdi/libwdi/libwdi'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/c/work/libwdi/libwdi'
make: *** [all] Error 2

mcuee, 2018-06-24:
Probably the Wiki can be updated to mention the workaround.

mcuee, 2018-06-27:
I have not tried MinGW-w64 sourceforge's personal build of multilib enabled toolchain, They are a bit old also.
https://sourceforge.net/projects/mingw-w64/files/Multilib%20Toolchains%28Targetting%20Win32%20and%20Win64%29/

mcuee, 2018-06-27:
Since you have updated the wiki, no problem to close the ticket.

pbatard, 2018-06-27:
Yeah, at this stage I'm no longer planning to fiddle with anything but the MinGW-w64 provided by MSYS2, coz it's just too much of a pain to maintain. I wish they would complete support of multilib there, because as I mentioned, their gcc executable has multilib support, but they don't provide the needed libraries, so it fails:

```
$ ./autogen.sh
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
configure.ac:12: installing './compile'
configure.ac:13: installing './config.guess'
configure.ac:13: installing './config.sub'
configure.ac:2: installing './install-sh'
configure.ac:2: installing './missing'
examples/Makefile.am: installing './depcomp'
configure: loading site script /mingw64/etc/config.site
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking build system type... x86_64-w64-mingw32
checking host system type... x86_64-w64-mingw32
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe
checking if the linker (C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /mingw64/bin/nm -B
checking the name lister (/mingw64/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -pR
checking the maximum length of command line arguments... 8192
checking how to convert x86_64-w64-mingw32 file names to x86_64-w64-mingw32 format... func_convert_file_msys_to_w32
checking how to convert x86_64-w64-mingw32 file names to toolchain format... func_convert_file_msys_to_w32
checking for C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... file_magic ^x86 archive import|^x86 DLL
checking for dlltool... dlltool
checking how to associate runtime and link libraries... func_cygming_dll_for_implib
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /mingw64/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
[... 135 lines cut ...]
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lgcc
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: skipping incompatible C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/libgcc_eh.a when searching for -lgcc_eh
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: skipping incompatible C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0\libgcc_eh.a when searching for -lgcc_eh
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: skipping incompatible C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/libgcc_eh.a when searching for -lgcc_eh
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lgcc_eh
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: skipping incompatible C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/../../../../x86_64-w64-mingw32/lib/libmoldname.a when searching for -lmoldname
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: skipping incompatible C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/../../../../x86_64-w64-mingw32/lib\libmoldname.a when searching for -lmoldname
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: skipping incompatible C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/../../../../x86_64-w64-mingw32/lib/libmoldname.a when searching for -lmoldname
[... cut ...] 4-mingw32/lib/libmsvcrt.a when searching for -lmsvcrt
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lmsvcrt
collect2.exe: error: ld returned 1 exit status
make[2]: *** [Makefile:497: installer_x86.exe] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/d/libwdi/libwdi'
make[1]: *** [Makefile:410: all-recursive] Error 1
make[1]: Leaving directory '/d/libwdi'
make: *** [Makefile:342: all] Error 2
```