pbatard/libwdi #131: libwdi driver gets "overridden" by badly written HP driver

karel-3d, 2018-07-07:
I am not sure if this is an actual issue or not...

We are a HW maker, we are installing to our users on windows7 driver by using https://github.com/pbatard/libwdi/blob/master/examples/wdi-simple.c . So far so good.

However, we found out that on some users' PCs, the device gets recognized as "HP Printer (BIDI)". When I dug deeper, I found out that the HP driver gets loaded, because in the inf file ([which I copied here to github gist](https://gist.github.com/karel-3d/1624149b5a3652b3596443cc20c3f22c)) there is this line

`%DESCR1%=LEDM,USB\MS_COMP_WINUSB`

and when I look at "driver node strong name" in Windows 7 device manager in the "HP Printer (BIDI)" that is not actually a HP printer but our device, I indeed see

`oem45.inf:Models.ntamd64.6.1:LEDM:1.0.0.227:usb\ms_comp_winusb`

so it seems the HP driver automatically gets to be used on all winusb devices despite libwdi driver being installed through the simple.c - and indeed when the HP thing is uninstalled in device manager (by right clicking -> remove -> uninstall), libwdi driver gets loaded correctly.

My question here is - is it somehow possible to "force" libwdi driver to be used, even when the HP thing is installed? (Since it can be an issue with more badly written drivers than just the HP scanner thing.) Maybe by some change in the libwdi INF, or by some other windows call?

I am looking at [this MS support document](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/driver-matching-criteria), but that is very cryptic.

karel-3d, 2018-07-07:
Hm, there seem to be some other issue in play on the users' computer, since I cannot replicate it on my own windows7 PC, and when both HP driver and libwdi driver are installed I see the correct driver and not HP one. :/

karel-3d, 2018-07-07:
Maybe the issue is that they connected the device *before* the libwdi driver was installed, windows remembers to use HP driver, and then even after installation, the bad driver gets used?

pbatard, 2018-07-28:
I'm afraid I don't think I have any plans to look into your problem as it has nothing to do with libwdi. And considering that nobody else has replied, I will close this issue.

closed 2018-07-28 by pbatard

pbatard, 2018-08-06:
There are driver priorities assigned by Microsoft according to whether the driver comes from Windows Update, and other factors. That is not something libwdi can control, as this is an internal setting that only Microsoft can alter. Otherwise, libwdi would expose that setting.

Therefore, if you are unlucky enough for a driver you don't want to have higher priority than the one you want, then the only solution is to override that driver manually.

karel-3d, 2018-11-29:
for future searchers - if someone finds this issue by googling - it was NOT in HP driver, it was in too short timeout in wdi-simple, see linked issue. 

however, the HP Driver is something that sometimes gets automatically installed by some Windows 7 instances for all unknown winusb devices, for reasons that elude me.