pbatard/libwdi #134: Restoring Original Drivers of Kingst VIS Logic Analyzer

ptrailblazer89, 2018-08-04:
I Wanted to Try Sigrok - Open GUI for this Analyzer as the Supplier Provided Software only has capability of Sampling of Max 4 MHz 

Some People on Internet Suggested it might work for fx2falw as imitation - So I was forced to turn to Zadig to replace original Drivers  with all variations of drivers and combinations provided with Zadig 

Unfortunately Sigrok did not recognize the device, but now the original software wont recognize the Device Either 

I tried a lot tried to { 1. Delete Drivers from USBlyzer 2. Removed  oem<xx>.inf files containing Kingst Keyword} - But Windows 10 Always Installs libwdi(Autogenerated) drivers and oem<xx>.inf files are autogenerated

It would have saved lot of hours for me to just provide option to "UNCONNECT" Zadig from the Device

P.S.: The Supplier also provides driver files which i Installed

pbatard, 2018-08-04:
> to replace original Drivers with all variations of drivers and combinations provided with Zadig

You are supposed to know precisely which driver you need to install before you use Zadig (WinUSB, libusbK, libusb0). Whatever guide you followed should provide that information. If you don't, then of course whatever you're trying to achieve is unlikely to work (and there is nothing Zadig or libwdi can do to help you).

Please understand that Zadig is an __advanced__ utility, meant to be used by people who know what they are doing (and specifically what driver they need to install, and for which device).

> I tried a lot

See [this](https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it).
I can guarantee that the steps provided here, but the secret is to __insist__ and possibly make sure you replug the device in all the ports it's ever been plugged, and go through the uninstall process for each.

Then you will need to reinstall the original driver.

> It would have saved lot of hours for me to just provide option to "UNCONNECT" Zadig from the Device

The problem is that it would take me __months__ to provide that feature (which still won't be the same as what you have in mind), and my time is limited, especially when it comes to libwdi, which is very much low priority project right now for which I have __exceedingly__ little time to devote.

Then again, the project is Open Source, so if someone is willing to invest their own time, I'll be happy to add the feature.

pbatard, 2018-08-04:
> It is provided with sigrok - which is User Utility/Software ...

And I am not responsible for how other people choose to present the software, or how complete the guide they provide is. Clearly the sigrok guide wasn't clear enough to help you find out what you needed to do and which driver to install, so you should take that matter up to them.

If a regular person takes the wheel a formula 1 and crashes, you don't go to the makers of the formula 1 car and tell them they should have made it safer for untrained drivers to pilot one.

It's up to the sigrok people to make sure you have all the data you need to properly use Zadig, as I really don't have time to dumb down the software for people who don't know what they are doing, or add foolproof features.

> Can you say Precise Steps to Install
Kingst Logic Analyzer Driver having replaced with all combinations of drivers {WinUSB, libUSB, libUSBK , USB(CDC)}

No. All I can tell is what I indicated above: Uninstall the driver as many times as you need, until you can reinstall the original one. You may have to reboot and replug your device over and over again (which is also why providing an "uninstaller" is not as easy as it may seem). This is the best advice I can provide here.

ptrailblazer89, 2018-08-04:
This Worked: 
[Robert's Blog - Removing libwdi drivers ](http://mirobetm.blogspot.com/2014/08/removing-libwdi-drivers.html)

1. First uninstall Zidag , Your application (Device)which is not working(preferably with revo-uninstaller)
(Can Reboot Here)

2. (Use [Everything Search ](https://www.voidtools.com/) to Find something.txt (lot easier))
under each oem< xx >.inf entry --->Find  string - (libwdi autogenerated)  in something.txt  file
and to delete that driver   use following Command Line Below:

    pnputil.exe -f -d oem< yy >.inf (Use Force Delete if It doesnt WOrk)

(Also Can use [Drive Store Explorer](https://archive.codeplex.com/?p=driverstoreexplorer) to delete .inf Files)

3. Reboot System
4. Install Original Application With its Drivers 
5. Upon Successful ---> Control Panel ---> < Your Device > ---> Properties --->Driver
    Driver Provider: Shows < Device Manufacturer Name > instead of libwdi

> Then again, the project is Open Source, so if someone is willing to invest their own time ...
> I really don't have time to dumb down the software for people ...

Software/Internet can find unexpected/intricate loop holes in your argument
Software can be **Courteous** and provide option for Restoring Machines Previous Working State

pbatard, 2018-08-04:
> Software/Internet can find unexpected/intricate loop holes in your argument

I don't see how. I clearly stated that I don't have the __time__ to implement the feature you request. If you refute my argument, then you are implying that I am lying about my lack of development time for libwdi, which is a very interesting point to try to make to the developer of a project...

> Software can be Courteous and provide option for Restoring Machines Previous Working State

Software can do tons of things when you have __unlimited__ time to implement them.
However, when you have realistic constraints in terms of time, then, if you are a semi-competent developer, you will establish clear goals as to what should be within the scope of your project and what is out of scope, so that you can deliver something that will benefit the maximum number of people.

Considering that the ability to restore/uninstall drivers is mostly (but not always) requested by people who didn't pay as much attention as they should have with regards to what you were doing, or didn't try to understand what what the action they were about to perform was likely to achieve (i.e. people who might very well benefit more from learning a lesson about not trying things at random, rather than having software put safeguards all over the place), as well as the overall lack of time I know I can spend on libwdi, I deliberately decided that driver uninstallation would remain out of scope from the get go (that is, as long as the expectation would be that __I__ would be the one to implement the feature).

You want to construe that as __uncourteous__ or something else, suit yourself. However, that is not the reality of things.

ptrailblazer89, 2018-08-04:
> I don't have the time to implement the feature you request

Accepted ...
Not Even WARNING-DANGER Signs - Extreme Precaution Must be Observed Have a Look At the Manual - RISK OF BREAKING, SOLE RESPONSIBILITY OF USER etc. - **Message Box** were Shown when the operation was being Performed

> I am lying about my lack of development time for libwdi, which is a very interesting point to try to make to the developer of a project...

 I Did not mean that, what I Meant was(Global/Internet Perspective)
 
 Customer <----> Intermediary Company 1 <-----> Formula 1 car Company (which included no airbags)
 Customer <----> Sigrok                              <-----> Zidag
 
 Consumer software's when provided to customer must "put safeguards all over the place" / include "foolproof features." for safety of People/In General Public Interest.
 Sigrok Software to be used by Consumer Did not Take Care of that when it Distributed the software
 (assuming by your arguments that Zidag is intended for Software Developers (who have capacity to undo what software did) as major Audience and not customers)
 
Whom should the (Hypothetical)Government Sue(Find Fault)(In Which Forum should Question End Up)
{ //This is what I Meant---Whole Framework
+ Customer, --->For Driving So Speedily assuming F1 car included Airbags - which is generally the case
+ Intermediary Company 1, ------> For Distributing **those** F1 Cars to Customers 
+ Formula 1 Car Company------> For Not Providing Air Bags (Lack of FUNDS)
}

Leaving Customers in Unknown state --> UN-courteous

pbatard, 2018-08-04:
> Not Even WARNING-DANGER Signs 

Since Zadig is not designed for people who don't know what they are doing or don't understand that, of course, replacing the driver for a device may mean that your device may cease to be accessible by some applications, that would annoy the people who are the core target audience of Zadig.

Besides, from producing another software (Rufus), that is aimed at the general public and that puts a SUPER LARGE WARNING SIGN that formatting a drive will erase all the data on it, and still receiving way too many people from people who completely seem to disregard that warning, I do have some empirical evidence on how effective such a warning would be for the people it would be meant to target.

As much as you are trying to put arguments that will try to make you look somewhat better, there's no escaping the fact that you tried things completely at random in Zadig, and reinstalled drivers over and over again, which doesn't exactly strike me as a behaviour that you can try to shift the blame to someone or something else (if your car door doesn't open, you don't immediately resort to using a crowbar and then a sledgehammer to make it budge, and then try to blame the car manufacturer for using a handle that you couldn't figure out how to use).

> Consumer software's when provided to customer must "put safeguards all over the place" / include "foolproof features." for safety of People/In General Public Interest.

And I'd like to see Universal Basic Income implemented, in the _general public interest_, as I have seen punctual evidence that, this would greatly improve the overall welfare of our society. However, I am also realistic about the __constraints__ this requires, and how it is not as easy to implement as snapping one's fingers.

> Customer, --->For Driving So Speedily assuming F1 car included Airbags

As far as I'm concerned, a customer who steps in a F1, but doesn't realize that they aren't driving a completely different beast as a sedan, does have some issues, especially if they do run into a problem, and then try to sue the makers of that F1...

ptrailblazer89, 2018-08-06:
> Since Zadig is not designed for people who don't know what they are doing or don't understand that,...

Software/Internet can find unexpected/intricate loop holes in your argument:
1. Generally(but not always) Open Source software's are done to help customers solve problems **immediately** (loop holes-which companies have left off e.g. Everything Search) - not leave them in unknown state.
2. Zadig Landed up in Customer Zone without putting the gravity of situation/Impression of how Software should be handled or how serious the Authors intentions are.

> I do have some empirical evidence on how effective such a warning would be for the people it would be meant to target.

Even Police, Military People Show Signs and Warn Users Before Shooting or Taking Drastic Action Needless{"This is Police Put Down Your Weapon Now" ,"Restricted Personal Only", ....}

>  and my time is limited, especially when it comes to libwdi,

Still If there is no time to implement the Software --- Could have shown a flowchart of how to undo/Restore original drivers which takes less time(drawing flowcharts) than implementing and integrating the code into software(People with bit (but not in Software DRIVERS) Software/Computer Knowledge would benefit) ....

> there's no escaping the fact that you tried things completely at random in Zadig, and reinstalled drivers over and over again, which doesn't exactly strike me as a behavior that you can try to shift the blame to someone or something else

Accepted, How Would Normal Customer Try {Using Permutations and Combinations - The Easiest Way First}

> people who might very well benefit more from learning a lesson about not trying things at random

Just wanted to give a software a try to see what it can do ...
E.g. Take {CCleaner, Glary Utilites, ...} Do People/Customers Understand how and what exactly they clean in registry they only  press buttons and see if their computer became Faster (and take the Word of Author that it does Good). Software Evolved in Internet that way ... Lot are over exaggerated Features-Which Have No Scale or mathematical metrics to measure their potential properly. 
Take AVG Tune Up Utilities - 
[Why Does Glasswire “Load” My System as shown by AVG-TuneUp Utilities](https://forum.glasswire.com/t/why-does-glasswire-load-my-system-as-shown-by-avg-tuneup-utilities/6511)

> if your car door doesn't open, you don't immediately resort to using a crowbar and then a sledgehammer to make it budge,  and then try to blame the car manufacturer for using a handle that you couldn't figure out how to use

Over-exaggeration, Normal Customers Try to Open the Car Door with combinations of buttons and levers on Car handle, They assume it is normal for one combination to work and the door opens.
(the designer takes care of this to make it ****especially easy** for Customer**(not designer) and the intended operation **which user expects** is accomplished)
Customer Will Not Know/think that this car is specially designed for Developers and there is small button in handle when pressed Locks out entire opening Mechanism and cannot open door unless given to developer/experienced person to undo it.

pbatard, 2018-08-06:
> can find unexpected/intricate loop holes in your argument:

That's not an argument. That's how I designed the software. I __never__ designed Zadig to be used by people who have no clue about driver installation.

You want to rewrite history, fine. But that won't make it true.

>  Open Source software's are done to help customers solve problems immediately

[Citation needed]. Zadig is a byproduct of libusb, which is a library that is used __exclusively__ by developers, not end-users/"customers". So, no, Open Source software is not done to help customers/end-users. There are many, many example of Open Source software that disproves your statement. Just because, as an end-user, you only see the end-user side doesn't mean there doesn't exist an equally vast non end-user side.

> Zadig Landed up in Customer Zone

Zadig is a __sample__ of libwdi, which you will have hard time qualifying as a "customer" friendly library (since libraries are hardly something "customers" tend to concern them with). See https://github.com/pbatard/libwdi/tree/master/examples. This is a not a standalone project, since the actual standalone project itself a developer focused library. Need to say more?

I chose to present Zadig as a standalone download in the hope that it would help some people. Now if you want to pretend I should be held responsible of how people __choose__ to perceive the software and that I no longer have any __rights__, as the developer, to declare what it was designed for, and who is supposed to use it, then all I can say is good luck trying to impose your dictatorial statements onto developers...

> Still If there is no time to implement the Software

Ah, the good old "anything that I don't need to do myself, and that I only have a very partial view of should only take 5 minutes to solve". Good luck with this as well.

> Over-exaggeration, Normal Customers Try to Open the Car Door with combinations of buttons and levers on Car handle, They assume it is normal for one combination to work and the door opens.

As far as I'm concerned, trying to use all drivers presented to you, without stopping a single minute to think and see if that is likely to accomplish anything is about as stupid as using a crowbar and then a sledgehammer to try to open a car door with a locking mechanism you simply can't figure out. Didn't want to spell it out openly, but since you are still pressing on that issue...

Developer design with a specific goal in mind. I can understand that you might be angry that this goal doesn't coincide with the goal you would like the same software to have, and why you would then try to present bullshit argument to pretend software designers should simply do the will of ill-advised users. Sadly, this is not how software development works. But, even if I am going to close this issue, I'll be happy to humour you if you still want to waste your time pretending that I should simply do your bidding because, as the "customer", you know better than the developers who designed the software...

closed 2018-08-06 by pbatard