pbatard/libwdi #136: Zadig not work @USB3.0

AndySong2018, 2018-08-18:
zadig not work at USB3.0  .

IUSB3\VID_XXXX&PID_XXXX.

AndySong2018, 2018-08-20:
> You need to provide a log. Unless you do so, I will have to close this issue.

Thanks .
My device is Typc C with ethernet and 3 USB2.0 ports.

AndySong2018, 2018-08-20:
Thanks .
My device is Type C with ethernet and 3 USB2.0 ports.

________________________________
From: Andy Song <Andy_Song_2018@outlook.com>
Sent: Monday, August 20, 2018 12:25 PM
To: pbatard/libwdi; pbatard/libwdi
Cc: Author
Subject: Re: [pbatard/libwdi] Zadig not work @USB3.0 (#136)

Thanks .
My device is Typc C with ethernet and 3 USB2.0 ports.

AndySong2018, 2018-08-20:
Please see the log.

Thanks.

________________________________
From: Andy Song <Andy_Song_2018@outlook.com>
Sent: Monday, August 20, 2018 12:26 PM
To: pbatard/libwdi; pbatard/libwdi
Cc: Author
Subject: Re: [pbatard/libwdi] Zadig not work @USB3.0 (#136)

Thanks .
My device is Type C with ethernet and 3 USB2.0 ports.

________________________________
From: Andy Song <Andy_Song_2018@outlook.com>
Sent: Monday, August 20, 2018 12:25 PM
To: pbatard/libwdi; pbatard/libwdi
Cc: Author
Subject: Re: [pbatard/libwdi] Zadig not work @USB3.0 (#136)

Thanks .
My device is Typc C with ethernet and 3 USB2.0 ports.

pbatard, 2018-08-20:
You still haven't provided a log. Again, unless you do so, I will just have to close this issue.

You will need to use the issue tracker at https://github.com/pbatard/libwdi/issues/136 to attach your log. It will not work through e-mail.

pbatard, 2018-08-20:
Also, if I understand you correctly you are trying to __replace__ the driver for a USB 3.0 hub (that also includes ethernet) with WinUSB? What exactly are you trying to accomplish? Why do you think you need to do that? And with what WinUSB based application do you plan to use that device (knowing that the minute you replace the driver, it will obviously cease to function as a hub + ethernet device that Windows can use).

Do you really understand what Zadig is meant to be used for and what it does?