pbatard/libwdi #139: WCID not installing compatible driver

dvj, 2018-08-30:
I have a USB device in which I've supported WCID. When plugging in to a Windows 10 machine the first time, drivers get installed. The installed driver does not seemingly allow my device to work with libusb though.

[image]
The driver installs as "v10.0.10586.0", whereas Zadig installs v6.1.7600.16385.

When installing the driver with Zadig, the driver files also include WdfCoInstaller01011.dll and WinUSBColInstaller2.dll, whereas the WCID auto-installed driver does not.

I cannot use a program to connect to the device through libusb. Is there something I am missing about how this should be working?

I understand this is probably not a libwdi issue per se, but reading through the WCID Devices wiki page makes it seem like just providing a WCID compatible firmware would be enough to make the device work with libusb. The libwdi wiki pages are by far the best resource on this topic.

CraigHutchinson, 2020-06-24:
@dvj  Hi, did you manage to resolve your issue in the end? I am experiencing a similar problem on my device and was wondering if it could be WinUSB v10 vs v6 related (Confirmed via device Manager also)

dvj, 2020-06-26:
@CraigHutchinson Even after following Pete's great advice, and spending some more time with our device, I did not get this to load the correct version of winusb without intervention.  We ended up dropping Windows support for the device, so unfortunately I haven't put more time into this since then.

That said, your issue sounds similar to what I was running into, so please do post if you make progress!

CraigHutchinson, 2020-06-26:
@dvj I made very good progress in my problem. I havn't however verified this was the only bit that fixed the problem:
- Identified an error using USB View from Microsoft which reported an erroneous descriptor for IAD on my composite device
- Rebuilt LibUSB from source-code for latest master (think this may be the key bit but not confirmed and now it works don't have time to re-confirm)
- After adding the WCID Extended properties to add a InterfaceGUID then the PC application could use my device successfully. 

NOTE: I feel the critical part may still be LibUSB compiled from latest source as there was a report that LibUSB could be the root of the issue we are/where facing re composite-devices on windows.

TLDR version; Update LibUSB and

NoahAndrews, 2024-10-14:
This issue comes up in a web search for this, so I'll put this here in case it helps someone else. My employer's product does support WCID, and that works without issue the vast majority of the time. Sometimes though, somehow Windows gets the impression that it does not. I'm able to fix that issue using USBDeview. The flags I used were `USBDeview.exe /RunAsAdmin /remove_by_pid 0483;A30E` (replacing the VID and PID with those of your device). After disconnecting and reconnecting the device, it starts working correctly.