pbatard/libwdi #141: The SelfSignFile function is intercepted and prompted by antivirus software

hmxcpu, 2018-09-17:
The libwdi-1.3.0 package is compiled after download, runs "zadig" and is intercepted and prompted by anti-virus software at the SelfSignFile function "RemoveCertFromStore" and "AddCertToStore".
But with your "zadig.exe," there's no such thing.

[image]

hmxcpu, 2018-09-17:
No problems were found with libwdi, and the software installed on my computer was 360 security guard.
But I'm surprised to run the "zadig.exe" you end up providing, without prompting.

hmxcpu, 2018-09-17:
Is the software to be compiled through digital signature, will not be 360 prompt?

pbatard, 2018-09-17:
Possibly. I __really__ can't answer that question for you. I don't use 360 security guard, so, __of course__, I don't have the faintest idea what can trip it, and I certainly did not, at any stage, develop libwdi and thought "well, let's do this so as not to trip security guard".

If you want to get to the bottom of this issue, I will respectfully ask that you conduct your own investigation, as I neither have the time, nor the interest, to do that for you.

hmxcpu, 2018-09-18:
Thank you for your reply. According to your prompt, I have submitted my question to 360.