pbatard/libwdi #143: Install Windows-own USBSTOR file back

LitzeSmurf, 2018-09-24:
Using USB communication with my three subsequent TEASI GPS devices on Windows 7 has been a pain in the ass for a couple of years. Nearly always I got either the well known message "_Drive must be formatted_" or any entry missed in the Explorer. When I discovered Zadig recently, I assumed this struggle would be over. But it took [this trick](https://community.spiceworks.com/how_to/1473-how-to-force-a-manual-driver-isntall-in-win-7) to install the 'original' **Windows driver USBSTOR** back, the one that at the connection of the device will be called in a PC with a fresh Windows installation. Surprisingly, it now invoked a _serious warning_ from Windows itself . . . Anyhow, the issue is completely solved this way.
Is there a reason why this USBSTOR driver misses in Zadig? I assume that the hidden twist that my Windows once made for these TEASI devices can happen with other users also. Your comment would be appreciated! Thanks.

pbatard, 2018-09-24:
> Is there a reason why this USBSTOR driver misses in Zadig?

Yes:

1. __All__ of its driver installation files are provided by Windows, so you don't actually need an external application to install it.
2. It's not a generic driver. If I were to add support for every __non-generic__ USB driver on the planet (because there's no reason to do it only for USBSTOR. Why should __you__ have preferential treatment with your request, and not __others__ who have requested a similar thing for a different USB driver?), then the application would be hundreds of MB in size, and would be completely unmaintainable. There's no middle ground there.
3. Your issue about __reverting__ (installing the driver back) is [covered in the FAQ](https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it), and is out of scope of Zadig. Zadig is not and will never be designed to be a non tech-savvy application. It aims at letting you complete a driver installation operation that Windows will not allow you to do on your own, so that's what it does. On the other hand, installing or replacing a driver with a native one is something Windows will let you do, and you are 100% supposed to know how to install Device Manager to achieve that. The goal of Zadig is not and will never be to duplicate what Device Manager can do. If you can do it with Device Manager (and uninstalling one of the drivers provided by Zadig to restore the original Windows driver __is__ something you can do with Device Manager), then Zadig will not duplicate that feature.

&rarr; Rejected.

closed 2018-09-24 by pbatard

LitzeSmurf, 2018-09-24:
Thanks for your quick response. I merely asked a simple question and I solved the issue already. So I can easily accept your rejection. Nevertheless, you better accept that in my case unfortunately this issue could NOT be solved by Windows (device manager). I tried that so many times and that is only one reason why it took me so long. But fortunately that is what was addressed in the link I gave. And I am so happy I found out that it finally worked!
It's good to see that you keep your scope well. Have a nice day.