pbatard/libwdi #144: error in cross-compilation on Linux Debian - windows.h not found

YannickStephane, 2018-09-27:
Hi, 
After a successful "configure", when i run "make" I have an error because windows.h is not found and I don't know why.
configure command is :
./configure --host=i686-w64-mingw32 host_alias=mingw32 --with-wdkdir=../wdk --prefix=$(pwd)libwdi/.libs

make  all-recursive
make[1] : on entre dans le répertoire « /home/mistertest/jenkins/workspace/libb2binterface-windows-32bits-build/lib/libwdi »
Making all in libwdi
make[2] : on entre dans le répertoire « /home/mistertest/jenkins/workspace/libb2binterface-windows-32bits-build/lib/libwdi/libwdi »
  CC       installer_x86-installer.o
installer.c:24:21: fatal error: windows.h: Aucun fichier ou dossier de ce type
 #include <windows.h>

Please can you give a complete procedure how to make a good cross-compilation ?
Thank you!

karel-3d, 2018-11-29:
We made cross-compilation work in docker here

https://github.com/trezor/trezord-go/blob/master/release/windows/Dockerfile.wdi_driver

note - we use our slightly changed fork of the code, but cross compilation works the same with our version and the version here.

karel-3d, 2018-11-29:
In your case, you are missing what is in the docker as `mingw-w64-common`, that is, actual windows header files.

If your OS is debian, install mingw-w64-common; ubuntu will probably have something else; I have no idea about fedora et al

pbatard, 2018-12-27:
I don't think I'll have a chance to look into this any time soon (too busy). I suggest you try to troubleshoot it yourself and, if you find there is an issue with the libwdi build scripts, submit a patch.

closed 2018-12-27 by pbatard