pbatard/libwdi #149: Zadig constantly displays the error "Driver installation canceled by user".

Foreld, 2018-12-27:
As it was already written in the title: it constantly produces an error "Driver installation canceled by user". How not tried, I did not find the answer. I have administrator rights, I tried on other versions, but also unsuccessfully. Tell me how to fix it.
In addition, in the lower left corner is written "Updates: Unable to acces version data".

pbatard, 2018-12-27:
You will need to provide your full log. Go to _Options_ &rarr; _Advanced Mode_, then retry what you were attempting and copy/paste the __full__ log.

Note: The `Unable to access version data` message is normal and has nothing to do with your issue (It happens because we had to switch servers. This will be fixed in a future version.)

Foreld, 2018-12-27:
Zadig 2.3.701
Windows 10 64-bit (Build 17134)
ini file 'zadig.ini' not found - default parameters will be used
default driver set to 'WinUSB'
1 device found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1F3A&PID_EFE8&REV_02;3
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_FF&Prot_FF
libwdi:debug [wdi_create_list] Driverless USB device (13): USB\VID_1F3A&PID_EFE8\5&15C311E1&0&8
libwdi:debug [wdi_create_list] Device description: 'CLV-S-NESY'
Checking for Zadig updates...

Checking release channel...

Updates: Unable to acces version data

0 devices found.
1 device found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1F3A&PID_EFE8&REV_02;3
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_FF&Prot_FF
libwdi:debug [wdi_create_list] Driverless USB device (13): USB\VID_1F3A&PID_EFE8\5&15C311E1&0&8
libwdi:debug [wdi_create_list] Device description: 'CLV-S-NESY'
Using inf name: CLV-S-NESY.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\Foreld\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\Foreld\usb_driver\CLV-S-NESY.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Foreld\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Foreld\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'clv-s-nesy.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Foreld\usb_driver\clv-s-nesy.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Foreld\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Foreld\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] successfully created file 'C:\Users\Foreld\usb_driver\CLV-S-NESY.cat'
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] acquired existing key container
libwdi:debug [CreateSelfSignedCert] generated new keypair
libwdi:warning [CreateSelfSignedCert] could not create self signed certificate: [#800706D9] � ������� ����������� �������� ����� �� �������� ��������� �������� �����.
libwdi:warning [wdi_prepare_driver] could not sign cat file
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter CLV-S-NESY.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_1F3A&PID_EFE8\5&15C311E1&0&8'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_1F3A&PID_EFE8&REV_02;3'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-3972951681-1334078062-1713021035-1001'
libwdi:debug [installer process] using syslog 'C:\Windows\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_1F3A&PID_EFE8&REV_02;3 - please wait...
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_1F3A&PID_EFE8&REV_02;3]
libwdi:debug [syslog] >>>  Section start 2018/12/27 22:02:36.079
libwdi:debug [syslog]       cmd: "C:\Users\Foreld\usb_driver\installer_x64.exe" "CLV-S-NESY.inf"
libwdi:debug [syslog]      ndv: INF path: C:\Users\Foreld\usb_driver\CLV-S-NESY.inf
libwdi:debug [syslog]      ndv: Install flags: 0x00000001
libwdi:debug [syslog]      ndv: {Update Device Driver - USB\VID_1F3A&PID_EFE8\5&15C311E1&0&5}
libwdi:debug [syslog]      ndv:      Search options: 0x00000080
libwdi:debug [syslog]      ndv:      Searching single INF 'C:\Users\Foreld\usb_driver\CLV-S-NESY.inf'
libwdi:debug [syslog]      dvi:      {Build Driver List} 22:02:36.096
libwdi:debug [syslog]      dvi:           Searching for hardware ID(s):
libwdi:debug [syslog]      dvi:                usb\vid_1f3a&pid_efe8&rev_02;3
libwdi:debug [syslog]      dvi:                usb\vid_1f3a&pid_efe8
libwdi:debug [syslog]      dvi:           Searching for compatible ID(s):
libwdi:debug [syslog]      dvi:                usb\class_ff&subclass_ff&prot_ff
[... 52 lines cut ...]
libwdi:debug [syslog]      flq:                Copying 'c:\users\foreld\usb_driver\amd64\WinUSBCoInstaller2.dll' to 'C:\Users\Foreld\AppData\Local\Temp\{d114b971-728c-694a-b77c-d85f7986d117}\amd64\WinUSBCoInstaller2.dll'.
libwdi:debug [syslog]      flq:                Copying 'c:\users\foreld\usb_driver\CLV-S-NESY.cat' to 'C:\Users\Foreld\AppData\Local\Temp\{d114b971-728c-694a-b77c-d85f7986d117}\CLV-S-NESY.cat'.
libwdi:debug [syslog]      flq:                Copying 'c:\users\foreld\usb_driver\clv-s-nesy.inf' to 'C:\Users\Foreld\AppData\Local\Temp\{d114b971-728c-694a-b77c-d85f7986d117}\clv-s-nesy.inf'.
libwdi:debug [syslog]      sto:           {Copy Driver Package: exit(0x00000000)} 22:02:36.133
libwdi:debug [syslog]      pol:           {Driver package policy check} 22:02:36.196
[... cut ...]
libwdi:debug [process_message] switching timeout back to finite
libwdi:debug [installer process] operation cancelled by the user
libwdi:debug [process_message] installer process completed
Driver Installation: Cancelled by User
1 device found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1F3A&PID_EFE8&REV_02;3
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_FF&Prot_FF
libwdi:debug [wdi_create_list] Driverless USB device (13): USB\VID_1F3A&PID_EFE8\5&15C311E1&0&8
libwdi:debug [wdi_create_list] Device description: 'CLV-S-NESY'

Foreld, 2018-12-28:
All included, still does not work.

pbatard, 2018-12-28:
As I said, this is a problem with your environment. Zadig cannot sign the CAT file because your system is misconfigured. Maybe you stopped a service, maybe you changed something else, but all I know is __YOU__ will have to figure out how to sort your environment so that the CAT signing process works, because there's nothing Zadig can do if the signing services it relies on cannot be provided by Windows.

Foreld, 2018-12-28:
All right, thanks anyway.

ikaroskurumi, 2021-09-03:
hola yo tambien recibia ese error cuando tenia un windows lite como el minios la  solucion que le encontre era de desabilitar la necesidad de controladores firmados en widows 10  por medio del reinicio avanzado.  ya que el error esta en que  ZAGDIG no puede firmar tus controladores y por ende windows no te los va a instalar por seguridad

mcuee, 2021-09-05:
> hola yo tambien recibia ese error cuando tenia un windows lite como el minios la solucion que le encontre era de desabilitar la necesidad de controladores firmados en widows 10 por medio del [...]

Google Translate:
hello I also received that error when I had a windows lite like minios the solution I found was to disable the need for signed drivers in widows 10 through advanced restart. since the error is that ZAGDIG cannot sign your drivers and therefore windows will not install them for safety

mcuee, 2021-09-05:
Disable the need for signed driver is just a work-around. Apparently the Windows installation is problematic. It is better to find the root cause of the issue than trying to disable the Windows security feature (which is meant for developers only).

mcuee, 2021-09-05:
https://github.com/pbatard/libwdi/issues/225
You should try the latest version 2.6 to see if the issue goes away.

erkobg, 2021-09-07:
> #225
> You should try the latest version 2.6 to see if the issue goes away.
Not working
https://github.com/pbatard/libwdi/issues/225#issuecomment-914253615

pbatard, 2021-09-07:
Note that code `[#800706D9]` which is what OP is observing, is for `This system is missing required cryptographic services`. So I don't think using 2.6 will help when Windows indicates that it is missing cryptographic components.

This is what OP will need to fix if they want to run Zadig.

erkobg, 2021-09-07:
Is there any particular  Cryptographic service I need to check ? or missing component?