pbatard/libwdi #152: Can i change back?

syuasahi, 2019-01-10:
I change my device(camera) to Winusb and it worked. 
But i want to change back to WUDFWpdMtp  cause i can't use a regular  application and also can't find folder.
How can i fix this problem. 
Sorry English not my mother language

pbatard, 2019-01-10:
https://github.com/pbatard/libwdi/wiki/FAQ#help-zadig-replaced-the-driver-for-the-wrong-device-how-do-i-restore-it

closed 2019-01-10 by pbatard

syuasahi, 2019-01-10:
I had try that way but still it become another one not a origin one. 
i try it at my old laptop and i run Zadig and it show up is different and can function.
i got the screenshot at this. 
sorry for make this problem.
https://photos.app.goo.gl/LZQKoq1Dms3vzwAk6

pbatard, 2019-01-10:
Sorry, if you managed to remove the drive installed by Zadig, then you will have to find by yourself how to reinstall the one you want.

Zadig did exactly what you told it to do: You told it to replace the existing driver for your device with WinUSB, and it replaced the drive for your device with WinUSB. Zadig does __NOT__ store the previous driver, so if you find that, after uninstalling WinUSB, the old driver you want is not reinstalled, then you'll have to check with the manufacturer of your device how to reinstall it.

syuasahi, 2019-01-10:
Ok, i think i need to find someone have more computer skill than me to fix it.
Still thank for your help.

YoungCarti, 2025-07-31:
I think you need to go to Device Manager, search for the device and right click on it and roll back the driver