pbatard/libwdi #153: Update all Wiki links

E3V3A, 2019-01-20:
Most links to MS sites are broken. Please update the links.

pbatard, 2019-01-20:
The Wiki is open to every GitHub users for edition. If you found some broken links, can you please update them? Remember that Open Source is usually a collaborative effort...

E3V3A, 2019-01-20:
I know and usually do, but in this case I am so new to using this project that I am not able to guess what to look for in any new links that would replace what was there.

pbatard, 2019-01-20:
Well, can you at least point to the links that you have found to be broken. Clearly, if you have reported this issue, it's because you have this data, and providing this is will be a lot more helpful than waiting for a developer, who is busy on many other things, to manage to find some time to try to figure out where exactly are these links that you say are broken. Again, collaborative effort. Thank you.

Mihaylov93, 2019-01-24:
I updated some of the Microsoft broken links. 

Here is the list of the updated links:
```
Determining Whether a Device Is Plugged In and removal of phantom devices
Writing a Device Installation Application
Determining Whether a Device Is Plugged In and removal of phantom devices
Troubleshooting Device Installation with the SetupAPI Log File
```

pbatard, 2019-03-25:
Links have been updated. Closing.

closed 2019-03-25 by pbatard