pbatard/libwdi #155: Windows 10 on ARM64 support

gbaman, 2019-03-24:
I am well aware this is probably pushing it, but I have an ARM64 based Windows 10 Pro machine sitting in front of me currently and am attempting to get it working with an RTL-SDR. 

Zadig runs fine and doesn't seem to have any issues, unfortunately though when I select the usual "Bulk-In, Interface (Interface 0)" and hit Install Driver for "WinUSB", I get a "The driver installation failed" error.
I have attached the log. 

I have no idea if the drivers etc will work on this machine, will admit I am no major expert in Windows 10 Pro on ARM64 (simply a machine that has been handed to me for this project), but am intrigued if it would be possible to make it work.
Finally, the machine itself is an HP Envy x2.

[zadig.log](https://github.com/pbatard/libwdi/files/3000964/zadig.log)

pbatard, 2019-03-24:
> Zadig runs fine

Wait, are you running the x86 version of Zadig that is published at https://zadig.akeo.ie?
That will not work, first of all because it will run on ARM64 in x86 emulation mode (which I doubt Microsoft allows installing ARM64 drivers from) and second, because there are __no__ ARM64 embedded in Zadig in the first place, so even if it "worked", it wouldn't be able to find the files it requires for ARM64.

Also:
```
Windows 10 32 bit (Build 17134)
(...)
Driver Installation: FAILED (Attempted to use a 32 bit installer on a 64 bit machine)
```
Zadig detects a 32-bit x86 emulation (likely because that's what Microsoft used by default) and therefore tries to run the 32-bit installer, which of course the system doesn't like.

For Zadig to work on ARM64, I'd have to recompile an ARM64 version and edit the embedded files to include ARM64 files.

I may do that __eventually__, but I'm afraid I'm too busy with other stuff to look into that in the short term (short term meaning 0-6 months).

Now, if somebody else wants to take a stab at it and send a patch I can review, that's a different story...

gbaman, 2019-03-24:
Haha yeah, I was installing a batch of random laptops with SDR# for use in a school next week (for use with High Altitude Ballooning STEM project) and hadn't even crossed my mind that this one was different till I hit the error, then started digging around.

What you are saying does make sense, it does only emulate x86 based apps and not x86_64 based apps full stop.
If you have somewhere rough you could point me to on where to start, I don't mind taking a glance at recompiling it. No guarantees though... :)

pbatard, 2019-03-25:
> If you have somewhere rough you could point me to on where to start

Oh boy...

* Well, first you'll have to find if and where MS provides ARM64 versions of their CoInstaller DLLs.
* Once you have that, you'll need to edit [this mess](https://github.com/pbatard/libwdi/blob/master/libwdi/embedder_files.h) to handle ARM64 (and while you're at it you may also want to add ARM).
* You'll also want to edit [this](https://github.com/pbatard/libwdi/blob/master/libwdi/winusb.inf.in) and [this](https://github.com/pbatard/libwdi/blob/master/libwdi/usbser.inf.in) to add ARM64 support.
* Then you'll need to edit the MS project files so lbwdi/Zadig can be compiled for ARM64 (and while you're at it you may also want to add ARM - btw [this](https://pete.akeo.ie/2017/05/compiling-desktop-arm-applications-with.html) might help you.
* There's also some more stuff that will be needed, such as updating the check for update in Zadig so that the ARM64 version doesn't try to download the x86 version, or reporting the arch on startup, but I've already done that [in Rufus](https://github.com/pbatard/rufus), so you don't really have to do it as I should be able to reuse that code.
* Last __but not least__ you will need to test, test and test again to make sure that your changes work on ARM64 machines! Adding the code is 20% of the work. Testing is 80%!

vielmetti, 2019-04-28:
Thanks for writing this up - I am also trying to get Zadig running on Windows 10 on ARM64 to support a Lenovo Yoga C620 using an Adafruit USB-to-serial cable so I can talk to a Jetson Nano.

Thanks for the detailed situation report here, I'll hope to get some help to look at this.

jkunkee, 2019-10-13:
WOW64 programs - so x86 on x64 as well as on ARM64 - can't install drivers. The API exists but just returns an error.

The ARM64 SDK is supposed to be fully featured, so if you can't find the DLLs you need, let me know the SDK piece you'd expect them to be in and the path you expect the file to be at and I can talk to a friend.

libusb v1 recompiles for ARM64 by just retargeting the VS projects for Windows 10. (VS2019 has some simplifications for this.)

ARM64 should be pretty straightforward to add as a configuration under VS 2017 or VS 2019--I recommend 2019.

You may find IsWow64Process2 handy for probing architectures, but to run on Windows 7 you'll have to use GetProcAddress to see if it's available.

ARM32 support is likely not going to be useful unless you port to IoT Edition, but VS does make it easy.

I just got an RTL-SDR and would be happy to test things on my Windows 10 on ARM systems, especially if there's a decent way to make sure things are uninstalled. (Just the 'delete drivers, yes' workflow in Device Manager, I expect.)

(Hi @vielmetti ! If you to get some time to look at this, hit me up--I'd be happy to at least kibbitz. I think these machines have great potential for mobile maker scenarios.)

(Yes, I work for Microsoft. I even work on Windows 10 on ARM(64) and have ported OSS projects as part of my day job, but I am here as a hobbyist, user, and fan and not as an employee or representative of my employer.)

pbatard, 2019-10-13:
Hi Jon,

First of all, whether in an official position or not, it's nice to know that there are some people from Microsoft looking at independent projects that relate to the installation of Windows drivers. And I'm actually glad you decided to chime in and offer to see if you can use your connection, because you may actually be able to help with the biggest roadblock I have at the moment (besides time to work on this feature) which is best illustrated by the picture below:

[image]

Namely: if you install the latest Windows 10 WDK, you still don't get any of the WDF and WinUSB coinstaller DLLs for arm64 (Still no `arm64` directory). And what's more, even for ARM, there's no `winusbcoinstaller2.dll` provided, which means we can't support that platform either.

At the very least, libwdi requires the following 2 DLLs to be able to support a platform:
* `WdfCoInstaller01011.dll`
* `winusbcoinstaller2.dll`

So, if possible, can you find out with your colleagues what the story is with regards to finally providing these files for ARM/ARM64 in the WDK, just like they are provided for x86 and x64?

With regards to your other points, I agree that VS2019 has made things easier for ARM64, and I'm planning to retarget libwdi for VS2019 when I get a chance (Since I'm already using VS2019 internally). My only issue here is that, whereas AppVeyor added the VS2019 target no that long ago (but still not in an official mode), they don't have MinGW support working on that target. So if I switch from VS2017 to VS2019, my MinGW libwdi builds will break, and I'd rather not have to go through what I had to do for Rufus, which is to use a mix of 2017 and 2019 (and still end up with a Coverity that doesn't work &mdash; C'mon AppVeyor, update your platforms already!!).

Apart from that, I agree that adding ARM64 support shouldn't be that difficult (as usual, it's going to be the ability to allocate time to complete that task that'll be a challenge). I'll keep in mind your recommendation to use `IsWow64Process2()`, as this seems more useful than `IsWow64Process()` indeed.

jkunkee, 2019-10-14:
That's what my system shows too; I'll see what I can do.

jkunkee, 2019-10-14:
Ah, that would do it. I got a quick answer: the CoInstallers are not needed on Windows 10 on ARM because the first version to ship was 1709. You can drop the references to them for ARM64. (I'm afraid I'm currently rather behind on what the coinstallers do, so I'm probably not much more help there. :/

pbatard, 2019-10-14:
Thanks for the info.

This is actually going to make things slightly more complex for ARM/ARM64, coz we'll need to conditionally sort out a `[USB_Install.CoInstallers]` section from the [`winusb.inf` template](https://github.com/pbatard/libwdi/blob/master/libwdi/winusb.inf.in)... I'm hoping we can filter these things out with `[USB_Install.CoInstallers.x86]` and `[USB_Install.CoInstallers.amd64]` but this will require some testing. And of course I'll need to add conditional logic on whether to search for the coinstallers or not depending on the platform.

No idea when I'll have a chance to look into that, but most likely it won't be before a couple of months... I do appreciate the help however, as it solves one of the main question I had about how we might be able to proceed for ARM/ARM64.

vielmetti, 2020-02-23:
This writeup from @jkunkee gives a good state-of-the-art report, if you find this issue you'll be interested in https://daskunkee.blogspot.com/2020/02/tech-esp32-programming-from-arm64.html

sidd-kishan, 2021-06-06:
Please pursue it we at Windows on Rasberry Pi community will be glad to extend support in testing your drivers and tools for ARM64. We have successfully booted WOA on Rasberry Pi 4 B+ and hence have Cheap ARM64 Windows On ARM device with sort of current hardware but way cheaper than a surface pro device.

pbatard, 2021-06-06:
Yes, I am well aware of it. I am [one](https://github.com/tianocore/edk2-platforms/commits?author=pbatard) of the many people who helped develop the UEFI firmware needed to run Windows on the Raspberry Pi and I also [helped figure out a way to run Windows off the Pi 4 USB 3.0 ports while using more than 1 GB RAM](https://pete.akeo.ie/2020/07/abusing-microsofts-symbol-servers-for.html). And the reason I was involved in this effort was __precisely__ so that I could have access to a cheap platform to test ARM64 versions of Rufus or libwdi.

However, the downside of all this is that I find myself having to allocate time to maintain the Raspberry Pi firmware... which, outside of other prioritization matters (I'm afraid that libwdi will never be that high a priority for me) actually takes time away from tasks such as adding Windows 10 ARM64 support for libwdi...

jkunkee, 2021-06-09:
One day, there will be a low-cost, supported options for developers--I get to work on one--but for now the [Snapdragon Development Kit](https://www.qualcomm.com/news/releases/2021/05/24/qualcomm-enhances-support-developers-new-snapdragon-developer-kit-windows) is slated to be released this summer. It won't be on a patch with the indomitable, inexpensive, and community-maintained Raspberry Pi, but Qualcomm will provide and maintain the BSP and is intended to be much less than $1000.

dougmsbbs, 2021-07-02:
I'd like to chime in here to say please pursue this. Getting a RTL up and running on a Pi 4 running Windows 10 ARM64 would be a real boost to a lot of people. I have two projects stalled just for lack of a driver I can install for RTL-SDR's. I'd be happy to help do any testing needed when it gets that far.

Or is there some other way to get an RTL driver loaded, maybe through pnputil?

mcuee, 2021-07-06:
How many people really care about Windows on ARM? I am an ARM fan. My main computer is the Mac Mini M1 (based on ARM64). My main Linux machine is the Raspberry Pi 400 (along with some other ARM boards). But I do not really care for Windows on ARM. My work and home Windows laptops are running on Intel CPU (AMD CPU is also okay).

Ref: libusb does support ARM32/ARM64 on Windows with VS, but not MinGW.
https://github.com/libusb/libusb/issues/921

pbatard, 2021-07-06:
Hi Xiaofan,

I think more people care about Windows on ARM than you give them credit for. Having worked on the effort to make Windows run on the Pi 4, I can tell you that there is genuine interest for an cheap yet powerful enough ARM64 platform that can run modern Windows, as buying into the intel/AMD framework can still be prohibitive for a lot of people for varied reasons.

And the thing is, Microsoft is perfectly placed to compete with Apple on ARM at a junction where apple will never ever go: _cheap platforms_.

In other words, if Microsoft embraced cheap ARM64 platforms like the Pi 4 (which we know can run Windows 10, or even 11, with "good enough" performance for every day tasks like web surfing, e-mail, document editing and so on), they would probably find a vast sway of users to follow them there, that aren't going to find their match with other options, and especially __not__ Apple MX or Linux on ARM. And that's because the existing Windows ecosystem, with a market dominance that means that if you're looking for an application chances are that you're always going to find at least a Windows version of it, is way too attractive to pass when given the opportunity. Which eventually means that, for a lot of users with limited budget, the choice is between secondhand intel/AMD platforms that run Windows, but that come with the drawbacks of being secondhand, or modern ARM64 platforms that may actually turn out to cheaper than secondhand intel/AMD ones, and run Windows.

So, provided that Microsoft are able to read the writing on the wall (which, judging from what's happening with the Pi 4 Windows effort, they sadly don't appear to be able to do. But then again, it wouldn't be Microsoft if they were able to make decisions that actually benefit Windows users in the long run...), you should find that quite a lot of people will eventually care about Windows on ARM, and, even if Microsoft really does make the beginnings of that platform a lot less inviting that it has any rights to be (because, right now, one really has to wonder if Microsoft isn't effectively trying to kill Windows on ARM before it can gain any traction, on account of their complete disregard for all the things that effectively contributed to make Windows __the__ dominant OS on x86, such as [not getting into some kind of lock in with a specific vendor when it comes to platform development](https://www.qualcomm.com/news/releases/2021/05/24/qualcomm-enhances-support-developers-new-snapdragon-developer-kit-windows)), I am considering that there is more than enough demand to warrant an ARM64 version of Zadig/libwdi.

Thus, I am still committed to produce such a version. But, for the foreseeable future, I still have quite a few things that take precedence over that (and the planned release of Windows 11 just added a new major one to that list)...

mcuee, 2021-07-06:
Thanks for the detailed explanation. 

Yes it will be good if there are really cheap cheap yet powerful enough ARM64 platform that can run modern Windows. Apparently Raspberry Pi 4 is not yet powerful enough in that aspect. Qualcomm stuff is expensive. And they do not have enough native ARM apps (UWP is very much locked) so that they have to rely on x86/x64 emulation -- that will mean powerful ARM based solution from Qualcomm or similar -- not cheap stuff like Raspberry Pi.  So I do not see that happen any time soon.

Windows on ARM is pretty much locked, I believe it will be more so in the future with Windows 11.

Personally I run Linux on Raspberry Pi 400 and it is really pretty smooth (forget Youtube 1080p for a moment). And you have enough apps. Then Apple has done a great job for Apple Silicon Macs that they run both native ARM apps and x64 apps very well. I use homebrew and within a few month the ARM64 forumlas are mostly working for me already. The main missing thing is Virtual Box but I have alternatives like UTM to run ARM Linux if I want (I do not need now because of Raspberry Pi 400).

Anyway, it is just my personal opinion. Sorry for the rant here. And it is great that you are planning to support ARM on Windows with libwdi.

dougmsbbs, 2021-07-06:
There are A LOT of people out here that care about Windows on ARM. They're just getting discouraged that Microsoft got us this far, and then just seemed to lose interest in finishing it.

For a lot of people it's a matter of older software that there is no replacement for. I'm in that boat myself. But guess what, the old XP stuff works pretty darn well on a Pi 4 and Windows 10. You just need to give it the memory to run right, so get the 8GB version, and don't skimp on the SD card size . Booting Linux for the RTL stuff, then an emulator on top of that for the Windows program is not a good solution, but it's what a lot of people are trying to get by.

I have no doubt that as soon as an RTL-SDR driver that works on Windows 10 ARM64 on a Raspberry Pi 4 becomes available it'll have thousands of downloads in very short order. Maybe Microsoft is just used to thinking in terms of millions of new downloads a week, while I think in the thousand's, but still...

mcuee, 2021-07-07:
Good to know that there are "There are A LOT of people out here that care about Windows on ARM" out there that care about Windows on ARM. To me it seems the main interests are on Raspberry Pi 4 which somehow gets a license to run Windows on ARM. Raspberry Pi is an exception for Windows on ARM, not the norm.

If you read Microsoft document, it is all talking about Qualcomm stuff, which is more expensive than many Intel/AMD laptops.
https://docs.microsoft.com/en-us/windows/uwp/porting/apps-on-arm

Anyway, Pete is committed to support libwdi on ARM and that is good.

mcuee, 2021-07-07:
@jkunkee 
Re: https://daskunkee.blogspot.com/2020/02/tech-esp32-programming-from-arm64.html
>I believe Zadig relies on libusb's binary releases, so this needs to change upstream before Zadig Just Works(TM).

I do not think this is a real issue, remember Pete is the original developer of libusb Windows, so he knows the ins and outs of libusb Windows. So there will be no issue for him to host the ARM binary for libusb Windows.

Most importantly libwdi/Zadig do not depend on libusb-1.0.dll at all.

sidd-kishan, 2021-07-08:
here is the arm64 version of the lib usb dll

mcuee, 2021-07-08:
Based on https://github.com/a1ien/rusb/issues/68, vcpkg has already libusb-1.0 Windows binary. However, the issue here has nothing to do with that, as libwdi does not depend on libusb-1.0.dll binary at all.

mcuee, 2021-09-13:
I checked out some Youtube video, now it seems to be much easier to install Windows 10 or Windows 11 onto Raspberry Pi 4 and 400. So I kind of change a bit of my mind in this topic.

As for myself, I will not touch Windows on my Raspberry Pi 400 (Linux is good) and I have no intention to get an 8GB Raspberry Pi 4B. Maybe I will touch Windows 11 on next generation Raspberry Pi 5.

mcuee, 2021-09-13:
>Namely: if you install the latest Windows 10 WDK, you still don't get any of the WDF and WinUSB coinstaller DLLs for arm64 (Still no arm64 directory). And what's more, even for ARM, there's no winusbcoinstaller2.dll provided, which means we can't support that platform either.

This is still true with Windows 11 WDK. I just installed Windows 11 WDK.
https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk

mcuee, 2021-09-13:
I believe the way on ARM64 is probably different from ARM32/x86/AMD64.

pbatard, 2021-09-13:
As Jon mentioned:

> the CoInstallers are not needed on Windows 10 on ARM because the first version to ship was 1709. You can drop the references to them for ARM64.

So the way to approach WinUSB on ARM64 should be to use a `.inf` that's similar to the one that was posted [in this thread](https://answers.microsoft.com/en-us/windows/forum/windows_10-hardware-winpc/installing-winusb-driver-on-windows-10-iot/aea01587-37ea-4748-9c65-20843e9bdbf4), updated for ARM64.

Unfortunately, it's because the `.inf` for ARM64 is going to be much simpler than the `.inf` for x64 and win32 that it makes things more complex to automate, because [the `.inf.in` we use for WinUSB](https://github.com/pbatard/libwdi/blob/master/libwdi/winusb.inf.in) is tailored for systems where CoInstallers are needed, which means I'm going to have to add a different path and a different `.inf.in` for ARM64 installation.

However, this whole thing may be rendered completely moot by Windows 11, since Microsoft is no longer trusting certificates that are installed in _Trusted Publishers_ for the signing of driver packages.

Which means that Zadig cannot be used on regular Windows 11 to install WinUSB (fails with ` Driver package failed signature validation. Error = 0x800B0100`), and since it looks like the new way to get a custom driver package allowed for installation on a system is to go through a [Microsoft web portal](https://docs.microsoft.com/en-us/microsoft-store/device-guard-signing-portal), this may very well spell the end of Zadig/libwdi altogether.

As such, unless there's a way to make libwdi/Zadig relevant for Window 11, I'm not sure spending time adding ARM64 support is that great of an investment...

mcuee, 2021-09-14:
> However, this whole thing may be rendered completely moot by Windows 11, since Microsoft is no longer trusting certificates that are installed in _Trusted Publishers_ for the signing of driver packages, and that is what we have been relying on to be able to install drivers on Windows 7 up to Windows 10 without having to ask users to enable test signing.
> 
> Which means that Zadig cannot be used on regular Windows 11 to install WinUSB (fails with ` Driver package failed signature validation. Error = 0x800B0100`), and since it looks like the new way to get a custom driver package allowed for installation on a regular Windows 11 system is to go through a [Microsoft web portal](https://docs.microsoft.com/en-us/microsoft-store/device-guard-signing-portal), this may very well spell the end of Zadig/libwdi altogether.

Hmm, where do you get this info that "Microsoft is no longer trusting certificates that are installed in _Trusted Publishers_ for the signing of driver packages".

For Windows 10, Microsoft also says that Attestation Signing (you need an EV certificate to create an account to sign in the Microsoft Partner Center Developer Dashboard and then get the signed driver package) or HLK (aka WHQL) is needed but in reality Zadig just works.

mcuee, 2021-09-14:
BTW, Zadig becomes more important now that Cross-Signing is dead.
https://www.osr.com/blog/2020/10/15/microsoft-driver-updates-allowed-win7-win8/
>As of 1 July 2021, Microsoft is eliminating option #2, Cross-Signing. This will leave passing the HLK the only option for releasing drivers for Windows versions other than Windows 10.

pbatard, 2021-09-14:
> Hmm, where do you get this info that "Microsoft is no longer trusting certificates that are installed in Trusted Publishers for the signing of driver packages"?

Testing and user reports.

I validated that even if you have the cert of your driver package installed into Trusted Publishers on Windows 11, Windows declares that your driver package's signature chain ends up with a certificate that isn't trusted.

Ergo, for Windows 11, _ "Microsoft is no longer trusting certificates that are installed in Trusted Publishers for the signing of driver packages"_.

> As of 1 July 2021, Microsoft is eliminating option #2, Cross-Signing. This will leave passing the HLK the only option for releasing drivers for Windows versions other than Windows 10.

Yeah. And I'm having all the trouble in the world with this %^&*() requirement, because in parallel Microsoft have also managed to botch the web registration process that grants access to HLK, [which prevents eligible folks (that is, everyone that has an EV cert) from registering](https://www.microsoftpartnercommunity.com/t5/Feedback-Support-Discussions/Cannot-complete-Hardware-Developer-Program-registration-page/m-p/47832#M2021).

So I have currently been sitting on a rather expensive EV certificate, issued more than a month ago, that I can't use... But that's a separate issue from what we're discussing here, and it doesn't remove the fact that Microsoft now appears to have taken a hard stance against delegating any trust to third parties, and especially end users.

So much for using a trust mechanism that is designed to be transitive and where the end user should at least always have some modicum of control over what can and cannot be installed on their system (through a process that should be achievable entirely offline)...

mcuee, 2021-09-14:
Wow, that is bad. I do not have Windows 11 installed now but I got a laptop which can upgrade to Windows 11 once it is released. I hope things can change over time.

mcuee, 2021-09-14:
As for the HLK Portal issue, you may want to ask in OSR forum at https://community.osr.com/categories/ntdev. I feel it is more useful than Microsoft sites. Experts like Peter Viscarola  from OSR, Tim Roberts, and Microsoft's Doron Holan.

Ref: another issue with HLK which is different from yours.
 https://community.osr.com/discussion/293066/windows-hlk-signing-error

pbatard, 2021-09-14:
> I do not have Windows 11 installed now but I got a laptop which can upgrade to Windows 11 once it is released

If you want, and you have a decent enough USB 3.0 drive (where random I/O speeds are good, so preferably SSD class or SSD into USB enclosure) you can test Windows 11 by creating a Windows To Go drive with Rufus. This will run Windows off the USB altogether without touching the internal disks, and this is pretty much how I am testing Windows 11.

You can of course get the latest Windows 11 ISO from the Insider program.

Even as they relinquished slightly on TPM, I don't expect Microsoft to change their stance on no longer trusting "Trusted Publishers" for driver packages, do, yeah, some folks are going to be in for some disappointment when they find that they can no longer rely on an application to automate driver installation for them.

> As for the HLK Portal issue, you may want to ask in OSR forum at https://community.osr.com/categories/ntdev. I feel it is more useful than Microsoft sites. Experts like Peter Viscarola from OSR, Tim Roberts, and Microsoft's Doron Holan.

The problem is that this is a pure Microsoft issue (or most likely an issue between Microsoft and whatever subcontractor they used to design their hardware program registration web app), so, even if OSR includes Microsoft employees, I don't think they're going to be able to do much more than ask whoever is facing the problem but to create a support ticket (which we all did and are currently waiting on a slow resolution). And that's not to say I haven't also been exploring other ways to try to raise the visibility of this issue inside the Microsoft hierarchy, because there's really so much time I have for this utter bullshit of a registration process, when I need to have UEFI bootloaders Secure Boot signed for the release of Windows 11...

mcuee, 2021-09-14:
I will ask in the OSR forum about the point that "Microsoft is no longer trusting certificates that are installed in Trusted Publishers for the signing of driver packages". I think it is an important aspect which may affect quite some users using libusb related projects and potentially other projects as well.

pbatard, 2021-09-14:
Agreed. I think this is an important point to make there, especially as I still don't know exactly which of _driver isolation_ or _device guard_ or something else is responsible for the tightening of the rules, and I am also hopeful that the OSR experts may be able to point out to some kind of workaround to get driver packages installed, that doesn't involve having to seek Microsoft's formal approval...

mcuee, 2021-09-15:
Question asked in OSR forum:
https://community.osr.com/discussion/293115/windows-11-and-alternative-driver-installation-method-in-libwdi

mcuee, 2021-10-06:
I got a new WIndows 10 laptop and upgraded it to WIndows 11. Zadig seems to work fine without any issues. 

First test: I installed WinUSB driver for a generic HID device, no issues at all.

```
Zadig 2.6.755
Windows 10 64-bit (Build 22000)
ini file 'zadig.ini' not found in 'C:\work\libusb' - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
6 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8087&PID_0026&REV_0002
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_E0&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 22.80.0.4
libwdi:debug [wdi_create_list] BTHUSB USB device (0): USB\VID_8087&PID_0026\5&586B51A&0&10
libwdi:debug [wdi_create_list] Device description: 'Intel(R) Wireless Bluetooth(R)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1C7A&PID_0575&REV_1072
libwdi:debug [wdi_create_list] Compatible ID: USB\MS_COMP_WINUSB
libwdi:debug [wdi_create_list] Driver version: 3.7.1.1
libwdi:debug [wdi_create_list] WUDFRd USB device (2): USB\VID_1C7A&PID_0575\077E2F9A
libwdi:debug [wdi_create_list] Device description: 'EgisTec EH575'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_25A7&PID_FA23&REV_0241&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.22000.1
libwdi:debug [wdi_create_list] HidUsb USB device (3): USB\VID_25A7&PID_FA23&MI_00\6&2E1AA5FE&0&0000
libwdi:debug [wdi_create_list] Device description: '2.4G Receiver (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_25A7&PID_FA23&REV_0241&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.22000.1
libwdi:debug [wdi_create_list] HidUsb USB device (5): USB\VID_25A7&PID_FA23&MI_01\6&2E1AA5FE&0&0001
libwdi:debug [wdi_create_list] Device description: '2.4G Receiver (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0925&PID_7001&REV_0001
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22000.1
libwdi:debug [wdi_create_list] HidUsb USB device (7): USB\VID_0925&PID_7001\5&586B51A&0&1
libwdi:debug [wdi_create_list] Device description: 'Generic HID'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F2&PID_B6DD&REV_2710&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22000.176
libwdi:debug [wdi_create_list] usbvideo USB device (8): USB\VID_04F2&PID_B6DD&MI_00\6&2F8D90E&0&0000
libwdi:debug [wdi_create_list] Device description: 'HD User Facing (Interface 0)'
Checking for Zadig updates...

Checking release channel...

No new release version found.

Using inf name: Generic_HID.inf
Successfully extracted driver files.
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\xiaof\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\xiaof\usb_driver\Generic_HID.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\xiaof\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\xiaof\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'generic_hid.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\xiaof\usb_driver\generic_hid.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\xiaof\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\xiaof\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\xiaof\usb_driver\Generic_HID.cat'
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\VID_0925&PID_7001 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\VID_0925&PID_7001 (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\VID_0925&PID_7001 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\xiaof\usb_driver\Generic_HID.cat'
[... 259 lines cut ...]
libwdi:debug [wdi_create_list] Driver version: 3.7.1.1
libwdi:debug [wdi_create_list] WUDFRd USB device (2): USB\VID_1C7A&PID_0575\077E2F9A
libwdi:debug [wdi_create_list] Device description: 'EgisTec EH575'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_25A7&PID_FA23&REV_0241&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.22000.1
libwdi:debug [wdi_create_list] HidUsb USB device (3): USB\VID_25A7&PID_FA23&MI_00\6&2E1AA5FE&0&0000
libwdi:debug [wdi_create_list] Device description: '2.4G Receiver (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_25A7&PID_FA23&REV_0241&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
[... cut ...]
libwdi:debug [wdi_create_list] WinUSB USB device (7): USB\VID_0925&PID_7001\5&586B51A&0&1
libwdi:debug [wdi_create_list] Device description: 'Generic HID'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F2&PID_B6DD&REV_2710&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22000.176
libwdi:debug [wdi_create_list] usbvideo USB device (8): USB\VID_04F2&PID_B6DD&MI_00\6&2F8D90E&0&0000
libwdi:debug [wdi_create_list] Device description: 'HD User Facing (Interface 0)'

```

mcuee, 2021-10-06:
Second test: replace FT2232H (0403:6010) USB Composite Parent driver with WinUSB, no issues as well.

```
Zadig 2.6.755
Windows 10 64-bit (Build 22000)
ini file 'zadig.ini' not found in 'C:\work\libusb' - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
7 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8087&PID_0026&REV_0002
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_E0&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 22.80.0.4
libwdi:debug [wdi_create_list] BTHUSB USB device (1): USB\VID_8087&PID_0026\5&586B51A&0&10
libwdi:debug [wdi_create_list] Device description: 'Intel(R) Wireless Bluetooth(R)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1C7A&PID_0575&REV_1072
libwdi:debug [wdi_create_list] Compatible ID: USB\MS_COMP_WINUSB
libwdi:debug [wdi_create_list] Driver version: 3.7.1.1
libwdi:debug [wdi_create_list] WUDFRd USB device (3): USB\VID_1C7A&PID_0575\077E2F9A
libwdi:debug [wdi_create_list] Device description: 'EgisTec EH575'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_25A7&PID_FA23&REV_0241&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.22000.1
libwdi:debug [wdi_create_list] HidUsb USB device (4): USB\VID_25A7&PID_FA23&MI_00\6&2E1AA5FE&0&0000
libwdi:debug [wdi_create_list] Device description: '2.4G Receiver (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_25A7&PID_FA23&REV_0241&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.22000.1
libwdi:debug [wdi_create_list] HidUsb USB device (6): USB\VID_25A7&PID_FA23&MI_01\6&2E1AA5FE&0&0001
libwdi:debug [wdi_create_list] Device description: '2.4G Receiver (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6010&REV_0700&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 2.12.36.4
libwdi:debug [wdi_create_list] FTDIBUS USB device (8): USB\VID_0403&PID_6010&MI_01\6&C7EA758&0&0001
libwdi:debug [wdi_create_list] Device description: 'Lattice FTUSB Interface Cable (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6010&REV_0700&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 2.12.36.4
libwdi:debug [wdi_create_list] FTDIBUS USB device (9): USB\VID_0403&PID_6010&MI_00\6&C7EA758&0&0000
libwdi:debug [wdi_create_list] Device description: 'Lattice FTUSB Interface Cable (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F2&PID_B6DD&REV_2710&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22000.176
libwdi:debug [wdi_create_list] usbvideo USB device (10): USB\VID_04F2&PID_B6DD&MI_00\6&2F8D90E&0&0000
libwdi:debug [wdi_create_list] Device description: 'HD User Facing (Interface 0)'
10 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6010&REV_0700
libwdi:debug [wdi_create_list] Compatible ID: USB\DevClass_00&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22000.1
libwdi:debug [wdi_create_list] usbccgp USB device (0): USB\VID_0403&PID_6010\5&586B51A&0&1
libwdi:debug [wdi_create_list] Device description: 'Lattice FTUSB Interface Cable (Composite Parent)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8087&PID_0026&REV_0002
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_E0&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 22.80.0.4
libwdi:debug [wdi_create_list] BTHUSB USB device (1): USB\VID_8087&PID_0026\5&586B51A&0&10
libwdi:debug [wdi_create_list] Device description: 'Intel(R) Wireless Bluetooth(R)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F2&PID_B6DD&REV_2710
libwdi:debug [wdi_create_list] Compatible ID: USB\DevClass_00&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22000.1
libwdi:debug [wdi_create_list] usbccgp USB device (2): USB\VID_04F2&PID_B6DD\0001
libwdi:debug [wdi_create_list] Device description: 'HD User Facing (Composite Parent)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1C7A&PID_0575&REV_1072
libwdi:debug [wdi_create_list] Compatible ID: USB\MS_COMP_WINUSB
libwdi:debug [wdi_create_list] Driver version: 3.7.1.1
libwdi:debug [wdi_create_list] WUDFRd USB device (3): USB\VID_1C7A&PID_0575\077E2F9A
libwdi:debug [wdi_create_list] Device description: 'EgisTec EH575'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_25A7&PID_FA23&REV_0241&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.22000.1
libwdi:debug [wdi_create_list] HidUsb USB device (4): USB\VID_25A7&PID_FA23&MI_00\6&2E1AA5FE&0&0000
libwdi:debug [wdi_create_list] Device description: '2.4G Receiver (Interface 0)'
[... 299 lines cut ...]
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_25A7&PID_FA23&REV_0241
libwdi:debug [wdi_create_list] Compatible ID: USB\DevClass_00&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22000.1
libwdi:debug [wdi_create_list] usbccgp USB device (5): USB\VID_25A7&PID_FA23\5&586B51A&0&3
libwdi:debug [wdi_create_list] Device description: '2.4G Receiver (Composite Parent)'
[... cut ...]
libwdi:debug [wdi_create_list] Driver version: 10.0.22000.176
libwdi:debug [wdi_create_list] usbvideo USB device (8): USB\VID_04F2&PID_B6DD&MI_00\6&2F8D90E&0&0000
libwdi:debug [wdi_create_list] Device description: 'HD User Facing (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\ROOT_HUB30&VID8086&PID9A13&REV0001
libwdi:debug [wdi_create_list] Compatible ID: 
libwdi:debug [wdi_create_list] Driver version: 10.0.22000.65
libwdi:debug [wdi_create_list] USBHUB3 USB device (9): USB\ROOT_HUB30\4&1A63DC7C&0&0

```

pbatard, 2021-10-06:
Yup, I am confirming the same results with the official release.

It looks like Microsoft may have been using the Insider builds to test features, that they didn't carry into the release, though I'm a bit worried they might not try to bring them up again in a future feature release.

WIGASTY, 2021-11-01:
**Driver Installation:FAILED on Windows 11**

Using inf name: SysDVR_-_Video_(Interface_0).inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\wigas\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\wigas\usb_driver\SysDVR_-_Video_(Interface_0).inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:info [wdi_prepare_driver] Test signing is: Disabled
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\wigas\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\wigas\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'sysdvr_-_video_(interface_0).inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\wigas\usb_driver\sysdvr_-_video_(interface_0).inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\wigas\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\wigas\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\wigas\usb_driver\SysDVR_-_Video_(Interface_0).cat'
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_057E&PID_3006&MI_00 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_057E&PID_3006&MI_00 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\VID_057E&PID_3006&MI_00 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\VID_057E&PID_3006&MI_00 (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\VID_057E&PID_3006&MI_00 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\wigas\usb_driver\SysDVR_-_Video_(Interface_0).cat'
libwdi:info [SelfSignFile] Successfully deleted private key
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter SysDVR_-_Video_(Interface_0).inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_057E&PID_3006&MI_00\6&1696B214&0&0000'
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_057E&PID_3006&REV_0100&MI_00'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-2656495222-189657690-2984995851-1001'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] Switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_057E&PID_3006&REV_0100&MI_00 - please wait...
libwdi:debug [syslog]  but terminated in a root certificate which is not trusted by the trust provider.
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 00:14:54.654
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE} 00:14:54.654
libwdi:debug [syslog]      sig:                          Key      = WinUSB_Generic_Device.inf
libwdi:debug [syslog]      sig:                          FilePath = C:\WINDOWS\System32\DriverStore\Temp\{03af70c6-abd4-c249-8c50-8cb95fc0b163}\WinUSB_Generic_Device.inf
libwdi:debug [syslog]      sig:                          Catalog  = C:\WINDOWS\System32\DriverStore\Temp\{03af70c6-abd4-c249-8c50-8cb95fc0b163}\WinUSB_Generic_Device.cat
libwdi:debug [syslog] !    sig:                          Verifying file against specific Authenticode(tm) catalog failed.
libwdi:debug [syslog] !    sig:                          Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 00:14:54.670
libwdi:debug [syslog] !!!  sig:                     Driver package catalog file certificate does not belong to Trusted Root Certificates, and Code Integrity is enforced.
libwdi:debug [syslog] !!!  sig:                     Driver package failed signature validation. Error = 0x800B0109
libwdi:debug [syslog]      sto:                {DRIVERSTORE IMPORT VALIDATE: exit(0x800b0109)} 00:14:54.670
libwdi:debug [syslog] !!!  sig:                Driver package failed signature verification. Error = 0x800B0109
libwdi:debug [syslog] !!!  sto:                Failed to import driver package into Driver Store. Error = 0x800B0109
libwdi:debug [syslog]      sto:           {Stage Driver Package: exit(0x800b0109)} 00:14:54.670
libwdi:debug [syslog]      dvs:      {Driver Setup Import Driver Package - exit (0x800b0109)} 00:14:54.670
[... cut ...]
libwdi:debug [wdi_create_list] RTSUER USB device (8): USB\VID_0BDA&PID_0129\20100201396000000
libwdi:debug [wdi_create_list] Device description: 'USB2.0-CRW'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F2&PID_B52B&REV_9552&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.22478.1000
libwdi:debug [wdi_create_list] usbvideo USB device (9): USB\VID_04F2&PID_B52B&MI_00\6&2CA746EB&0&0000
libwdi:debug [wdi_create_list] Device description: 'USB2.0 VGA UVC WebCam (Interface 0)'

pbatard, 2021-11-01:
@WIGASTY, either you're trying to use libwdi/Zadig on ARM64, which is not supported, or you are trying to piggyback on an unrelated issue (and also you have truncated the log so we can't even tell what version of Zadig and the exact version of Windows you are using, including the build number).

At any rate, I have deleted your comment, because it is out of scope.

radiolee, 2021-11-10:
https://github.com/edk2-porting/edk2-sdm845
I have a Xiaomi-Mi8(dipper) old 845 device and run Win11 arm64 with no issues. I want use usb2ttl ( ch341, pl2302, ft232, etc..) to connect mcu, and I also play rtl-sdr. These old cell phone really cheap and maybe you have one or more :-)

mcuee, 2021-11-17:
Ref: 
https://github.com/mcuee/libusb-win32/issues/29
https://github.com/mcuee/libusbk/issues/35

Dreamail, 2022-08-27:
Are there any futher progress?

ArminiusTux, 2022-12-29:
Dear @Dreamail,

I am also eagerly awaiting a Windows ARM64 build of Zadig.

Imagine the pure irony of this Microsoft device being the first useable TV broadcasting solution for Windows 11 ARM, with the help of [3rd party drivers](https://zadig.akeo.ie/) by @pbatard and a [customised interface & DVB application](https://www.altx.ro/projects/dvb-usb-direct-interface/) by @altxro . ;-)

> ; Xbox_USB_Tuner.inf
> ; Copyright (c) 2010-2016 Pete Batard <pete@akeo.ie> (GNU LGPL)
> [Strings]
> DeviceName = "Xbox USB Tuner"
> VendorName = "Microsoft Corp."
> SourceName = "Xbox USB Tuner Install Disk"
> DeviceID   = "VID_045E&PID_02D5"
> DeviceGUID = "{FC07B70E-28E7-4564-924C-EEF3D8C21DD8}"
> 
> [Version]
> Signature   = "$Windows NT$"
> Class       = "USBDevice"
> ClassGuid   = {88bae032-5a81-49f0-bc3d-a4ff138216d6}
> Provider    = "libwdi"
> CatalogFile = Xbox_USB_Tuner.cat
> DriverVer   = 06/02/2012, 6.1.7600.16385
> 
> [ClassInstall32]
> Addreg = WinUSBDeviceClassReg
> 
> [WinUSBDeviceClassReg]
> HKR,,,0,"Universal Serial Bus devices"
> HKR,,Icon,,-20
> 
> [Manufacturer]
> %VendorName% = libusbDevice_WinUSB,NTx86,NTamd64,NTarm

mcuee, 2023-01-20:
@Dreamail and @ArminiusTux 

Sorry but just wondering if you can help test the following PR for libusb project.
* https://github.com/libusb/libusb/pull/1202

Dreamail, 2023-01-20:
> @Dreamail and @ArminiusTux 
> 
> Sorry but just wondering if you can help test the following PR for libusb project.
> * https://github.com/libusb/libusb/pull/1202

Any prebulit binary?.

mcuee, 2023-01-20:
> Any prebulit binary?.

Yes, here it is for WIndows on ARM64.
https://github.com/libusb/libusb/pull/1202#issuecomment-1398094170

ectoplasmosis, 2023-01-25:
Any update on this? We have many devices relying on x86 libusb32 drivers which we would love to use with Win11 on ARM64.

mcuee, 2023-02-06:
Since this will take some time, the current work-around is to modify the FW to use WCID (Microsoft OS Descriptors for USB Devices).
* https://github.com/pbatard/libwdi/wiki/WCID-Devices
* https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/microsoft-defined-usb-descriptors

Success story:
https://github.com/LibUsbDotNet/LibUsbDotNet/issues/138#issuecomment-1418688319

Dreamail, 2023-02-07:
So it is impossible to use libusb on arm device without modify firmware?

mcuee, 2023-02-07:
> So it is impossible to use libusb on arm device without modify firmware?

If your device is recognized by Windows on ARM/ARM64, and use WinUSB driver, it should work without modifying the firmware.

If your device is recognized by Windows on ARM/ARM64, and use USB HID driver, it should work without modifying the firmware.

mcuee, 2023-02-07:
https://github.com/libusb/libusb/pull/1202#issuecomment-1419953889

> TV tuner (DVB)
> Drivers created by Zadig and modified (ARM64) using solely the winusb.sys (v10.0.22621.608)

@ArminiusTux 

Just wondering how you get the driver installed for your DVB TV tuner? Are you saying that you are able to install the driver without signing the driver package?

mcuee, 2023-02-07:
@Dreamail and @ArminiusTux 
Try this tip as well to see if it works for you or not.
https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb-installation#installing-winusb-by-specifying--the-system-provided-device-class

**Installing WinUSB by specifying the system-provided device class**
When you connect your device, you might notice that Windows loads Winusb.sys automatically (if the IHV has defined the device as a WinUSB Device). Otherwise follow these instructions to load the driver:

Plug in your device to the host system.
Open Device Manager and locate the device.
Select and hold (or right-click) the device and select Update driver software... from the context menu.
In the wizard, select Browse my computer for driver software.
Select Let me pick from a list of device drivers on my computer.
From the list of device classes, select Universal Serial Bus devices.
The wizard displays WinUsb Device. Select it to load the driver.

pbatard, 2023-02-07:
> So it is impossible to use libusb on arm device without modify firmware?

It's not impossible. You just have to craft the `.inf` (and sign the resulting driver package) yourself, which is by all means something that's definitely feasible.

Please remember that **all** Zadig/libwdi does is automate an operation (creating an installing a `.inf`) that, if you know what you're doing &mdash; and there should be plenty of resources to guide you there, you should be able to accomplish on your own.

Zadig/libwdi only started as a _"People won't want to have to edit a `.inf` manually, so let's see if we can automate that for them"_ project but, underneath it all, it just replicates what's described by Microsoft at https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb-installation#writing-a-custom-inf-for-winusb-installation (same article as the one Xiaofan pointed above).

Sadly, Microsoft did not really update that page for ARM64, but, on principle, you should be able to craft the `.inf` you need and install the driver package yourself. So, no, it is not impossible to use libusb on ARM without modify firmware. However, you may have to work for it...

As a matter of fact, outside of being a bit too busy with other stuff (and having had to relegate Zadig/libwdi as a **low priority** project), I'm kind of waiting for people to actually start writing and publishing their own ARM64 `.inf`'s, so that I can then have a few good examples of how one should go around automating the whole thing.

mcuee, 2023-02-07:
This is probably a bit too complicated as an WinUSB inf example, but I think it can be used as a reference.

```
; WCHLinkWDM.INF
; Driver for WCHLink V1.2
; WDM/DLL for Windows XP/Vista/7/8/8.1/10/11/ SERVER 2003/2008/2012/2016/2019/2022
; Copyright (C) WCH 2022
;

[Version]
Signature = "$Chicago$"
Class     = WCH
ClassGuid = {77989ADF-06DB-4025-92E8-40D902C03B0A}
Provider  = %WinChipHead%
DriverVer = 08/23/2022, 1.2.2022.08
CatalogFile = WCHLinkWDM.CAT

[ControlFlags]
ExcludeFromSelect = USB\VID_1A86&PID_8010&MI_00,USB\VID_1A86&PID_8012&MI_00

[ClassInstall32]
Addreg = WCHReg

[ClassInstall]
Addreg = WCHReg

[WCHReg]
HKR, , , 0, %ClassName%
HKR, , Icon, , -5
HKR, , NoInstallClass, , 1

[Manufacturer]
%WinChipHead% = WinChipHead,NT,NTamd64,NTia64,NTarm64

[WinChipHead]
%WCHLink.DeviceDesc% = WCHLink.Install, USB\VID_1A86&PID_8010&MI_00
%WCHLinkDAP.DevDesc% = USB_Install, USB\VID_1A86&PID_8012&MI_00

[WinChipHead.NT]
%WCHLink.DeviceDesc% = WCHLink.Install, USB\VID_1A86&PID_8010&MI_00
%WCHLinkDAP.DevDesc% = USB_Install, USB\VID_1A86&PID_8012&MI_00

[WinChipHead.NTamd64]
%WCHLink.DeviceDesc% = WCHLink.Install.NTamd64, USB\VID_1A86&PID_8010&MI_00
%WCHLinkDAP.DevDesc% = USB_Install, USB\VID_1A86&PID_8012&MI_00

[WinChipHead.NTia64]
%WCHLink.DeviceDesc% = WCHLink.Install.NTia64, USB\VID_1A86&PID_8010&MI_00
%WCHLinkDAP.DevDesc% = USB_Install, USB\VID_1A86&PID_8012&MI_00

[WinChipHead.NTarm64]
%WCHLink.DeviceDesc% = WCHLink.Install.NTarm64, USB\VID_1A86&PID_8010&MI_00
%WCHLinkDAP.DevDesc% = USB_Install, USB\VID_1A86&PID_8012&MI_00

[USB_Install]
Include=winusb.inf
Needs=WINUSB.NT

[USB_Install.Services]
Include = Winusb.inf
AddService = WinUSB,0x00000002,WinUSB_ServiceInstall

[USB_Install.HW]
AddReg = USB_Install.HW.AddReg

[USB_Install.HW.AddReg]
HKR,,DeviceInterfaceGUIDs,0x10000,"{CDB3B5AD-293B-4663-AA36-1AAE46463776}"

[WinUSB_ServiceInstall]
DisplayName     = %USB_Install_SvcDesc%
ServiceType     = 1
StartType       = 3
ErrorControl    = 1
ServiceBinary   = %12%\WinUSB.sys

[WCHLink.Install]
CopyFiles = WCHLink.CopyFiles.SYS, WCHLink.CopyFiles.DLL
AddReg    = WCHLink.9X.AddReg, WCHLink.AddReg

[WCHLink.Install.NT]
CopyFiles = WCHLink.CopyFiles.SYS, WCHLink.CopyFiles.DLL
AddReg    = WCHLink.NT.AddReg, WCHLink.AddReg

[WCHLink.Install.NTamd64]
CopyFiles = WCHLink.CopyFiles.SYSA64, WCHLink.CopyFiles.DLLA64
AddReg    = WCHLink.NT.AddReg, WCHLink.AddReg

[WCHLink.Install.NTia64]
CopyFiles = WCHLink.CopyFiles.SYSI64, WCHLink.CopyFiles.DLLI64
AddReg    = WCHLink.NT.AddReg, WCHLink.AddReg

[WCHLink.Install.NTarm64]
CopyFiles = WCHLink.CopyFiles.SYSM64, WCHLink.CopyFiles.DLLM64
AddReg    = WCHLink.NT.AddReg, WCHLink.AddReg

[WCHLink.CopyFiles.SYS]
WCHLinkWDM.SYS, , , 2

[WCHLink.CopyFiles.SYSA64]
WCHLinkW64.SYS, , , 2

[WCHLink.CopyFiles.SYSI64]
;WCHLinkI64.SYS, , , 2

[WCHLink.CopyFiles.SYSM64]
WCHLinkM64.SYS, , , 2

[WCHLink.CopyFiles.DLL]
WCHLinkDLL.DLL, , , 2

[WCHLink.CopyFiles.DLLA64]
WCHLinkDLL.DLL, , , 2

[WCHLink.CopyFiles.DLLI64]
WCHLinkDLL.DLL, , , 2

[WCHLink.CopyFiles.DLLM64]
WCHLinkDLL.DLL, , , 2

[WCHLink.9X.AddReg]
HKR, , DevLoader, , *NTKERN
HKR, , NTMPDriver, , WCHLinkWDM.SYS

[WCHLink.NT.AddReg]

[WCHLink.AddReg]
HKLM, SOFTWARE\WinChipHead\IC\WCHLink, WDM, 0x00010001, 0x00000025
HKLM, SOFTWARE\WinChipHead\IC\WCHLink, DLL, 0x00010001, 0x00000022
HKLM, SOFTWARE\WinChipHead\IC\WCHLink, Function, , "USB slave"

[WCHLink.Install.NT.Services]
AddService = WCHLink, 2, WCHLink.Service

[WCHLink.Install.NTamd64.Services]
AddService = WCHLink_A64, 2, WCHLink.ServiceA64

[WCHLink.Install.NTia64.Services]
AddService = WCHLink_I64, 2, WCHLink.ServiceI64

[WCHLink.Install.NTarm64.Services]
AddService = WCHLink_M64, 2, WCHLink.ServiceM64

[WCHLink.Service]
DisplayName   = "WCHLinkWDM"
ServiceType   = 1
StartType     = 3
ErrorControl  = 1
ServiceBinary = %10%\System32\Drivers\WCHLinkWDM.SYS

[WCHLink.ServiceA64]
DisplayName   = "WCHLinkWDM_A64"
ServiceType   = 1
StartType     = 3
ErrorControl  = 1
ServiceBinary = %10%\System32\Drivers\WCHLinkW64.SYS

[WCHLink.ServiceI64]
DisplayName   = "WCHLinkWDM_I64"
ServiceType   = 1
StartType     = 3
ErrorControl  = 1
ServiceBinary = %10%\System32\Drivers\WCHLinkI64.SYS

[WCHLink.ServiceM64]
DisplayName   = "WCHLinkWDM_M64"
ServiceType   = 1
StartType     = 3
ErrorControl  = 1
ServiceBinary = %10%\System32\Drivers\WCHLinkM64.SYS

[DestinationDirs]
DefaultDestDir      = 10, System32\Drivers
WCHLink.CopyFiles.DLL = 11
WCHLink.CopyFiles.DLLA64 = 10, Syswow64 
WCHLink.CopyFiles.DLLI64 = 10, Syswow64 
WCHLink.CopyFiles.DLLM64 = 10, Syswow64 
WCHLink.CopyFiles.SYS = 10, System32\Drivers
WCHLink.CopyFiles.SYSA64 = 10, System32\Drivers
WCHLink.CopyFiles.SYSM64 = 10, System32\Drivers
;WCHLink.CopyFiles.SYSI64 = 10, System32\Drivers

[SourceDisksFiles]
WCHLinkWDM.SYS  = 1
WCHLinkDLL.DLL  = 1
WCHLinkW64.SYS  = 1
WCHLinkM64.SYS  = 1

[SourceDisksFiles.amd64]
WCHLinkW64.SYS  = 1
WCHLinkDLL.DLL  = 1

[SourceDisksFiles.arm64]
WCHLinkM64.SYS  = 1
WCHLinkDLL.DLL  = 1

[SourceDisksNames]
1 = %DISK_NAME%, , ,

[SourceDisksNames.amd64]
1 = %DISK_NAME%, , ,

[SourceDisksNames.ia64]
;1 = %DISK_NAME%, , ,

[SourceDisksNames.arm64]
1 = %DISK_NAME%, , ,

[Strings]
ClassName        = "Interface"
WinChipHead      = "wch.cn"
USB_Install_SvcDesc = "WinUSB"
WCHLink.DeviceDesc  = "WCH-LinkRV"
WCHLinkDAP.DevDesc  = "WCH CMSIS-DAP"
DISK_NAME = "WCH-Link Installation Disk"
```

ArminiusTux, 2023-02-07:
> Just wondering how you get the driver installed for your DVB TV tuner? Are you saying that you are able to install the driver without signing the driver package?

Indeed @mcuee , I bent the rules and cheated. 😉
[image]

> Zadig/libwdi only started as a _"People won't want to have to edit a `.inf` manually, so let's see if we can automate that for them"_ project but, underneath it all, it just replicates what's described by Microsoft

We (i.e. "the common & lazy people) still need such a tool, let it be for monitor drivers (just .inf tweaking) only. 👌

> I'm kind of waiting for people to actually start writing and publishing their own ARM64 `.inf`'s, so that I can then have a few good examples of how one should go around automating the whole thing.

Hopefully @pbatard, this genuine ARM64 driver from Samsung for the **Galaxy Book 2 (SM-W737)** - [posted here](https://github.com/ljhholer/Samsung-Galaxy-Book-2-Hardware-Drivers) by @ljhholer, is worth a closer look.🕵️‍♂️

<details>
  <summary>WinUSBSKY.inf</summary>
  
for the "[Fx2 Learning Kit Device](https://osrfx2.sourceforge.net/)"
  
  ```javascript
  ;
;
; Installs WinUsb
;

[Version]
Signature = "$Windows NT$"
Class     = USBDevice
ClassGUID = {88BAE032-5A81-49f0-BC3D-A4FF138216D6}
Provider  = %ManufacturerName%
CatalogFile = WinUSBInstallation.cat
DriverVer = 06/24/2018,1.0.0.1

; ========== Manufacturer/Models sections ===========

[Manufacturer]
%ManufacturerName% = Standard,NTARM64

[Standard.NTARM64]
;%DeviceName% =USB_Install, USB\VID_0547&PID_1002
%DeviceName% =USB_Install, USB\VID_0483&PID_DF11

; ========== Class definition ===========

;[ClassInstall32]
;AddReg = ClassInstall_AddReg

;[ClassInstall_AddReg]
;HKR,,,,%ClassName%
;HKR,,NoInstallClass,,1
;HKR,,IconPath,%REG_MULTI_SZ%,"%systemroot%\system32\setupapi.dll,-20"
;HKR,,LowerLogoVersion,,5.2

; =================== Installation ===================

[USB_Install]
Include = winusb.inf
Needs   = WINUSB.NT

[USB_Install.Services]
Include =winusb.inf
Needs   = WINUSB.NT.Services

[USB_Install.HW]
AddReg=Dev_AddReg

[Dev_AddReg]
;HKR,,DeviceInterfaceGUIDs,0x10000,"{9f543223-cede-4fa3-b376-a25ce9a30e74}"
HKR,,DeviceInterfaceGUIDs,0x10000,"{5CBC7CBE-C056-45BD-B422-0049D94D11E9}"

; [DestinationDirs]
; If your INF needs to copy files, you must not use the DefaultDestDir directive here.  
; You must explicitly reference all file-list-section names in this section.

; =================== Strings ===================

[Strings]
ManufacturerName="Samsung Electronics Co.,Ltd."
ClassName="Universal Serial Bus devices"
;DeviceName="Fx2 Learning Kit Device"
DeviceName="Samsung WinUSB Device"
REG_MULTI_SZ = 0x0001
;
  ```
  
</details>

mcuee, 2023-02-08:
> > Just wondering how you get the driver installed for your DVB TV tuner? Are you saying that you are able to install the driver without signing the driver package?
> 
> Indeed @mcuee , I bent the rules and cheated. 😉 [image]

Thanks a lot for the info. Now it is clear to me.

mcuee, 2023-02-08:
> To avoid the above, an alternative to Zadig is to use dpscat utility from libusbK project which can be used to sign any driver package. But I have not checked how easy to build dpscat under Windows on ARM64. https://github.com/mcuee/libusbk/tree/master/libusbK/src/dpscat

@ArminiusTux 
Here is the libusbK binary for Windows on ARM64 (built using VS2019), including dpscat. Please give it a try. Thanks.
[libusbK_bin_arm64.zip](https://github.com/mcuee/libusbk/files/10680835/libusbK_bin_arm64.zip)

pbatard, 2023-02-08:
Gonna leave [this one] (https://github.com/libusb/libusb/pull/1202#issuecomment-1421515015) here as well, since it shows the changes that can be applied to a libwdi generated `.inf` to apply it to ARM64.

ArminiusTux, 2023-02-11:
> Here is the libusbK binary for Windows on ARM64 (built using VS2019), including dpscat. Please give it a try.

<details>
  <summary>Well @mcuee , the signing tool apparently does not like my butchered inf.</summary>

[image]

[image]
</details>

<details>
  <summary>No problems (re)signing the original though:</summary>
  
[image]
</details>

[29 more replies]

closed 2023-03-01 by pbatard