pbatard/libwdi #157: runas when already admin

doub, 2019-05-13:
I was browsing the code when I noticed something that doesn't look right to me in install_driver_internal at the following line: https://github.com/pbatard/libwdi/blob/bc2336e/libwdi/libwdi.c#L1561

It seems the privilege elevation is run when we're already admin, and presumably not run (in the else clause) when it would be needed. I think the bug (in libwdi) might be hidden (in the samples) by the fact that the executables are compiled with /MANIFESTUAC:"level='requireAdministrator'".

Or am I misunderstanding this code?

pbatard, 2019-05-13:
> It seems the privilege elevation is run when we're already admin

Remember, libwdi is first and foremost a __library__ project. When used as a library, it runs at the privilege level of the calling application, and since this is a generic library, we can make no assumption that the calling application is already running elevated.

The fact that the __samples__ that are associated with the library are compiled to run elevated is irrelevant and should not be extrapolated to mean that everybody who uses the library will also call it elevated.

And that is the reason why we do need code that will attempt to run the library elevated if the process that called the library isn't.

As I believe I answered your question, I will close this issue.

closed 2019-05-13 by pbatard

doub, 2019-05-13:
I understand the need to elevate the privileges. I'm trying to use the library from a non-admin process.

What I was saying is that the if condition is inverted, and privilege elevation is only attempted when it's not needed, and not attempted when it's needed. I actually went a little bit further and confirmed that with the code in the repo, if the app is not marked as "requireAdministrator", then IsUserAnAdmin() returns false, which runs the else clause, which calls CreateProcess (and fails) instead of ShellExecute+runas.

pbatard, 2019-05-13:
On what platforms have you tested? IIRC correctly, IsUserAnAdmin() was designed for pre-Windows 7 environments in mind (because, even on platforms that don't have elevation, we need to find out whether the user has the correct rights).

Unless you have validated that IsUserAnAdmin() does return false for all platforms from XP to Windows 10, I don't believe that there's anything that needs changing.

doub, 2019-05-13:
I tested it on Windows 10. And [the documentation of IsUserAnAdmin](https://docs.microsoft.com/en-us/windows/desktop/api/shlobj_core/nf-shlobj_core-isuseranadmin) also says that it returns false when you're not an admin.

pbatard, 2019-05-13:
Again, you need to test on more than Windows 10. I am pretty sure this was designed around Windows XP or Windows 7 elevation detection, and, especially if you can't point to an actual issue with keeping the current code, I see no reason to alter or remove it.

Please bear in mind that the library needs to handle more than a single type of platforms and that things that you might think don't make much sense for a platform are very much required for another.

doub, 2019-05-13:
I can't help but feel like you've misread what I wrote. Which one is elevating privileges? Is that ShellExecute or CreateProcess?

pbatard, 2019-05-13:
Do you see an actual issue or not? Unless you see an actual issue with the code, I'm not planning to invest time looking closely into it, as this automatically becomes a low priority item. Maybe you feel like I should drop whatever else I am doing to look more closely into this, but I'm not willing to do that unless you can demonstrate an actual issue. I'm simply too busy right now to deal with non issues.

doub, 2019-05-13:
Yes, I see an issue. But your mind is set that it's a non-issue. Let's agree to disagree and not waste any more of your time, I'll adjust my local copy. Cheers.

pbatard, 2019-05-13:
No worries. If you want to send a pull request, that you have tested at least on Windows 7 and Windows 10, I'll look at it, but otherwise, I'm afraid my current priorities don't provide much incentive to look at fixing libwdi unless there is a clear issue likely to affect a lot of people (and I'm afraid your initial _"code doesn't look right"_ doesn't provide me with much inclination to look back at code that, again, I'm pretty sure was added for older platforms and that nobody seems to have had any issue with until now).