pbatard/libwdi #158: Window 7 Driver Installation: FAILED (Requested resource not found)

xdimh, 2019-05-13:
Zadig 2.4.721
Windows 7 SP1 64-bit
ini file 'zadig.ini' not found - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
5 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_093A&PID_2510&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.24145
libwdi:debug [wdi_create_list] HidUsb USB device (2): USB\VID_093A&PID_2510\6&106A4439&0&5
libwdi:debug [wdi_create_list] Device description: 'USB Optical Mouse'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1C4F&PID_0026&REV_0110&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.24145
libwdi:debug [wdi_create_list] HidUsb USB device (4): USB\VID_1C4F&PID_0026&MI_00\7&2B6CCC1E&0&0000
libwdi:debug [wdi_create_list] Device description: 'USB Keyboard (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1C4F&PID_0026&REV_0110&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.24145
libwdi:debug [wdi_create_list] HidUsb USB device (5): USB\VID_1C4F&PID_0026&MI_01\7&2B6CCC1E&0&0001
libwdi:debug [wdi_create_list] Device description: 'USB Keyboard (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_6868&PID_0200&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_07&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 1.2.6.0
libwdi:debug [wdi_create_list] libusb0 USB device (6): USB\VID_6868&PID_0200\GP-58
libwdi:debug [wdi_create_list] Device description: 'GP-58'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8888&PID_902E&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.24145
libwdi:debug [wdi_create_list] HidUsb USB device (9): USB\VID_8888&PID_902E\2011
libwdi:debug [wdi_create_list] Device description: 'Receiver'
Checking for Zadig updates...

Checking release channel...

Updates: Unable to acces version data

Using inf name: GP-58.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\Administrator\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\Administrator\usb_driver\GP-58.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Administrator\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Administrator\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'gp-58.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Administrator\usb_driver\gp-58.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Administrator\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Administrator\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] successfully created file 'C:\Users\Administrator\usb_driver\GP-58.cat'
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_6868&PID_0200 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_6868&PID_0200 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] created new key container
libwdi:debug [CreateSelfSignedCert] generated new keypair
libwdi:info [CreateSelfSignedCert] created new self-signed certificate 'CN=USB\VID_6868&PID_0200 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] successfully created certificate 'CN=USB\VID_6868&PID_0200 (libwdi autogenerated)'
libwdi:info [SelfSignFile] added certificate 'CN=USB\VID_6868&PID_0200 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] successfully signed file 'C:\Users\Administrator\usb_driver\GP-58.cat'
libwdi:info [SelfSignFile] successfully deleted private key
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter GP-58.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_6868&PID_0200\GP-58'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_6868&PID_0200&REV_0100'
[... 442 lines cut ...]
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_093A&PID_2510&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.24145
libwdi:debug [wdi_create_list] HidUsb USB device (2): USB\VID_093A&PID_2510\6&106A4439&0&5
libwdi:debug [wdi_create_list] Device description: 'USB Optical Mouse'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1C4F&PID_0026&REV_0110&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.24145
libwdi:debug [wdi_create_list] HidUsb USB device (4): USB\VID_1C4F&PID_0026&MI_00\7&2B6CCC1E&0&0000
libwdi:debug [wdi_create_list] Device description: 'USB Keyboard (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1C4F&PID_0026&REV_0110&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.24145
libwdi:debug [wdi_create_list] HidUsb USB device (5): USB\VID_1C4F&PID_0026&MI_01\7&2B6CCC1E&0&0001
libwdi:debug [wdi_create_list] Device description: 'USB Keyboard (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_6868&PID_0200&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_07&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 1.2.6.0
libwdi:debug [wdi_create_list] libusb0 USB device (6): USB\VID_6868&PID_0200\GP-58
libwdi:debug [wdi_create_list] Device description: 'GP-58'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8888&PID_902E&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.24145
libwdi:debug [wdi_create_list] HidUsb USB device (9): USB\VID_8888&PID_902E\2011
libwdi:debug [wdi_create_list] Device description: 'Receiver'

pbatard, 2019-05-13:
I think this may be an issue with your Windows 7 installation.

The core of your issue is this:

```
Unable to load INF: 'C:\Windows\System32\DriverStore\FileRepository\winusb.inf_amd64_neutral_2878b0df142d775e\winusb.inf'(00000003)
Error 3: The system cannot find the path specified.
Could not find include INF file "winusb.inf". Error = 0x00000003
```

Which indicates that Windows thinks that your system should be able to provide a generic x64 WinUSB driver, yet these driver installation files (this part comes from the system, and not libwdi) cannot be found where they should be located.

All in all, considering that other Win7 users are not reporting a similar issue, it certainly looks like an environmental problem with your machine. Have you applied all the Windows updates? Have you perhaps applied Group Policies that relate to drivers? And does there exist any `winusb.inf` under `C:\Windows\System32\DriverStore\FileRepository\`?

xdimh, 2019-05-15:
@pbatard Thank you for your reply！problem is solved when I put the winusb.inf in the correct position.

pbatard, 2019-05-15:
Okay. I will close this issue then.

closed 2019-05-15 by pbatard