pbatard/libwdi #161: Feature Request - Multiple versions of WinUSB in Zadig

Fliptron, 2019-07-04:
I'm not sure if my request makes sense, but here are the behaviors I've seen.

All of this is on a Windows 7 Pro x64 with SP1.  CPU is AMD FX8370

USB Hardware:  8 channel logic analyzer (clone of Saleae's product)
                         Just released Joulescope  (high speed current and voltage measurement)

If I plug in the logic analyzer, and there is no WinUSB yet associated with that USB
port, Windows driver install fetches a WinUSB from Windows Update Server.
The version installed is V6.0.6000.16388 and it shows up in device manager as Saleae.
The Saleae software then recognizes the logic analyzer, and the analyzer can be used.
If that driver is already installed, then no additional driver install activity occurs, and
the Saleae software works fine.

An open source alternative to the Saleae software is sigrok/Pulseview.  It does not see the logic analyzer if the associated WinUSB is V6.0.6000.16388 .  They warn of this issue, and recommend using Zadig  [](https://sigrok.org/wiki/Windows) to update the driver. If I use Zadig to update the driver to Winusb, the version installed is V6.1.7600.16385  .  For reasons I still have not figured out, after I have done this, sigrok/Pulseview still can't see the hardware, and now the Saleae software can't see it either.

To get back to the version of WinUSB that works with the Saleae software, I can't do it with Zadig. Windows device manager may show a roll-back option, but on many attempts it does not.
The only consistent way I have found to get back to the older version is to use device manager and select uninstall and check the "delete the driver" option. Then if I unplug and re-plug the logic analyzer, the Windows based driver install will then retrieve V6.0.6000.16388

Separately from the above, I have a Joulescope , and it too needs WinUSB, but it needs version
V6.1.7601.23403.  I think it may also require that .inf file associates the product name with VID/PID .  (at this point I am way out of my depth as to what is going on, I really don't have a good understanding of how USB internals work).  (and I think the Saleae may also need this name association)

So my feature request (if it makes sense) is to support multiple versions of WinUSB, and maybe support setting the name (if it makes sense).
Thanks

pbatard, 2019-07-05:
Not planning to do that: Too much work, for a platform that is going to be obsoleted soon (Windows 7 EOL is January 2020) for a "workaround" that's only likely to ever help a couple people and might not even work, and adding loads of unwanted complexity, confusion (for users, who will now be asking what version of WinUSB they should install), and extra weight to the application (these files are not small), and that's without taking into account that the version as reported by Zadig is a Frankenstein's monster of many things (there's the driver and then there's the DLL) so what you think you might be seeing as a version might not be accurate at all, and Windows is doing its own thing with regards to the WinUSB driver files actually being installed behind the scenes.

Please remember that there's no real magic to what Zadig does &mdash; all it provides, really, is a driver bundle that you __can__ recreate on your own if you fetch the relevant files from Microsoft. Therefore, if you think that the driver bundle installed by Zadig is not suitable for your usage, you should really look into creating your own, with exactly the driver and DLL files you think you need, rather than ask a third party, who doesn't have access to your hardware (and isn't really interested in investigating your issue) to do that for you &rarr; Rejected.

closed 2019-07-05 by pbatard

Fliptron, 2019-07-05:
Fair enough. Thanks for taking the time to explain the situation, and making some recommendations.