pbatard/libwdi #163: Zadig - install-filter.exe not found while trying to install libusb-win32 filter driver

johmd, 2019-07-22:
I get a (localized) error message telling me that install-filter.exe was not found while trying to install the libusb-win32 filter driver in Zadig. The debug log does not name a path, but Sysinternals Process Monitor shows me, that Zadig tries to locate install-filter.exe in C:\Users\Nutzer\usb_driver\ (and after failing there looking in directories from the PATH variable, which probably should be considered dangerous) while the platform specific executables are located in the amd64 and x86 sub-directories.

The relevant line in the code seems to be https://github.com/pbatard/libwdi/blob/bc2336edbf0b10341156287fc797598d090b9958/libwdi/libwdi.c#L1567

Complete log follows:

```
Zadig 2.4.721
Windows 7 SP1 64-bit
ini file 'zadig.ini' not found - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
2 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_03EB&PID_2104&REV_0200
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7600.16385
libwdi:debug [wdi_create_list] WinUSB USB device (2): USB\VID_03EB&PID_2104\0000B0012345
libwdi:debug [wdi_create_list] Device description: 'AVRISP mkII'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_80EE&PID_0021&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.17514
libwdi:debug [wdi_create_list] HidUsb USB device (3): USB\VID_80EE&PID_0021\5&18F54CB7&0&1
libwdi:debug [wdi_create_list] Device description: 'USB Tablet'
Using inf name: AVRISP_mkII.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [wdi_prepare_driver] K driver available - adding the libusbK DLLs to the libusb-win32 inf
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\Nutzer\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\Nutzer\usb_driver\AVRISP_mkII.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'libusb0.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Nutzer\usb_driver\amd64\libusb0.dll'
libwdi:debug [AddFileHash] 'libusb0.sys': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Nutzer\usb_driver\amd64\libusb0.sys'
libwdi:debug [AddFileHash] 'libusb0_x86.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Nutzer\usb_driver\amd64\libusb0_x86.dll'
libwdi:debug [AddFileHash] 'libusbk.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Nutzer\usb_driver\amd64\libusbk.dll'
libwdi:debug [AddFileHash] 'libusbk_x86.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Nutzer\usb_driver\amd64\libusbk_x86.dll'
libwdi:debug [AddFileHash] 'avrisp_mkii.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Nutzer\usb_driver\avrisp_mkii.inf'
libwdi:debug [AddFileHash] 'libusb0.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Nutzer\usb_driver\x86\libusb0.dll'
libwdi:debug [AddFileHash] 'libusb0.sys': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Nutzer\usb_driver\x86\libusb0.sys'
libwdi:debug [AddFileHash] 'libusb0_x86.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Nutzer\usb_driver\x86\libusb0_x86.dll'
libwdi:debug [AddFileHash] 'libusbk.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Nutzer\usb_driver\x86\libusbk.dll'
libwdi:info [CreateCat] successfully created file 'C:\Users\Nutzer\usb_driver\AVRISP_mkII.cat'
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_03EB&PID_2104 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_03EB&PID_2104 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] created new key container
libwdi:debug [CreateSelfSignedCert] generated new keypair
libwdi:info [CreateSelfSignedCert] created new self-signed certificate 'CN=USB\VID_03EB&PID_2104 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] successfully created certificate 'CN=USB\VID_03EB&PID_2104 (libwdi autogenerated)'
libwdi:info [SelfSignFile] added certificate 'CN=USB\VID_03EB&PID_2104 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] successfully signed file 'C:\Users\Nutzer\usb_driver\AVRISP_mkII.cat'
libwdi:info [SelfSignFile] successfully deleted private key
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:info [install_driver_internal] could not find installer executable
Driver Installation: FAILED
2 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_03EB&PID_2104&REV_0200
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7600.16385
libwdi:debug [wdi_create_list] WinUSB USB device (2): USB\VID_03EB&PID_2104\0000B0012345
libwdi:debug [wdi_create_list] Device description: 'AVRISP mkII'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_80EE&PID_0021&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.17514
libwdi:debug [wdi_create_list] HidUsb USB device (3): USB\VID_80EE&PID_0021\5&18F54CB7&0&1
libwdi:debug [wdi_create_list] Device description: 'USB Tablet'

```

niczoom, 2019-08-17:
I get the same here, an error dialog popped up with "Windows cannot find 'install-filter.exe'. Make sure you typed the name correctly ...". The "Installing Driver..." with green progress bar sits there displaying messages about the amount of time it can take. Once I click OK on the error dialog the install stops with "Driver Installation: FAILED", log below:

> Using inf name: USB_Keyboard_(Interface_0).inf
Successfully extracted driver files.
libwdi:info [wdi_prepare_driver] K driver available - adding the libusbK DLLs to the libusb-win32 inf
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\Hardings\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\Hardings\usb_driver\USB_Keyboard_(Interface_0).inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'libusb0.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Hardings\usb_driver\amd64\libusb0.dll'
libwdi:debug [AddFileHash] 'libusb0.sys': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Hardings\usb_driver\amd64\libusb0.sys'
libwdi:debug [AddFileHash] 'libusb0_x86.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Hardings\usb_driver\amd64\libusb0_x86.dll'
libwdi:debug [AddFileHash] 'libusbk.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Hardings\usb_driver\amd64\libusbk.dll'
libwdi:debug [AddFileHash] 'libusbk_x86.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Hardings\usb_driver\amd64\libusbk_x86.dll'
libwdi:debug [AddFileHash] 'usb_keyboard_(interface_0).inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Hardings\usb_driver\usb_keyboard_(interface_0).inf'
libwdi:debug [AddFileHash] 'libusb0.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Hardings\usb_driver\x86\libusb0.dll'
libwdi:debug [AddFileHash] 'libusb0.sys': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Hardings\usb_driver\x86\libusb0.sys'
libwdi:debug [AddFileHash] 'libusb0_x86.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Hardings\usb_driver\x86\libusb0_x86.dll'
libwdi:debug [AddFileHash] 'libusbk.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Hardings\usb_driver\x86\libusbk.dll'
libwdi:info [CreateCat] successfully created file 'C:\Users\Hardings\usb_driver\USB_Keyboard_(Interface_0).cat'
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_046D&PID_C31F&MI_00 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_046D&PID_C31F&MI_00 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] created new key container
libwdi:debug [CreateSelfSignedCert] generated new keypair
libwdi:info [CreateSelfSignedCert] created new self-signed certificate 'CN=USB\VID_046D&PID_C31F&MI_00 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] successfully created certificate 'CN=USB\VID_046D&PID_C31F&MI_00 (libwdi autogenerated)'
libwdi:info [SelfSignFile] added certificate 'CN=USB\VID_046D&PID_C31F&MI_00 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] successfully signed file 'C:\Users\Hardings\usb_driver\USB_Keyboard_(Interface_0).cat'
libwdi:info [SelfSignFile] successfully deleted private key
Installing driver. Please wait...
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:info [install_driver_internal] operation cancelled by the user
Driver Installation: FAILED

All the files seem to be in the 'usb_driver' folder. When I try to run 'installer_x64.exe' from there a dialog just flashes and nothing happens.

I installed the program 'libusb-win32-devel-filter-1.2.6.0.exe' and ran 'install-filter-win.exe'. This installs the filter drivers without an issue.

mcuee, 2021-07-06:
If you are main interested in the filter driver, I will recommend libusbk. libusbk has the libwdi (albeit older version) based installer for libusb0.sys/libusbk.sys/WinUSB (similar for Zadig) and GUI filter installaer as well.

mcuee, 2021-07-07:
https://github.com/pbatard/libwdi/wiki/Zadig
Interestingly I have not used libusb-win32 filter driver for very long. And the above Zadig wiki does not mention the installation of filter driver as well. To be honest, the filter driver is only useful for libusb-win32 now and the users are strongly encouraged to migrate to libusb-1.0 API (libusb-1.0 Windows). 

Or if your application is only Windows, you can consider libusbk as well but the API is quite a bit different.

mcuee, 2021-09-13:
I can reproduce the issue in the latest Zadig release as well. For now the workaround is to use the filter driver installer from libusbk-inf-wizard as part of the libusbk release here: https://github.com/mcuee/libusbk/releases

mcuee, 2021-09-13:
> https://github.com/pbatard/libwdi/blob/bc2336edbf0b10341156287fc797598d090b9958/libwdi/libwdi.c#L1567

Indeed that may be an issue. I checked out the usb_driver folder and file name of "install-filter.exe" is the same for x86 and amd64. So the codes may not be correct.

mcuee, 2021-09-13:
BTW, usbdk may be a replacement for the filter driver, and libusb-1.0 Windows backend supports usbdk.

pbatard, 2021-09-21:
Xiaofan, thanks for your input on this. I think the missing subdirectory in the path is indeed the problem and I'll try to fix that in the next release.

commit d506f19e (2021-10-31): [core] fix filter-driver installation path

* Closes #163
closed 2021-10-31 by pbatard