pbatard/libwdi #164: External disk lost connection after my mistake (Western Digital My Passport unit)

Yoshideking, 2019-07-24:
Hi,

so I'm not well versed on computer but I'm not new on this, so I hope I can explain myself.

So I needed to connect my Nintendo Switch to the computer, and I think I made a mistake at first because I didn't replace the "device" at first, so I installed libusbK (v.3.0.7.0) on My Passport 25E2 (Western Digital external disk). And now I can't get to gain access again with this device.

The Western Digital utility doesn't recognize it. So I think I could test other drivers with zadig, but none seems to work.

Could you please lend me a hand, please?

I'm on Windows 10.

Data zadig gives me:
My Passport 25E2
Driver: WinUSB (v.6.1.7600.16385)
USB ID: 1058 25E2
WCID: X

I know I mess it up, but I can repair it with your advice.

pbatard, 2019-07-24:
https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it

Yoshideking, 2019-07-25:
Hi, I did everything and after this step (4):

                "Once the uninstallation is complete, unplug and replug your device. The original system driver should now be reinstalled. In our case, we would see the Flash Drive reappearing under the Disk drives category, which is what we want."

The device stills appears under:

driver provider: libwdi
driver date: 10.feb.2017
digital firm: USB\VID_1058&PID_25E2 (libwdi autogenerated)

pbatard, 2019-07-25:
The trick is to insist and make sure you repeat the process for all the USB ports you might have plugged your device in until the libwdi installed driver disappears.

I have never seen a system where what I explained above didn't work after persisting.

Alternatively, you can look around for driver uninstallation applications (but you will have to do your homework, as I'm not going to help you with that).

Yoshideking, 2019-08-22:
Problem solved thx to the tutorial linked. I just unninstal many times until it works.
I donne it, thanks. The tricks is to insist because I reinstall the driver many times with zadig.

My bad, thanks, your help is much appreciated.

closed 2019-08-22 by Yoshideking