pbatard/libwdi #165: Unable to install certificate on Windows 10

dhedstrom, 2019-08-14:
The certificate, my.cer is valid and already installed in the system, but I wanted to test the certificate installer. This was run from an Administrator WDK Win7 x64 Free Build Environment:

C:\dev>libwdi\x64\Release\examples\wdi-simple.exe --type 4 --name="My USB Device" --vid 0x2528 --pid 0x0004 --inf "my.inf" --progressbar --log 0 --cert "my.cer"
Extracting driver files...
libwdi:info [wdi_prepare_driver] custom driver - extracting binaries only (no inf/cat creation)
libwdi:info [extract_binaries] successfully extracted driver files to usb_driver
  Success
Installing certificate 'my.cer' as a Trusted Publisher...
libwdi:warning [AddCertToTrustedPublisher] could not create context for certificate: Unknown PKI error 0x8009310B

Thanks,
Dave

pbatard, 2019-08-15:
Any chance your `my.cer` contains Unicode encoded values? If so you might want to see [this](https://www.sysadmins.lv/blog-en/you-cannot-submit-a-certificate-request-generated-by-exchange-management-console-emc-or-exchange-management-shell-ems-to-microsoft-certificate-services.aspx).

At any rate the error reported comes from the Windows PKI API, not libwdi, so I don't think there's much that can be done here. Either you provide a certificate that the PKI APIs we use are happy with (doesn't matter if you can install it some other way), or you need to alter your certificate (or, you can also try to fill a bug with Microsoft so that they alter their Windows PKI implementation).

pbatard, 2019-08-22:
No update &rarr; Closing.

closed 2019-08-22 by pbatard