pbatard/libwdi #166: Get rid of WinUSB driver

ghatbkk, 2019-09-17:
I used Zadig to install a WinUSB driver for a POS printer for a project. Now, I want to go back to using the normal driver and am unable to delete or get rid of the WinUSB v6.1/7600/16385 driver. HELP!

pbatard, 2019-09-17:
[This](https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it) should answer your question, since the same applies whether you voluntary or involuntary installed the driver you want to remove.

And, __yes__, the solution is to insist. You may have to replug the device you installed WinUSB for in all the ports it was ever plugged, but the advice given in the FAQ does work.

ghatbkk, 2019-09-17:
It did work. I had to do it 3 times, but it eventually worked. Thanks!

closed 2019-09-17 by pbatard