pbatard/libwdi #168: Error: Driver installation cancelled by user, Windows 10

ratsrolocosx, 2019-10-10:
I am trying to download driver's onto my gamecube adapter, it's listed as WUP-028, and the driver installation fails everytime. I have tried running as admin, and restarting my computer. 

[Zadig.log](https://github.com/pbatard/libwdi/files/3710037/Zadig.log)

pbatard, 2019-10-10:
The problem is that, on your machine, Zadig/libwdi can not sign the `.cat` file:
```
libwdi:warning [CreateSelfSignedCert] could not obtain a key container: Unknown PKI error 0x8009001A
libwdi:warning [wdi_prepare_driver] could not sign cat file
```
As a result, since driver signature is enforced, the driver installation fails:
```
libwdi:debug [syslog] !!!  sig:                     Driver package catalog file does not contain a signature, and Code Integrity is enforced.
libwdi:debug [syslog] !!!  sig:                     Driver package failed signature validation. Error = 0xE0000247
```

Error `0x8009001A` is `NTE_KEYSET_ENTRY_BAD` which seems to indicate that a storage cryptographic container from your system is corrupt.

In other words, something must have been altered on your system, compared to a vanilla Windows 10 installation, that makes the call to [`CryptAcquireContext()`](https://docs.microsoft.com/en-us/windows/win32/api/wincrypt/nf-wincrypt-cryptacquirecontextw) fail with this error, and it is not something that Zadig/libwdi can fix.

According to [this](https://social.msdn.microsoft.com/Forums/en-US/f3ae9010-d5c0-4fd2-a863-b4527a605050/error-8009001a?forum=theforumofsho) you may have to the file from in `C:\Users\<userID>\AppData\Roaming\Microsoft\Crypto\RSA\` that got corrupted, but I'm afraid I can't help you identify which one.

At any rate, I'm afraid this issue is environmental, and it appears that you have corruption issues with your cryptography containers.

pbatard, 2019-10-18:
No update from OP &rarr; Closing

closed 2019-10-18 by pbatard