pbatard/libwdi #171: Zadig; Driver Installation Failed

TheMythicalCat, 2020-01-25:
I'm trying to install Wii U GameCube Controller drivers, and when I first did it it worked perfectly. But then it just... stopped working. So I went in to try to reinstall them. Whenever I try to, I get this in the log:
Zadig 2.4.721
Windows 10 64-bit (Build 18363)
ini file 'zadig.ini' not found - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
7 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C534&REV_2901&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.175
libwdi:debug [wdi_create_list] HidUsb USB device (1): USB\VID_046D&PID_C534&MI_00\7&225C0206&0&0000
libwdi:debug [wdi_create_list] Device description: 'USB Receiver (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_057E&PID_0337&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.175
libwdi:debug [wdi_create_list] HidUsb USB device (2): USB\VID_057E&PID_0337\15/07/2014
libwdi:debug [wdi_create_list] Device description: 'WUP-028'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C534&REV_2901&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.175
libwdi:debug [wdi_create_list] HidUsb USB device (4): USB\VID_046D&PID_C534&MI_01\7&225C0206&0&0001
libwdi:debug [wdi_create_list] Device description: 'USB Receiver (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C52B&REV_1209&MI_02
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.175
libwdi:debug [wdi_create_list] HidUsb USB device (6): USB\VID_046D&PID_C52B&MI_02\7&40C4738&0&0002
libwdi:debug [wdi_create_list] Device description: 'USB Receiver (Interface 2)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0A12&PID_0001&REV_8891
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_E0&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.449
libwdi:debug [wdi_create_list] BTHUSB USB device (7): USB\VID_0A12&PID_0001\6&22AE8F5B&0&6
libwdi:debug [wdi_create_list] Device description: 'CSR8510 A10'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C52B&REV_1209&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.175
libwdi:debug [wdi_create_list] HidUsb USB device (8): USB\VID_046D&PID_C52B&MI_01\7&40C4738&0&0001
libwdi:debug [wdi_create_list] Device description: 'USB Receiver (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C52B&REV_1209&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 1.10.78.0
libwdi:debug [wdi_create_list] HidUsb USB device (9): USB\VID_046D&PID_C52B&MI_00\7&40C4738&0&0000
libwdi:debug [wdi_create_list] Device description: 'USB Receiver (Interface 0)'
Checking for Zadig updates...

Checking release channel...

Updates: Unable to acces version data

Using inf name: WUP-028.inf
Successfully extracted driver files.
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\riley\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\riley\usb_driver\WUP-028.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\riley\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\riley\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'wup-028.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\riley\usb_driver\wup-028.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\riley\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\riley\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] successfully created file 'C:\Users\riley\usb_driver\WUP-028.cat'
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] created new key container
libwdi:debug [CreateSelfSignedCert] generated new keypair
libwdi:info [CreateSelfSignedCert] created new self-signed certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)'
[... 206 lines cut ...]
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.175
libwdi:debug [wdi_create_list] HidUsb USB device (4): USB\VID_046D&PID_C534&MI_01\7&225C0206&0&0001
libwdi:debug [wdi_create_list] Device description: 'USB Receiver (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C52B&REV_1209&MI_02
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.175
libwdi:debug [wdi_create_list] HidUsb USB device (6): USB\VID_046D&PID_C52B&MI_02\7&40C4738&0&0002
libwdi:debug [wdi_create_list] Device description: 'USB Receiver (Interface 2)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0A12&PID_0001&REV_8891
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_E0&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.449
libwdi:debug [wdi_create_list] BTHUSB USB device (7): USB\VID_0A12&PID_0001\6&22AE8F5B&0&6
libwdi:debug [wdi_create_list] Device description: 'CSR8510 A10'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C52B&REV_1209&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.175
libwdi:debug [wdi_create_list] HidUsb USB device (8): USB\VID_046D&PID_C52B&MI_01\7&40C4738&0&0001
libwdi:debug [wdi_create_list] Device description: 'USB Receiver (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C52B&REV_1209&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 1.10.78.0
libwdi:debug [wdi_create_list] HidUsb USB device (9): USB\VID_046D&PID_C52B&MI_00\7&40C4738&0&0000
libwdi:debug [wdi_create_list] Device description: 'USB Receiver (Interface 0)'

I've tried deleting usb_driver, I've restarted my computer a dozen times, I can't do anything with it in my drivers. I've read through every single link on the top page of Google. Nothing works.

pbatard, 2020-01-25:
Those `0xffff` and `0x8007ffff` seem to indicate that there is something wrong with your environment.

The fact that an installed driver stopped working, and that you get tons of errors when trying to install it again tends to indicate that there is something fundamentally wrong going on with your OS device driver stack.

I'm afraid you are the very first person to report such an issue, so there isn't much I can do to help.

You probably want to check the integrity of your Windows environment, by issuing [`sfc /scannow`](https://support.microsoft.com/en-ie/help/929833/use-the-system-file-checker-tool-to-repair-missing-or-corrupted-system) or using the refresh feature of Windows 10.

pbatard, 2020-03-17:
I will close this issue now.

closed 2020-03-17 by pbatard