pbatard/libwdi #172: Driver installation cancelled by user.

arn-soft, 2020-01-28:
I am facing this issue i read all of the issue people have faced with this driver. and this issue is repetitive one. i know it is not zadig problem but environment issue. can anybody got this working.
i am kind of stuck here. anybody can show a way to fix this environment issue will help a lot.

libwdi:warning [CreateSelfSignedCert] could not obtain a key container: Unknown PKI error 0x8009001A
libwdi:warning [wdi_prepare_driver] could not sign cat file

arn-soft, 2020-01-28:
Once in a blue moon it works too. i am adding a log here.
[Zadig.log](https://github.com/pbatard/libwdi/files/4120822/Zadig.log)

schollman, 2020-01-30:
For what it is worth, I was able to work around this same problem by using zadig after temporarily disabling the driver signature enforcement by following Solution #1 on this page. https://windowsreport.com/driver-signature-enforcement-windows-10/

robskie, 2020-02-03:
I'm also experiencing this issue after I installed ESET NOD32. Uninstalling ESET and renaming the RSA folder in %PROGRAMDATA%\Microsoft\Crypto to RSA.old solves the problem.

pbatard, 2020-03-17:
Yes, I'm afraid this kind of issue seems to be environmental, and not something Zadig can work around on its own. If your system can't sign a `.cat` file, it means that there's some underlying problem with how your PKI was configured, which may be due to administrative policies, security solutions and so on...

All I can say is that error `0x8009001A` is `NTE_KEYSET_ENTRY_BAD` / _"Keyset as registered is invalid."_ which unfortunately doesn't tell much.

Oh and @arn-soft, your log explicitly states this:
```
libwdi:info [wdi_prepare_driver] .cat generation disabled by user
```
which means that, of course installation will fail on a machine that enforces signature validation, which is the default of any modern Windows installation.

This is what happens if you __explicitly choose__ to uncheck the relevant option in Zadig, so make sure it is checked, okay:
[image]

But then, I have to point out that, besides the line above, you get a fairly explicit warning in the log, which is different from OP's:
```
libwdi:debug [syslog] !!!  sig:                     Driver package does not contain a catalog file, and Code Integrity is enforced.
libwdi:debug [syslog] !!!  sig:                     Driver package failed signature validation. Error = 0xE000022F
```

Considering that none of these problems are due to a bug in libwdi or Zadig, I will close this issue.

closed 2020-03-17 by pbatard

tombresson, 2025-01-30:
> I'm also experiencing this issue after I installed ESET NOD32. Uninstalling ESET and renaming the RSA folder in %PROGRAMDATA%\Microsoft\Crypto to RSA.old solves the problem.

Thank you, so very much!

I had a hunch NOD32 was intervening in the cert store, but I didn't know how to get rid of the existing keys. I was looking in `%AppData%\Microsoft\Crypto\RSA`, which doesn't help. This resolved my issue with not being able to generate a cert.