pbatard/libwdi #174: Uninstall this Zadic thing

delaware1999, 2020-02-10:
Hello I wanted to use the dslrdashboard  for my Canon EOS 1300D

I had to use first Zadic to get the USB drivers to work otherwise dslrdashboard will not find the EOS 1300D Camara.

But then I found out that dslrdashboard is not good !  I want to use now a other app.

But in windows my EOS 1300D Camara is not there anymore..
I  must deinstall this useless ZADIC USB Drivers now and I dont know how.. I used google but nothing helped really.

Can you tell me how to solve this problem.. I want that ma EOS 1300D shows up when I connect it with my window 7 system.. Please help me

pbatard, 2020-02-10:
https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it

closed 2020-02-26 by pbatard

chauhan01, 2023-09-02:
I faced the same issue. The only solution that worked for me is to delete winusb.sys manually. Look for the file in c/windows/system32/drivers/winusb.sys and delete the file. You may also need to change the permissions to delete the file. This video can help.
https://www.youtube.com/watch?v=FaVUe8jAn_E

pbatard, 2023-09-02:
> The only solution that worked for me is to delete winusb.sys manually.

I have to strongly advise against doing that.
`winusb.sys` is a __system file__ and you may have other hardware devices that requires this driver to be present to work (since any manufacturer can create a USB device that uses WinUSB rather than a proprietary driver, [so that users don't need to install anything for that device to work](https://github.com/pbatard/libwdi/wiki/WCID-Devices)).

So, if you do delete `winusb.sys`, you are likely to prevent other USB devices from working, which you may not even realise until months later.