pbatard/libwdi #176: Issue Installing (Other Issue)

JarrodC-jpg, 2020-03-16:
Hope you can help, trying to install the winUSB driver. Here is the intall log:

```
Using inf name: Unknown_Device_#1.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\Jarrod\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\Jarrod\usb_driver\Unknown_Device_#1.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Jarrod\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Jarrod\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'unknown_device_#1.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Jarrod\usb_driver\unknown_device_#1.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Jarrod\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Jarrod\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] successfully created file 'C:\Users\Jarrod\usb_driver\Unknown_Device_#1.cat'
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_0925&PID_3881 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_0925&PID_3881 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] created new key container
libwdi:debug [CreateSelfSignedCert] generated new keypair
libwdi:info [CreateSelfSignedCert] created new self-signed certificate 'CN=USB\VID_0925&PID_3881 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] successfully created certificate 'CN=USB\VID_0925&PID_3881 (libwdi autogenerated)'
libwdi:info [SelfSignFile] added certificate 'CN=USB\VID_0925&PID_3881 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] successfully signed file 'C:\Users\Jarrod\usb_driver\Unknown_Device_#1.cat'
libwdi:info [SelfSignFile] successfully deleted private key
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter Unknown_Device_#1.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0925&PID_3881\6&16BFD654&0&15'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_0925&PID_3881&REV_0000'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-1698916166-1716680693-442871980-1001'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_0925&PID_3881&REV_0000 - please wait...
libwdi:debug [syslog] 6
libwdi:debug [syslog] Check]
libwdi:debug [syslog] >>>  Section start 2020/03/16 12:33:01.477
libwdi:debug [syslog] >  [Device Installation Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2020/03/16 11:47:09.568
libwdi:debug [syslog] a3f1bf6e7e}#SBConnect&6&187ca6a2&0} 11:40:16.656
libwdi:debug [syslog]      dvs:      Installing best driver (oem95.inf) on device 'SWD\DRIVERENUM\{1BBB99B1-03DF-4468-81DC-11A3F1BF6E7E}#SBCONNECT&6&187CA6A2&0'.
libwdi:debug [syslog]      ump:      Installation will be processed asynchronously
libwdi:debug [syslog] !!!  dvs:      Device install failed for device.
libwdi:debug [syslog] !!!  dvs:      Error 0xe0000246: One of the installers for this device cannot perform the installation at this time.
libwdi:debug [syslog] !!!  dvs:      Failed to install device instance 'SWD\DRIVERENUM\{1bbb99b1-03df-4468-81dc-11a3f1bf6e7e}#SBConnect&6&187ca6a2&0'. Error = 0xe0000246
libwdi:debug [syslog]      dvs: {Driver Setup Update Device - exit(0x00000000)} 11:40:16.664
libwdi:debug [syslog] <<<  Section end 2020/03/16 11:40:16.667
libwdi:debug [syslog] <<<  [Exit status: FAILURE(0xe0000246)]
libwdi:debug [syslog] [Boot Session: 2020/03/16 11:42:03.500]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_0925&PID_3881&REV_0000]
libwdi:debug [syslog] >>>  Section start 2020/03/16 11:47:09.186
libwdi:debug [syslog]       cmd: "C:\Users\Jarrod\usb_driver\installer_x64.exe" "Unknown_Device_#1.inf"
libwdi:debug [syslog]   cpy:                     Published 'creativesbconnect.inf_amd64_8a4dd90c4cad77ad\creativesbconnect.inf' to 'oem95.inf'.
libwdi:debug [syslog]      idb:                     Indexed 2 device IDs for 'creativesbconnect.inf_amd64_8a4dd90c4cad77ad'.
libwdi:debug [syslog]      sto:                     Flushed driver database node 'DRIVERS'. Time = 0 ms
libwdi:debug [syslog]      sto:                     Flushed driver database node 'SYSTEM'. Time = 0 ms
libwdi:debug [syslog]      idb:                {Publish Driver Package: exit(0x00000000)} 11:40:16.616
libwdi:debug [syslog]      sto:                {DRIVERSTORE IMPORT END} 11:40:16.618
libwdi:debug [syslog]      dvi:                     Flushed all driver package files to disk. Time = 0 ms
libwdi:debug [syslog]      sig:                     Installed catalog 'CreativeSBConnect.cat' as 'oem95.cat'.
libwdi:debug [syslog]      sto:                {DRIVERSTORE IMPORT END: exit(0x00000000)} 11:40:16.631
libwdi:debug [syslog]      sto:           {Core Driver Package Import: exit(0x00000000)} 11:40:16.632
[... 79 lines cut ...]
libwdi:debug [syslog] >>>  Section start 2020/03/16 11:47:09.186
libwdi:debug [syslog]       cmd: "C:\Users\Jarrod\usb_driver\installer_x64.exe" "Unknown_Device_#1.inf"
libwdi:debug [syslog] ssed asynchronously
libwdi:debug [syslog] !!!  dvs:      Device install failed for device.
libwdi:debug [syslog] !!!  dvs:      Error 0xe0000246: One of the installers for this device cannot perform the installation at this time.
libwdi:debug [syslog] !!!  dvs:      Failed to install device instance 'SWD\DRIVERENUM\{1bbb99b1-03df-4468-81dc-11a3f1bf6e7e}#SBConnect&6&187ca6a2&0'. Error = 0xe0000246
libwdi:debug [syslog]      dvs: {Driver Setup Update Device - exit(0x00000000)} 11:40:16.664
libwdi:debug [syslog] <<<  Section end 2020/03/16 11:40:16.667
libwdi:debug [syslog] <<<  [Exit status: FAILURE(0xe0000246)]
libwdi:debug [syslog] [Boot Session: 2020/03/16 11:42:03.500]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_0925&PID_3881&REV_0000]
libwdi:debug [syslog] >>>  Section start 2020/03/16 11:47:09.186
libwdi:debug [syslog]       cmd: "C:\Users\Jarrod\usb_driver\installer_x64.exe" "Unknown_Device_#1.inf"
libwdi:debug [process_message] switching timeout back to finite
libwdi:debug [installer process] unhandled error 0xE0000246 (-536870330)
libwdi:debug [installer process] [3758096966] One of the installers for this device cannot perform the installation at this time.
libwdi:debug [process_message] installer process completed
Driver Installation: FAILED (Other error)
1 device found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0925&PID_3881&REV_0000
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_FF&Prot_FF
libwdi:debug [wdi_create_list] Driverless USB device (12): USB\VID_0925&PID_3881\6&16BFD654&0&15
libwdi:debug [wdi_create_list] could not read device description for 12: [#0000000D] The data is invalid.
libwdi:debug [wdi_create_list] Device description: 'Unknown Device #1'
```

pbatard, 2020-03-17:
Well, the log is pretty clear:
```
Error 0xe0000246: One of the installers for this device cannot perform the installation at this time.
```

This seems to indicate a purely environmental problem. In other words, this is not something Zadig can solve, because the issue is that, for one reason or another, your Windows system is unable to perform driver installation, and you would see the same issue if you just extracted the driver files and installed them manually, without using Zadig, by right clicking on the `.inf`.

Therefore, I'm afraid you're going to have to find out what exactly is preventing your system from installing drivers. It could be an administrative policy, or a service that is not running or a myriad of other factors that are next to impossible to troubleshoot without physically accessing the machine.

You may want to read [this post](https://answers.microsoft.com/en-us/windows/forum/all/windows-10-all-drivers-fail-to-install/085e06c4-ff99-4c81-882a-f1a6ab50b49f) that comes up when googling the error you get. But apart from that, I'm afraid I don't have any insight as to what might be the cause of your issue, besides the fact that, for one reason or another, your Windows installation must have been altered.

JarrodC-jpg, 2020-03-19:
Ahh I was afraid of that, Thank you!

pbatard, 2020-03-19:
I will close this issue then.

closed 2020-03-19 by pbatard