pbatard/libwdi #177: Zadig - Driver Installation: FAILED (Requested resource not found)

jakeasa123, 2020-03-24:
I build a tiny program which could use POS Printer to print some text with Python and Windows 10 64 bit few days ago.
I need to set up printer use libusbK driver, so I found Zadig and success; today, I tried to set up for 2nd PC (Windows 10 64 bit) to check program have some problem or not, but I get some error and the log as below:
> Zadig 2.4.721
Windows 10 64-bit (Build 18362)
ini file 'zadig.ini' not found - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
3 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_13FE&PID_5200&REV_0110
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_08&SubClass_06&Prot_50
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.1
libwdi:debug [wdi_create_list] USBSTOR USB device (1): USB\VID_13FE&PID_5200\070B694FA8BFA669
libwdi:debug [wdi_create_list] Device description: 'USB DISK 3.0'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0483&PID_5743&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_07&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.1
libwdi:debug [wdi_create_list] usbprint USB device (2): USB\VID_0483&PID_5743\012345678AB
libwdi:debug [wdi_create_list] Device description: 'Printer-80'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F3&PID_00A4&REV_1023
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.175
libwdi:debug [wdi_create_list] HidUsb USB device (3): USB\VID_04F3&PID_00A4\5&222E27AA&0&5
libwdi:debug [wdi_create_list] Device description: '2.4G RX'
Using inf name: Printer-80.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\UserName\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\UserName\usb_driver\Printer-80.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:info [CreateCat] successfully created file 'C:\Users\UserName\usb_driver\Printer-80.cat'
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_0483&PID_5743 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_0483&PID_5743 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] created new key container
libwdi:debug [CreateSelfSignedCert] generated new keypair
libwdi:info [CreateSelfSignedCert] created new self-signed certificate 'CN=USB\VID_0483&PID_5743 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] successfully created certificate 'CN=USB\VID_0483&PID_5743 (libwdi autogenerated)'
libwdi:info [SelfSignFile] added certificate 'CN=USB\VID_0483&PID_5743 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] successfully signed file 'C:\Users\UserName\usb_driver\Printer-80.cat'
libwdi:info [SelfSignFile] successfully deleted private key
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter Printer-80.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0483&PID_5743\012345678AB'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_0483&PID_5743&REV_0100'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-2323712114-2148394435-215344373-1001'
libwdi:debug [installer process] using syslog 'C:\Windows\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_0483&PID_5743&REV_0100 - please wait...
libwdi:debug [syslog] s\UserName\usb_driver\installer_x64.exe" "Printer-80.inf"
libwdi:debug [syslog]  11:09:38.256
libwdi:debug [syslog]      dvi:      {DIF_SELECTBESTCOMPATDRV} 11:09:38.272
libwdi:debug [syslog]      dvi:           Default installer: Enter 11:09:38.272
libwdi:debug [syslog]      dvi:                {Select Best Driver}
libwdi:debug [syslog]      dvi:                     Class GUID of device changed to: {ecfb0cfd-74c4-4f52-bbf7-343461cd72ac}.
libwdi:debug [syslog]      dvi:                     Selected Driver:
libwdi:debug [syslog]      dvi:                          Description - Printer-80
libwdi:debug [syslog]      dvi:                          InfFile     - c:\users\UserName\usb_driver\printer-80.inf
libwdi:debug [syslog]      dvi:                          Section     - LUsbK_Device
libwdi:debug [syslog]      dvi:                {Select Best Driver - exit(0x00000000)}
libwdi:debug [syslog]      dvi:           Default installer: Exit
libwdi:debug [syslog]      dvi:      {DIF_SELECTBESTCOMPATDRV - exit(0x00000000)} 11:09:38.303
libwdi:debug [syslog]      ndv:      Force Installing Driver:
libwdi:debug [syslog]      ndv:           Inf Name       - printer-80.inf
libwdi:debug [syslog]      ndv:           Driver Date    - 12/01/2014
libwdi:debug [syslog]      ndv:           Driver Version - 3.0.7.0
libwdi:debug [syslog]      sto:      {Setup Import Driver Package: c:\users\UserName\usb_driver\printer-80.inf} 11:09:38.318
[... 44 lines cut ...]
libwdi:debug [process_message] switching timeout back to finite
libwdi:debug [installer process] unable to locate the file hashes in cat file
libwdi:debug [process_message] installer process completed
Driver Installation: FAILED (Requested resource not found)
3 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_13FE&PID_5200&REV_0110
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_08&SubClass_06&Prot_50
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.1
libwdi:debug [wdi_create_list] USBSTOR USB device (1): USB\VID_13FE&PID_5200\070B694FA8BFA669
libwdi:debug [wdi_create_list] Device description: 'USB DISK 3.0'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0483&PID_5743&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_07&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.1
libwdi:debug [wdi_create_list] usbprint USB device (2): USB\VID_0483&PID_5743\012345678AB
libwdi:debug [wdi_create_list] Device description: 'Printer-80'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F3&PID_00A4&REV_1023
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.18362.175
libwdi:debug [wdi_create_list] HidUsb USB device (3): USB\VID_04F3&PID_00A4\5&222E27AA&0&5
libwdi:debug [wdi_create_list] Device description: '2.4G RX'
>
I'm sorry that I'm not good at English and not very often use github, so there had some delete line in the log.
In the log, 'UserName' is my 3rd PC's user name, it is Chinese text before I replace.

Sincerely.

pbatard, 2020-03-24:
The reason for your issue is this:
```
libwdi:debug [syslog] !!!  sig:                     Driver package INF file hash is not present in catalog file. Filename = printer-80.inf, Error = 0xE000024B
libwdi:debug [syslog] !!!  sig:                     Driver package appears to be tampered, and Code Integrity is enforced.
libwdi:debug [syslog] !!!  sig:                     Driver package failed signature validation. Error = 0xE000024B
```

But Zadig should have added the hash for the `.inf` in the catalog file, so I'm not sure I understand why that happens.

Possibly this could have to do with your username. I designed libwdi/Zadig to be fully Unicode compliant so it shouldn't care about the name for the user directory, but of course, I haven't tested a lot with non western directory names, so maybe there is some issue.

From what I understand, you edited the log to replace the actual user name with `UserName`, so it's a bit unfortunate, because it means I can't test your conditions. I also find it a bit suspicious that there is a trailing `~1` to your username when it's looking for files in AppData (e.g. `C:\Users\UserName~1\AppData\...`) but not when looking for the driver files (`c:\users\UserName\usb_driver`). This shouldn't matter for catalog file validation, but it makes it difficult for me to understand what your original data might have been.

One thing you can try to do is:
1. Instead of clicking _Install Driver_ click _Extract Files (Don't install)_
2. Go to the directory where the driver files have been installed (should be `c:\users\UserName\usb_driver\`) and right click on the `.inf` file then select install.
3. Check `C:\Windows\INF\setupapi.dev.log` to see if it displays the same error (the `!!!  sig:                     ... Error = 0xE000024B`).
4. If it has the same error, try to move `usb_driver\` to something like `C:\tmp\` or any directory that does not contain Chinese character and try the right click installation again.

jakeasa123, 2020-03-25:
Hello, pbatard. 

First, thanks to your kind reply!

Second, I'm sorry about replace user name in the log.
I do not own he PC that meet this problem, so I replace user name. It is a company's name, I'm sorry that I can't give you full name. Part's of user name is '錸xx科技有限公司'.

Third, I tried what you said few minutes ago.
1. Click _Extract Files (Don't install)_ 
2. Find `.inf` file in `c:\users\錸xx科技有限公司\usb_driver\` and right click on the `.inf` file and select Install. After select Install, I get an error message box.
3. Because of the error, I move `c:\users\錸xx科技有限公司\usb_driver\` to `c:\usb_driver\` and select Install again, but I get same error as step 2.

The error is Chinese, title: `安裝錯誤`, contain information: `指定的類別目錄檔案中沒有該檔案的雜湊。該檔案可能已損毀或被竄改。`
If translate by Google, title: `Installation error`, contain information: `The specified catalog file does not have a hash for that file. The file may be damaged or tampered with.`

Sincerely.

pbatard, 2020-03-25:
Thanks for the test. From the error you get with 3., it means that the problem is unlikely to be with the user name.

Can I ask you to please zip the `Printer-80.inf` and `Printer-80.cat` files you have in the `usb_driver\` driver directory and e-mail them to me (pete@akeo.ie) so I can take a look? Just the the `.inf` and `.cat` are needed, as the other files come from Zadig and are not modified.

And just in case you worry, the `.inf` and `.cat` don't contain any private information, so they should be safe to share. For instance, they don't even contain the name of the user that generated them, if you are worried about that.

jakeasa123, 2020-03-25:
Hello, pbatard.

Thanks to your kind and speedy reply again!

After previous reply which I wrote, I tried to add a new user with full English name as `test` at the same PC, and use Zadig again. Then, driver install success :)

I will send the files you said to you after this reply.

Sincerely.

pbatard, 2020-03-25:
Okay, I've finally got a chance to look at the files you sent (thanks!) and the `.cat` file is actually empty.

In other words, a normal `.cat` file should contain catalog entries for the various driver files, including the `.inf`, like so:
[image]

But yours does not contain any entries at all:
[image]

So that explains the error you get.

And it seems the log also confirms this because it should have had lines such as:

```
libwdi:debug [AddFileHash] 'xbox_controller.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\UserName\usb_driver\Printer-80.inf'
```
between:
```
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
```
and 
```
libwdi:info [CreateCat] successfully created file 'C:\Users\UserName\usb_driver\Printer-80.cat'
```
but none of that happened...

So it looks like, even though it created the `.inf` and the other driver files, libwdi failed to actually process them when generating the `.cat`.

This looks like a bug in libwdi and considering that this hasn't been reported before, I'm starting to think, again, that this could have something to do with using a user directory with a non-western name.

I will take a closer look at my code, and see if I can figure out what might be causing libwdi to ignore the files, because it definitely should not happen.

jakeasa123, 2020-03-26:
Hello, pbatard.

Thanks to your kind and speedy reply again!

I'll be glad if those file could help you, and wish you could found the problem and solve it.

Sincerely.

pbatard, 2020-03-28:
I think I may have found the issue, but I need to validate that this is the case.

I just sent you an e-mail with a link to a new version of Zadig that I hope solved the problem. Can I please ask you to test and confirm that this is the case?

Thanks!

jakeasa123, 2020-03-30:
Hello, pbatard.

I have seen the e-mail and downloaded new version of Zadig that you said.
But I can't use that PC right now, maybe need waits few hours or tomorrow.

I would give you the test result as soon as possible that I could.

Sincerely.

jakeasa123, 2020-03-30:
Hello, pbatard.

I have tried it about 5 minutes ago.

The new version of Zadig install libusbK on the PC that user name include Chinese text successful. 
After driver install successfully, my tiny program could print as my excepted.

Thank you!

Sincerely.

pbatard, 2020-03-30:
I'm very happy to hear that! Thank you very much for your testing.

I will close this issue then.

commit 21d23dd8 (2020-03-30): [pki] fix handling of non western paths during cat generation

* Closes #177
closed 2020-03-30 by pbatard