pbatard/libwdi #180: How to use libwdi to install a driver for a PCI/PCIe device?

trollcop, 2020-04-28:
Is it possible?

pbatard, 2020-04-28:
No, it's for USB devices only:
https://github.com/pbatard/libwdi/blob/d09916e83e8fba9f5abcbc9a42cf5f69726c28b0/libwdi/libwdi.c#L718

If you want to use libwdi for something else, you will need to fork and modify your own version.

closed 2020-04-28 by pbatard

trollcop, 2020-04-28:
Yes, I'm aware its for USB devices in its current state, and I did see that
enumeration, is supporting a PCI device as simple as changing that to PCI
and following same steps or do PCI devices require additional WHQL
signatures etc?

pbatard, 2020-04-28:
> is supporting a PCI device as simple as changing that to PCI and following same steps or do PCI devices require additional WHQL signatures etc?

No idea. I've only developed the library for USB device usage and I have not looked into any other type. So I'm afraid it's up to you to investigate...

trollcop, 2020-04-28:
Thanks for the information.