pbatard/libwdi #182: how to compile wdi-simple.exe??

khushbooY, 2020-05-05:
Sir I am not getting any link or idea how to compile wdi-simple.exe.
Actually i m trying to automate Zadig driver replacer for Win-USB and want to install Zadig driver while installing my application.there are some thing i don't get in ur wdi-simple.iss code could you please help!!

1.) how to compile wdi-simple.exe what does it actually need win-usb driver's inf or sys ??
2.) what does this part of code mean which file should be in these directories i didn't get??

/*

Embed WinUSB driver files from the following WDK location
NB: You must also make sure the WDF_VER, COINSTALLER_DIR and X64_DIR
match your WinUSB redist directrories
*/
#ifndef WDK_DIR
#define WDK_DIR "C:/Program Files (x86)/Windows Kits/10"
#endif
/* WDK WDF coinstaller version */
#define WDF_VER 1011

/* CoInstaller subdirectory for WinUSB redist files ("winusb" or "wdf") */
#define COINSTALLER_DIR "wdf"

/* 64bit subdirectory for WinUSB redist files ("x64" or "amd64") */
#define X64_DIR "x64"

3.) what things are required to create win.exe and use it??

I know my question is overwhelming but i seriouly didn't get it??

pbatard, 2020-05-05:
https://github.com/pbatard/libwdi/wiki/Compiling-and-debugging-libwdi-or-Zadig

pbatard, 2020-05-05:
I'm going to assume that the above did the trick and therefore close the issue.

closed 2020-05-05 by pbatard

khushbooY, 2020-05-06:
sir i am getting this error while compilation.
[image]

khushbooY, 2020-05-06:
i tried the alternative which is given but still same error could you please help

khushbooY, 2020-05-06:
[image]

khushbooY, 2020-05-06:
After the relaunching of visual studio by different cerdentails i get this
[image]

khushbooY, 2020-05-06:
Thank You sir It worked!!

pbatard, 2020-05-06:
Great. I'm very happy to hear that!