pbatard/libwdi #183: Put generated files into a better place and name

altimumdelta, 2020-05-05:
It would be cool if Zadig respected some of the good practices of putting generated files instead of throwing them into a unexpected place into a very generic directory name 

`\Users\Username\usb_driver`

pbatard, 2020-05-05:
What would you suggest?

altimumdelta, 2020-05-05:
Hmm, oh okay, ... well hmm I'm kinda out of ideas right now, because I haven't thought about it too much about the alternative. It does feel like a more special thing that perhaps could go somewhere more special and not the standard AppData ...

pbatard, 2020-05-05:
> because I haven't thought about it too much about the alternative.

Yeah, that the issue I had. Once you start thinking about alternatives, you realize that:
1. You want to have those files in the current user directory
2. You don't want to create individual folder for every driver they install

Which means you want to use `\Users\<username>\<something_descriptive>\` as the parent folder, and soon enough you find that `\Users\<username>\usb_driver\` is about as good at it'll get.

Especially you don't want to use _Zadig_ there, because Zadig is just a sample application that relies on libwdi, and people who use different libwdi apps would then end up with different directories for each one, which isn't too nice.

I'm going to close this issue then. But if you have a specific example as to what a _"more special thing"_ might look like, feel free to provide it.

closed 2020-05-05 by pbatard

altimumdelta, 2020-05-05:
I mostly have a problem figuring out the source application which created them, in this case it didn't took long to figure it out judging by the files and their names inside, but I still forget about it sometimes and check the folder out "what's this" .,. when you do maintenance with hundreds of folders and thousands of files it does get problematic. Many multi-platform apps have now started this trend of .dot folders in the user folder ... it's incredibly messy with so many locations on windows at least.

I guess I'm not familiar with libwdi and that this is not Zadig really and that Zadig is just one of them.