pbatard/libwdi #185: zadic --useinf fails

spiro-trikaliotis, 2020-05-25:
Hello,

zadic --useinf fails because it does not find its installer_x64.exe or installer_x86.exe.
The extraaction of the installer is done in libwdi_prepare_driver. But, as it is not called with --useinf, the extraction is not done, and libwdi does not find it.

The exact output is:

libwdi:error [install_driver_internal] this application does not contain the required 64 bit installer
libwdi:error [install_driver_internal] please contact the application provider for a 64 bit compatible version

pbatard, 2020-05-30:
Well, considering that I am not providing any version of zadic, the problem is that the version you compiled only embeds the 32-bit drivers.

Please pay close attention to the `EMBED` lines when compiling. If you don't see the amd64  version of the drivers being picked up, as is the case below, you need to sort that:
```
6>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x86\WdfCoInstaller01011.dll (2018.08.09)
6>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x86\winusbcoinstaller2.dll (2018.08.09)
6>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x64\WdfCoInstaller01011.dll (2018.08.09)
6>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x64\winusbcoinstaller2.dll (2018.08.09)
6>  EMBED  D:\libusb-win32\bin\x86\libusb0_x86.dll (2012.08.02)
6>  EMBED  D:\libusb-win32\bin\x86\install-filter.exe (2012.08.02)
6>  EMBED  D:\libusb-win32\bin\x86\libusb0.sys (2012.08.02)
6>  EMBED  D:\libusb-win32\bin\amd64\libusb0.dll (2012.08.02)
6>  EMBED  D:\libusb-win32\bin\amd64\libusb0.sys (2012.08.02)
6>  EMBED  D:\libusb-win32\bin\amd64\install-filter.exe (2012.08.02)
6>  EMBED  D:\libusb-win32\installer_license.txt (2012.08.02)
6>  EMBED  D:\libusbK\bin\sys\x86\libusbK.sys (2014.11.30)
6>  EMBED  D:\libusbK\bin\dll\x86\libusbK.dll (2014.11.30)
6>  EMBED  D:\libusbK\bin\sys\amd64\libusbK.sys (2014.11.30)
6>  EMBED  D:\libusbK\bin\dll\amd64\libusbK.dll (2014.11.30)
6>  EMBED  D:\libwdi\Win32\Debug\helper\installer_x86.exe (2020.05.30)
6>  EMBED  D:\libwdi\x64\Debug\helper\installer_x64.exe (2020.05.30)
6>  EMBED  D:\libwdi\libwdi\winusb.inf.in (2020.04.01)
6>  EMBED  D:\libwdi\libwdi\libusb0.inf.in (2020.04.01)
6>  EMBED  D:\libwdi\libwdi\libusbk.inf.in (2020.04.01)
6>  EMBED  D:\libwdi\libwdi\usbser.inf.in (2020.04.01)
6>  EMBED  D:\libwdi\libwdi\winusb.cat.in (2020.04.01)
6>  EMBED  D:\libwdi\libwdi\libusb0.cat.in (2020.04.01)
6>  EMBED  D:\libwdi\libwdi\libusbk.cat.in (2020.04.01)
6>  EMBED  D:\libwdi\libwdi\usbser.cat.in (2020.04.01)
```

Also please be mindful that I am planning to remove zadic from libwdi, as most people should use wdi-simple instead, so I will only provide limited support for it at this stage.

spiro-trikaliotis, 2020-05-30:
Hello Pete,
I am using the exact same compiled zadic.exe to install USB drivers. Both installer_x64.exe and install_x86.exe are EMBEDed (see compilation results at the end).

I set a breakpoint in extract_binaries(): It is not called if I use the -useinf option. It is only called in wdi_prepare_driver() (two times, one as special path for WDI_USER). 

But wdi_prepare_driver() is only called if "desc" is set (creating a device from scratch, line 157 of zadic.c), or if use_supplied_inf_flag is 0 (line 203 of zadic.c). Thus, if I set --useinf (and use_supplied_inf_flag is 1!), then wdi_prepare_driver() is not called, so extract_binaries is not called. either.

(Please note that I do not want libusb0 nor libusbk, so I commented out the location to them):

The output of the EMBEDer:
2>Embedding binary resources
2>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x86\WdfCoInstaller01011.dll (2019.04.18)
2>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x86\winusbcoinstaller2.dll (2019.04.18)
2>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x64\WdfCoInstaller01011.dll (2019.04.18)
2>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x64\winusbcoinstaller2.dll (2019.04.18)
2>  EMBED  C:\Users\spiro\Documents\libwdi\Win32\Debug\helper\installer_x86.exe (2020.05.30)
2>  EMBED  C:\Users\spiro\Documents\libwdi\x64\Debug\helper\installer_x64.exe (2020.05.30)
2>  EMBED  C:\Users\spiro\Documents\libwdi\libwdi\winusb.inf.in (2020.05.30)
2>  EMBED  C:\Users\spiro\Documents\libwdi\libwdi\libusb0.inf.in (2020.05.30)
2>  EMBED  C:\Users\spiro\Documents\libwdi\libwdi\libusbk.inf.in (2020.05.30)
2>  EMBED  C:\Users\spiro\Documents\libwdi\libwdi\usbser.inf.in (2020.05.30)
2>  EMBED  C:\Users\spiro\Documents\libwdi\libwdi\winusb.cat.in (2020.05.30)
2>  EMBED  C:\Users\spiro\Documents\libwdi\libwdi\libusb0.cat.in (2020.05.30)
2>  EMBED  C:\Users\spiro\Documents\libwdi\libwdi\libusbk.cat.in (2020.05.30)
2>  EMBED  C:\Users\spiro\Documents\libwdi\libwdi\usbser.cat.in (2020.05.30)

But, if you want to remove zadic, that is a different story. I will have a look if wdi-simple fits my needs, too.

pbatard, 2020-05-30:
> But, if you want to remove zadic, that is a different story. I will have a look if wdi-simple fits my needs, too.

Please do. If you see the same issue with `wdi-simple` I'll try to have a closer look at it.

spiro-trikaliotis, 2020-05-31:
The problem with zadic is still there. However, wdi-simple fulfills my needs, too, so I will close this issue, knowing that you will remove the erroneous zadic.exe.
Thank you for your support!

closed 2020-05-31 by spiro-trikaliotis

pbatard, 2020-05-31:
Great! One more reason to remove zadic, as I'd rather not have other users waste their time with it like you did. Thanks for the update.