pbatard/libwdi #188: Empty device name

asesidaa, 2020-06-03:
I am trying to install WinUSB for Logan Nano, but when i tried to list all the device, the device name is empty and the driver cannot be installed properly. The device name is correct in device manager
[image]

[image]

pbatard, 2020-06-03:
_Options_ &rarr; _List All Devices_ and _Options_ &rarr; _Advanced Mode_, then save the log and copy/paste it here.

asesidaa, 2020-06-03:
Zadig 2.5.730
Windows 10 64-bit (Build 19041)
ini file 'zadig.ini' not found in 'C:\Users\lg\Downloads\Programs' - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
15 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1038&PID_1248&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_01&SubClass_01&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.1
libwdi:debug [wdi_create_list] usbaudio USB device (0): USB\VID_1038&PID_1248&MI_00\8&1B3F8C09&1&0000
libwdi:debug [wdi_create_list] Device description: 'SteelSeries Siberia 840 (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C52B&REV_2407&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.1
libwdi:debug [wdi_create_list] HidUsb USB device (5): USB\VID_046D&PID_C52B&MI_01\7&2BA19684&2&0001
libwdi:debug [wdi_create_list] Device description: 'USB Receiver (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1038&PID_1240&REV_0600&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.1
libwdi:debug [wdi_create_list] HidUsb USB device (6): USB\VID_1038&PID_1240&MI_01\8&26C42363&0&0001
libwdi:debug [wdi_create_list] Device description: 'SteelSeries Siberia 840 (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_10A5&PID_0007&REV_0200&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 202.23.5.100
libwdi:debug [wdi_create_list] WUDFRd USB device (9): USB\VID_10A5&PID_0007&MI_00\6&1C1E4ADF&0&0000
libwdi:debug [wdi_create_list] Device description: 'FPC Sensor Controller L:0004 FW:202.23.5.97 (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0B95&PID_1790&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_FF&Prot_00
libwdi:debug [wdi_create_list] Driver version: 1.20.3.0
libwdi:debug [wdi_create_list] AX88179 USB device (10): USB\VID_0B95&PID_1790\00000000000171
libwdi:debug [wdi_create_list] Device description: 'AX88179'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_562E&REV_0011&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.1
libwdi:debug [wdi_create_list] usbvideo USB device (11): USB\VID_0BDA&PID_562E&MI_00\6&33C9A73F&1&0000
libwdi:debug [wdi_create_list] Device description: 'LG Camera (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C52B&REV_2407&MI_02
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.1
libwdi:debug [wdi_create_list] HidUsb USB device (12): USB\VID_046D&PID_C52B&MI_02\7&2BA19684&2&0002
libwdi:debug [wdi_create_list] Device description: 'USB Receiver (Interface 2)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8087&PID_0A2B&REV_0010
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_E0&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 21.90.0.4
libwdi:debug [wdi_create_list] BTHUSB USB device (13): USB\VID_8087&PID_0A2B\5&130477CA&0&5
libwdi:debug [wdi_create_list] Device description: '英特尔(R) 无线 Bluetooth(R)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_10A5&PID_0007&REV_0200&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 202.23.5.97
libwdi:debug [wdi_create_list] WUDFRd USB device (15): USB\VID_10A5&PID_0007&MI_01\6&1C1E4ADF&0&0001
libwdi:debug [wdi_create_list] Device description: 'FPC Sensor Controller FW Update (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0C45&PID_760A&REV_0105&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.1
libwdi:debug [wdi_create_list] HidUsb USB device (17): USB\VID_0C45&PID_760A&MI_01\7&F9246F6&1&0001
libwdi:debug [wdi_create_list] Device description: 'USB Keyboard (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1038&PID_1248&REV_0100&MI_03
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.1
libwdi:debug [wdi_create_list] HidUsb USB device (18): USB\VID_1038&PID_1248&MI_03\8&1B3F8C09&1&0003
libwdi:debug [wdi_create_list] Device description: 'Hid Interface (Interface 3)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0C45&PID_760A&REV_0105&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.1
libwdi:debug [wdi_create_list] HidUsb USB device (19): USB\VID_0C45&PID_760A&MI_00\7&F9246F6&1&0000
libwdi:debug [wdi_create_list] Device description: 'USB Keyboard (Interface 0)'
**libwdi:debug [wdi_create_list] Hardware ID: USB\VID_28E9&PID_0189&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FE&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 13.57.56.210
libwdi:debug [wdi_create_list] GDDFUDriver USB device (22): USB\VID_28E9&PID_0189\5&130477CA&0&1
libwdi:debug [wdi_create_list] Device description: ''**
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C52B&REV_2407&MI_00
[... cut ...]
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.1
libwdi:debug [wdi_create_list] HidUsb USB device (24): USB\VID_1038&PID_1240&MI_00\8&26C42363&0&0000
libwdi:debug [wdi_create_list] Device description: 'SteelSeries Siberia 840 (Interface 0)'

There is a device description in device manager.
[image]

I don't know whether no device description will make something go wrong, but if I continue and edit a new name, the device will become an unknown device in device manager.

pbatard, 2020-06-03:
> Device description: ''

The device is there, and you can use it. It just doesn't have a name.

In other words, the issue is that the people who manufactured your device set the name descriptor to the empty string, which I would think goes against USB recommendations.

If you click _Edit_ you will be able to enter whatever name you chose, and regardless of whether the name is empty or not, you can install the driver with Zadig.

I guess I __could__ add some code in libwdi that creates a fake name (e.g. "Unnamed USB Device") when the descriptor is empty, but that sounds like a waste of time, when the issue, really, is that the people who designed your device did a bad job with their descriptors, and having an empty name doesn't prevent Zadig operations.

Since there is no actual underlying problem (Zadig __does__ report the exact name it got from the device) and you should be able to accry out regular Zadig operations regardless , I will therefore close this issue.

pbatard, 2020-06-03:
Also note that Device Manager probably creates its own description, or picks it up from a different descriptor, and I am not going to reverse the Device Manager code to figure out what it does to fill its description, so that Zadig can match it. Unless you are seeing problems down the line, having a blank name in Zadig is not an issue.

closed 2020-06-03 by pbatard