pbatard/libwdi #191: embedder.exe fails when USER_DIR in config.h is defined

JSB2500, 2020-06-09:
Version: v1.3.1

When USER_DIR in config.h is defined, embedded.exe reports this error: "EMBED  C:\...\libwdi\libwdi\libusb0.dll embedder : error: Could not open file 'C:\...\libwdi\libwdi\libusb0.dll'."

Suggested fix: Change line 298 of embedder.c from:
`embeddable[i].reuse_last = 0;`
to:
`embeddable[i].reuse_last = embeddable_fixed[i].reuse_last;`

mcuee, 2021-07-07:
Yes I can reproduce this issue with latest git.

```
Build started...
1>------ Build started: Project: embedder, Configuration: Release Win32 ------
1>embedder.c
1>Generating code
1>4 of 18 functions (22.2%) were compiled, the rest were copied from previous compilation.
1>  2 functions were new in current compilation
1>  1 functions had inline decision re-evaluated but remain unchanged
1>Finished generating code
1>embedder.vcxproj -> C:\work\libusb\libwdi\libwdi\embedder.exe
2>------ Build started: Project: libwdi (static), Configuration: Release Win32 ------
3>------ Skipped Build: Project: libwdi (dll), Configuration: Release Win32 ------
3>Project not selected to build for this solution configuration 
2>Embedding binary resources
2>unable to stat 'C:\work\libusb\libwdi\libwdi\libusb0.dll' - assuming rebuild needed
2>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x86\WdfCoInstaller01011.dll (2021.01.14)
2>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x86\winusbcoinstaller2.dll (2021.01.14)
2>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x64\WdfCoInstaller01011.dll (2021.01.14)
2>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x64\winusbcoinstaller2.dll (2021.01.14)
2>  EMBED  C:\libusb-win32-1.2.6.0-bin\bin\x86\libusb0_x86.dll (2021.07.07)
2>  EMBED  C:\libusb-win32-1.2.6.0-bin\bin\x86\install-filter.exe (2021.07.07)
2>embedder : error : Could not open file 'C:\work\libusb\libwdi\libwdi\libusb0.dll'.
2>  EMBED  C:\work\libusb\libwdi\libwdi\libusb0.dll
2>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(145,5): error MSB3073: The command "cd C:\work\libusb\libwdi\libwdi\.msvc\\..
2>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(145,5): error MSB3073: embedder embedded.h
2>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(145,5): error MSB3073: :VCEnd" exited with code 1.
2>Done building project "libwdi_static.vcxproj" -- FAILED.
========== Build: 1 succeeded, 1 failed, 7 up-to-date, 1 skipped ==========

```

mcuee, 2021-07-07:
And I can confirm the fix works. 
```
Build started...
1>------ Build started: Project: embedder, Configuration: Release Win32 ------
1>embedder.c
1>Generating code
1>2 of 18 functions (11.1%) were compiled, the rest were copied from previous compilation.
1>  0 functions were new in current compilation
1>  0 functions had inline decision re-evaluated but remain unchanged
1>Finished generating code
1>embedder.vcxproj -> C:\work\libusb\libwdi\libwdi\embedder.exe
2>------ Build started: Project: libwdi (static), Configuration: Release Win32 ------
3>------ Skipped Build: Project: libwdi (dll), Configuration: Release Win32 ------
3>Project not selected to build for this solution configuration 
2>Embedding binary resources
2>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x86\WdfCoInstaller01011.dll (2021.01.14)
2>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x86\winusbcoinstaller2.dll (2021.01.14)
2>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x64\WdfCoInstaller01011.dll (2021.01.14)
2>  EMBED  C:\Program Files (x86)\Windows Kits\10\redist\wdf\x64\winusbcoinstaller2.dll (2021.01.14)
2>  EMBED  C:\libusb-win32-1.2.6.0-bin\bin\x86\libusb0_x86.dll (2021.07.07)
2>  EMBED  C:\libusb-win32-1.2.6.0-bin\bin\x86\install-filter.exe (2021.07.07)
2>  EMBED  C:\libusb-win32-1.2.6.0-bin\bin\x86\libusb0.sys (2021.07.07)
2>  EMBED  C:\libusb-win32-1.2.6.0-bin\bin\amd64\libusb0.dll (2021.07.07)
2>  EMBED  C:\libusb-win32-1.2.6.0-bin\bin\amd64\libusb0.sys (2021.07.07)
2>  EMBED  C:\libusb-win32-1.2.6.0-bin\bin\amd64\install-filter.exe (2021.07.07)
2>  EMBED  C:\libusb-win32-1.2.6.0-bin\installer_license.txt (2021.07.07)
2>  EMBED  C:\libusbK-3.0.8.0-bin\bin\sys\x86\libusbK.sys (2021.07.07)
2>  EMBED  C:\libusbK-3.0.8.0-bin\bin\dll\x86\libusbK.dll (2021.07.07)
2>  EMBED  C:\libusbK-3.0.8.0-bin\bin\sys\amd64\libusbK.sys (2021.07.07)
2>  EMBED  C:\libusbK-3.0.8.0-bin\bin\dll\amd64\libusbK.dll (2021.07.07)
2>  EMBED  C:\work\libusb\libwdi\Win32\Release\helper\installer_x86.exe (2021.07.07)
2>  EMBED  C:\work\libusb\libwdi\x64\Release\helper\installer_x64.exe (2021.07.07)
2>  EMBED  C:\work\libusb\libwdi\libwdi\winusb.inf.in (2021.06.09)
2>  EMBED  C:\work\libusb\libwdi\libwdi\libusb0.inf.in (2021.06.09)
2>  EMBED  C:\work\libusb\libwdi\libwdi\libusbk.inf.in (2021.06.09)
2>  EMBED  C:\work\libusb\libwdi\libwdi\usbser.inf.in (2021.06.09)
2>  EMBED  C:\work\libusb\libwdi\libwdi\winusb.cat.in (2021.06.09)
2>  EMBED  C:\work\libusb\libwdi\libwdi\libusb0.cat.in (2021.06.09)
2>  EMBED  C:\work\libusb\libwdi\libwdi\libusbk.cat.in (2021.06.09)
2>  EMBED  C:\work\libusb\libwdi\libwdi\usbser.cat.in (2021.06.09)
2>  EMBED  C:\work\cyusb\driver\bin\Win10\x64\cyusb3.cat (2021.07.07)
2>  EMBED  C:\work\cyusb\driver\bin\Win10\x64\cyusb3.inf (2021.07.07)
2>  EMBED  C:\work\cyusb\driver\bin\Win10\x64\cyusb3.pdb (2021.07.07)
2>  EMBED  C:\work\cyusb\driver\bin\Win10\x64\cyusb3.sys (2021.07.07)
2>  EMBED  C:\work\cyusb\driver\bin\Win10\x86\cyusb3.cat (2021.07.07)
2>  EMBED  C:\work\cyusb\driver\bin\Win10\x86\cyusb3.inf (2021.07.07)
2>  EMBED  C:\work\cyusb\driver\bin\Win10\x86\cyusb3.pdb (2021.07.07)
2>  EMBED  C:\work\cyusb\driver\bin\Win10\x86\cyusb3.sys (2021.07.07)
2>libwdi.c
2>libwdi_dlg.c
2>logging.c
2>pki.c
2>tokenizer.c
2>libwdi_static.vcxproj -> C:\work\libusb\libwdi\Win32\Release\lib\libwdi.lib
4>------ Build started: Project: wdi-simple, Configuration: Release Win32 ------
5>------ Build started: Project: zadic, Configuration: Release Win32 ------
6>------ Build started: Project: zadig, Configuration: Release Win32 ------
4>Generating code
5>Generating code
6>Generating code
4>37 of 93 functions (39.8%) were compiled, the rest were copied from previous compilation.
4>  0 functions were new in current compilation
4>  1 functions had inline decision re-evaluated but remain unchanged
4>Finished generating code
5>12 of 90 functions (13.3%) were compiled, the rest were copied from previous compilation.
5>  0 functions were new in current compilation
5>  2 functions had inline decision re-evaluated but remain unchanged
5>Finished generating code
6>13 of 253 functions ( 5.1%) were compiled, the rest were copied from previous compilation.
6>  0 functions were new in current compilation
6>  4 functions had inline decision re-evaluated but remain unchanged
4>wdi-simple.vcxproj -> C:\work\libusb\libwdi\Win32\Release\examples\wdi-simple.exe
6>Finished generating code
5>zadic.vcxproj -> C:\work\libusb\libwdi\Win32\Release\examples\zadic.exe
6>zadig.vcxproj -> C:\work\libusb\libwdi\Win32\Release\examples\zadig.exe
========== Build: 5 succeeded, 0 failed, 4 up-to-date, 1 skipped ==========

```

mcuee, 2021-07-07:
However, Zadig will often crash with the above configuration, probably due to the embedded Cypress USB driver.

mcuee, 2021-07-07:
I just tried with one build which remove WinUSB/libusbk/libusb-win32 but only with Cypress, with or without the changes, Zadig will crash. So I guess I do not know how to use the option of user defined driver yet. IMHO this is not well documented.

mcuee, 2021-07-07:
It seems to me I need to do something to create a template inf and cat file.

mcuee, 2021-07-20:
It seems to me this is not a real issue but rather lack of documentation to use the custom driver file. I was trying to embed cyusb3 but failed.

mcuee, 2021-07-20:
One simple solution is to use dpscat.exe from libusbk to sign a custom driver package, then you can use Windows Device Manager to install the file. Windows 10 will pop up a warning message but the installation will be successful.

Example run log:
```
C:\work\libusb\libusbK-3.1.0.0-bin\bin\exe\amd64\cyusb3\Win10\x64> .\dpscat.exe
Copyright(c) 2012 Travis Lee Robinson. (DUAL BSD/GPL)
Portions Copyright(c) Pete Batard. (LGPL)
C:\work\libusb\libusbK-3.1.0.0-bin\bin\exe\amd64\cyusb3\Win10\x64>
Actual section to install: Device.NTamd64
Found Hwid: USB\VID_04B4&PID_00FC
Found Hwid: USB\VID_04B4&PID_00FE
Found Hwid: USB\VID_04B4&PID_00FD
Found Hwid: USB\VID_04B4&PID_00FB
Hash calculated for: .\cyusb3.inf
Using INF guid..
Hash added..
Catalog file 'CYUSB3.cat' created..
Deleted existing certificate: CN="Cypress (cyusb3.inf) [Self]" (Root store)
Deleted existing certificate: CN="Cypress (cyusb3.inf) [Self]" (TrustedPublisher store)
Set Enhanced Key Usage, URL and CPS..
Created new key container..
Generated new keypair.
Created new self-signed certificate: CN="Cypress (cyusb3.inf) [Self]"
Added CN="Cypress (cyusb3.inf) [Self]" certificate to 'Root' and 'TrustedPublisher' stores..
Signed file: C:\work\libusb\libusbK-3.1.0.0-bin\bin\exe\amd64\cyusb3\Win10\x64\CYUSB3.cat
Deleted private key..
Success!
```

mcuee, 2021-07-20:
From Travis -- but I have not tried it yet.

>You can also use some of the files from a libusbK driver package to create a self-contained "InstallDriver.exe"
> 1) Use the 7za.exe included in the lbusbK driver package to zip all of your driver files to _DriverFiles.7z (You must use the 7za.exe from the libusbK package)
> 2) Copy the 7zDP_LZMA.* files, dpinst* files, and dpscat.exe into your cypress driver folder
> 3) Execute: "COPY /B "7zDP_LZMA.sfx"+"7zDP_LZMA.cfg"+"_DriverFiles.7z" "InstallDriver.exe"

mcuee, 2021-07-20:
In reality, once you use the dpscat to build the driver package, you can use different method to create the driver installer, not necessary to follow the above libusbk method or the method in libwdi Zadig or libwdi wid-simple example.

closed 2021-09-04 by pbatard

mcuee, 2023-07-20:
Example Installation using Zadig.

[image]

commit e1257a86 (2021-09-04): [core] fix errors when providing a user driver

* With thanks to JSB2500 for the fix suggestion
* Closes #191
* Also fix a CodeQL reported issue