pbatard/libwdi #194: libusb1 is not supported

jxltom, 2020-06-30:
The libusb1 in https://github.com/libusb/libusb is the latest version of libusb, however only libusb-win32 in https://sourceforge.net/projects/libusb-win32/ which is libusb 0.1 is supported. I'm curious why? Many thanks to the explanation!

pbatard, 2020-06-30:
drivers != libraries.

libwdi install __drivers__
Libusb is a __library__.

Libusb 0.1 (a.k.a. libusbwin32) happens to use a __custom driver__ (`libusb0.sys`), so libwdi enables the installation of that driver
libusb 1.0 happens to use __Microsoft's WinUSB driver__, so libwdi enables the installation of that driver.

There's no mystery at all there, you're missing the point that a library may not used a driver that is named after it.

closed 2020-06-30 by pbatard

jxltom, 2020-06-30:
Thanks for the clarification man!