pbatard/libwdi #196: wdi-simple - don't install driver by default

benlye, 2020-07-13:
We're using wdi-simple.exe to install device drivers for an open source project.  It works great, but we have problems from time to time with different AV products falsely detecting wdi-simple as malicious.  I'm trying to counter this by making it default to showing the usage rather than installing a device driver when it's run with no arguments.

My theory (hope) is that this will stop the AV products thinking bad things are happening when they run the executable in a sandbox because nothing will happen (unlike today when it immediately does scary system stuff).

Would you be interested in accepting this as a PR?  It's a super-simple change:
https://github.com/benlye/libwdi/commit/5e438ebb550a1da4aa6608bc8579031573b87e18

I'm assuming that people who want to automatically install their own driver are customizing the build anyway (because they don't really want the XBox driver), so could easily remove/disable it at the same time as they customize the default driver.  In our use case we need to install drivers for two different devices, so having a single driver auto-installed by default doesn't work for us anyway - we just run two separate wdi-simple.exe command lines from our package.

pbatard, 2020-07-13:
Sorry but no. I am __exceedingly__ tired of AV vendors flagging the software I produce, and working around these issues rather than trying to get the problem at the source, by having AV vendors stop flagging everything and anything as a false positive is not the way to go.

Please complain directly to the AV vendor. It will be a lot more helpful than adding workaround after workaround in applications that are perfectly legitimate and should not be flagged in the first place.

closed 2020-07-13 by pbatard

benlye, 2020-07-13:
Understood, and I know what you mean about being tired.  I'm already doing that (complaining to the AV vendors) as well for the version of wdi-simple.exe I've compiled for our project (with some success).  I just thought this step might make some of those false positives less likely to occur in the first place.

Thanks for the software.

pbatard, 2020-07-13:
The problem is that there is absolutely zero reason for these false positives to appear. Unless you are effectively creating malware, noone should have to concern themselves about writing code in a manner that is "less likely" to trigger a false positive. If we start to work around these issues in the source, it will just embolden AV vendors into pushing even more crap like this. And that is the reason I have to insist that this gets resolved at the source, where it should be. In the long run, it will be a lot more beneficial for everyone to get AV vendors to get their act together, than let them abuse their position to __bully__ more and more developers into wasting time doing stuff they should not have to do.