pbatard/libwdi #197: Zadig: Driver Installation: FAILED (Other error)

SingL3, 2020-08-27:
Hi, I am trying to install driver for a device and I got the below message:

```
Zadig 2.5.730
Windows 10 64-bit (Build 18363)
ini file 'zadig.ini' not found in 'D:\Downloads' - default parameters will be used
default driver set to 'WinUSB'
2 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0525&PID_A4A0&REV_0200&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_f0&Prot_01
libwdi:debug [wdi_create_list] Driverless USB device (1): USB\VID_0525&PID_A4A0&MI_01\6&3113FEE1&0&0001
libwdi:debug [wdi_create_list] could not read device description for 1: [#0000000D] ������Ч��
libwdi:debug [wdi_create_list] Device description: 'Unknown Device #1 (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0525&PID_A4A0&REV_0200&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_f0&Prot_00
libwdi:debug [wdi_create_list] Driverless USB device (3): USB\VID_0525&PID_A4A0&MI_00\6&3113FEE1&0&0000
libwdi:debug [wdi_create_list] Device description: 'iAP Interface (Interface 0)'
Using inf name: iAP_Interface_(Interface_0).inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\admin\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\admin\usb_driver\iAP_Interface_(Interface_0).inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\admin\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\admin\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'iap_interface_(interface_0).inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\admin\usb_driver\iap_interface_(interface_0).inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\admin\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\admin\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\admin\usb_driver\iAP_Interface_(Interface_0).cat'
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_0525&PID_A4A0&MI_00 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_0525&PID_A4A0&MI_00 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\VID_0525&PID_A4A0&MI_00 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\VID_0525&PID_A4A0&MI_00 (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\VID_0525&PID_A4A0&MI_00 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\admin\usb_driver\iAP_Interface_(Interface_0).cat'
libwdi:info [SelfSignFile] Successfully deleted private key
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter iAP_Interface_(Interface_0).inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0525&PID_A4A0&MI_00\6&3113FEE1&0&0000'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_0525&PID_A4A0&REV_0200&MI_00'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-3156512955-109854751-1387415816-1001'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
Driver Installation: FAILED (Other error)
```

Do you have any suggestion for me?

pbatard, 2020-09-20:
Can you zip and attach your full `C:\Windows\INF\setupapi.dev.log`? Unfortunately, it looks like the Zadig installer was unable to read its data (there should have been excerpt from this log between `libwdi:debug [installer process] syslog reader thread started` and `Driver Installation: FAILED (Other error)` but for some reasons they did not appear in your output).

Without seeing `setupapi.dev.log` it's impossible to tell what happened.

pbatard, 2020-11-30:
No update &rarr; Closing.

closed 2020-11-30 by pbatard