pbatard/libwdi #198: Zadig: Failed (Administrator Privileges)

Quortdal, 2020-08-29:
Hello,
Every time I try to install any of the available drivers, it fails stating, "Unable to run process with required administrative privileges." I'm attempting the install from the default admin account, though my account also has full admin access. Still nothing. I've tried everything I can think of. For context, there is no preinstalled driver on this PCB, so I can't flash firmware without the driver.
Any suggestions would be greatly appreciated. Thank you!

pbatard, 2020-08-29:
Please provide the full log from Zadig (Options &rarr; Advanced Mode).

I simply cannot help you without seeing the full log.

pbatard, 2020-09-20:
No log &rarr; Closed.

closed 2020-09-20 by pbatard