pbatard/libwdi #200: Zadig: Driver Installation: Installtion Cancel by User

heliarmk, 2020-09-22:
Hi, I am trying to install driver for a device and I got the below message:
```
Using inf name: ATm32U4DFU.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\guankai1\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\guankai1\usb_driver\ATm32U4DFU.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\guankai1\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\guankai1\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'atm32u4dfu.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\guankai1\usb_driver\atm32u4dfu.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\guankai1\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\guankai1\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\guankai1\usb_driver\ATm32U4DFU.cat'
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:warning [CreateSelfSignedCert] Could not obtain a key container: The requested key container is corrupted.
libwdi:warning [wdi_prepare_driver] could not sign cat file
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter ATm32U4DFU.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_03EB&PID_2FF4\1.0.0'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_03EB&PID_2FF4&REV_0000'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-1380817616-3362833225-652976467-44806'
libwdi:debug [installer process] using syslog 'C:\Windows\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_03EB&PID_2FF4&REV_0000 - please wait...
libwdi:debug [syslog]    cmd: "C:\Users\guankai1\usb_driver\installer_x64.exe" "ATm32U4DFU.inf"
libwdi:debug [syslog] eck]
libwdi:debug [syslog] >>>  Section start 2020/09/22 09:40:55.601
libwdi:debug [syslog]  Policy Check]
libwdi:debug [syslog] >>>  Section start 2020/09/22 09:40:23.873
libwdi:debug [syslog] 1\usb_driver\installer_x64.exe" "ATm32U4DFU.inf"
libwdi:debug [syslog] d: C:\Windows\system32\svchost.exe -k netsvcs -p -s DsmSvc
libwdi:debug [syslog] >>  Section start 2020/09/22 09:23:49.053
libwdi:debug [syslog] <<<  Section end 2020/09/22 09:25:10.802
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Installation Restrictions Policy Check]
libwdi:debug [syslog] >>>  Section start 2020/09/22 09:24:20.260
libwdi:debug [syslog] <<<  Section end 2020/09/22 09:27:58.029
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_03EB&PID_2FF4&REV_0000]
libwdi:debug [syslog] >>>  Section start 2020/09/22 09:24:36.912
libwdi:debug [syslog]       cmd: "C:\Users\guankai1\usb_driver\installer_x64.exe" "ATm32U4DFU.inf"
libwdi:debug [syslog] 020/09/22 08:46:32.264
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Delete Device - SWD\PRINTENUM\{F74F92AF-0757-4C1A-8AFE-D8F1436E6F8A}]
libwdi:debug [syslog] >>>  Section start 2020/09/22 09:07:49.763
libwdi:debug [syslog]       cmd: C:\Windows\System32\spoolsv.exe
libwdi:debug [syslog] !    dvi: Access denied from Query and Remove
libwdi:debug [syslog] <<<  Section end 2020/09/22 09:07:50.037
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Delete Device - SWD\PRINTENUM\{5B2E4CC7-7049-4D88-80B4-097F6247ACFA}]
libwdi:debug [syslog] >>>  Section start 2020/09/22 09:07:50.080
libwdi:debug [syslog]       cmd: C:\Windows\System32\spoolsv.exe
libwdi:debug [syslog] !    dvi: Access denied from Query and Remove
libwdi:debug [syslog] <<<  Section end 2020/09/22 09:07:50.103
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Delete Device - SWD\PRINTENUM\{07EA79B1-F5C5-4D12-8A93-2DD1123FD57A}]
libwdi:debug [syslog] >>>  Section start 2020/09/22 09:07:50.143
libwdi:debug [syslog]       cmd: C:\Windows\System32\spoolsv.exe
libwdi:debug [syslog] <<<  Section end 2020/09/22 09:07:50.172
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
[... 198 lines cut ...]
libwdi:debug [syslog] !    dvi: Access denied from Query and Remove
libwdi:debug [syslog] <<<  Section end 2020/09/22 09:07:50.296
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Delete Device - SWD\PRINTENUM\{C6947959-25CC-4905-BE82-4E424895A327}]
libwdi:debug [syslog] >>>  Section start 2020/09/22 09:07:50.332
libwdi:debug [syslog]       cmd: C:\Windows\System32\spoolsv.exe
libwdi:debug [syslog] !    dvi: Access denied from Query and Remove
libwdi:debug [syslog] <<<  Section end 2020/09/22 09:07:50.351
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_03EB&PID_2FF4&REV_0000]
libwdi:debug [syslog] >>>  Section start 2020/09/22 09:19:45.417
libwdi:debug [syslog]       cmd: "C:\Users\guankai1\usb_driver\installer_x64.exe" "ATm32U4DFU.inf"
libwdi:debug [process_message] switching timeout back to finite
libwdi:debug [installer process] Invalid CAT file signature or operation cancelled by user.
libwdi:debug [process_message] installer process completed
Driver Installation: Cancelled by User
1 device found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_03EB&PID_2FF4&REV_0000
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_00&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driverless USB device (3): USB\VID_03EB&PID_2FF4\1.0.0
libwdi:debug [wdi_create_list] Device description: 'ATm32U4DFU'
0 devices found.
0 devices found.
```
I found the zadig want to install the driver to a printer, but I want to install the driver for Atm32U4. Do you have any suggestion for me?

heliarmk, 2020-09-22:
After I uninstall the printer in my win. The zadig shows this error.
```
Using inf name: ATm32U4DFU.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\guankai1\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\guankai1\usb_driver\ATm32U4DFU.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\guankai1\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\guankai1\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'atm32u4dfu.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\guankai1\usb_driver\atm32u4dfu.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\guankai1\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\guankai1\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\guankai1\usb_driver\ATm32U4DFU.cat'
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:warning [CreateSelfSignedCert] Could not obtain a key container: The requested key container is corrupted.
libwdi:warning [wdi_prepare_driver] could not sign cat file
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter ATm32U4DFU.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_03EB&PID_2FF4\1.0.0'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_03EB&PID_2FF4&REV_0000'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-1380817616-3362833225-652976467-44806'
libwdi:debug [installer process] using syslog 'C:\Windows\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_03EB&PID_2FF4&REV_0000 - please wait...
libwdi:debug [syslog]     - 0x00100004
libwdi:debug [syslog]      inf:                          {Install from INF Section - exit(0x00000000)} 14:26:55.808
libwdi:debug [syslog]      dvi:                     {Install DEVICE exit (0x00000000)}
libwdi:debug [syslog]      dvi:                     Install Device: Configuring device class. 14:26:55.812
libwdi:debug [syslog]      dvi:                     Install Device: Configuring device class completed. 14:26:55.812
libwdi:debug [syslog]      dvi:                     Device Status: 0x01802400, Problem: 0x1 (0xc0000493)
libwdi:debug [syslog]      dvi:                     Install Device: Starting device 'USB\VID_046D&PID_C52B&MI_00\6&1E22DE71&0&0000'. 14:26:55.815
libwdi:debug [syslog]      dvi:                     Install Device: Starting device completed. 14:26:55.847
libwdi:debug [syslog]      dvi:                Default installer: Exit
libwdi:debug [syslog]      dvi:           {DIF_INSTALLDEVICE - exit(0x00000000)} 14:26:55.848
libwdi:debug [syslog]      dvi:           {DIF_NEWDEVICEWIZARD_FINISHINSTALL} 14:26:55.849
libwdi:debug [syslog]      dvi:                CoInstaller 1: Enter 14:26:55.849
libwdi:debug [syslog]      dvi:                CoInstaller 1: Exit
libwdi:debug [syslog]      dvi:                Default installer: Enter 14:26:55.862
libwdi:debug [syslog]      dvi:                Default installer: Exit
libwdi:debug [syslog]      dvi:           {DIF_NEWDEVICEWIZARD_FINISHINSTALL - exit(0xe000020e)} 14:26:55.863
libwdi:debug [syslog]      dvi:      {Install Device - exit(0x00000000)} 14:26:55.864
libwdi:debug [syslog]      dvi: {Core Device Install - exit(0x00000000)} 14:26:55.865
libwdi:debug [syslog]      dvi: {DIF_DESTROYPRIVATEDATA} 14:26:55.865
libwdi:debug [syslog]      dvi:      CoInstaller 1: Enter 14:26:55.866
libwdi:debug [syslog]      dvi:      CoInstaller 1: Exit
libwdi:debug [syslog]      dvi:      Default installer: Enter 14:26:55.867
libwdi:debug [syslog]      dvi:      Default installer: Exit
libwdi:debug [syslog]      dvi: {DIF_DESTROYPRIVATEDATA - exit(0xe000020e)} 14:26:55.868
libwdi:debug [syslog] <<<  Section end 2020/09/22 14:26:55.881
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_03EB&PID_2FF4&REV_0000]
libwdi:debug [syslog] >>>  Section start 2020/09/22 14:27:17.144
libwdi:debug [syslog]       cmd: "C:\Users\guankai1\usb_driver\installer_x64.exe" "ATm32U4DFU.inf"
libwdi:debug [syslog] ice: Configuring device class completed. 14:26:55.812
libwdi:debug [syslog]      dvi:                     Device Status: 0x01802400, Problem: 0x1 (0xc0000493)
libwdi:debug [syslog]      dvi:                     Install Device: Starting device 'USB\VID_046D&PID_C52B&MI_00\6&1E22DE71&0&0000'. 14:26:55.815
libwdi:debug [syslog]      dvi:                     Install Device: Starting device completed. 14:26:55.847
libwdi:debug [syslog]      dvi:                Default installer: Exit
[... cut ...]
libwdi:debug [installer process] Invalid CAT file signature or operation cancelled by user.
libwdi:debug [process_message] installer process completed
Driver Installation: Cancelled by User
1 device found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_03EB&PID_2FF4&REV_0000
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_00&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driverless USB device (2): USB\VID_03EB&PID_2FF4\1.0.0
libwdi:debug [wdi_create_list] Device description: 'ATm32U4DFU'
0 devices found.
0 devices found.
```

pbatard, 2020-11-30:
> ```
> libwdi:warning [CreateSelfSignedCert] Could not obtain a key container: The requested key container is corrupted.
> libwdi:warning [wdi_prepare_driver] could not sign cat file
> ```

This is your problem. Your system cannot sign `.cat` files. You need to figure out why that is the case (probably your Windows cryptographic services are broken in some way) because this is not a Zadig/libwdi problem.

On a proper Windows installation, these services will work, so if they don't, then it means that your altered your Windows installation in some way, and it is up to you to fix that.

closed 2020-11-30 by pbatard