pbatard/libwdi #201: My main keyboard has stopped working

Chart111122, 2020-09-28:
The keyboard I use is the HyperX alloy origins core and I was trying to get OpenRGB to detect my keyboard, and this was what I was directed to. It said to select the inference with the highest number. Which on the image they gave was lowest (so there was 1 and 2 and they chose 1) seeing that I chose 0. And now I cannot type with the keyboard but the lighting effects are still working. I have tried to look in the FAQ to see if there is a way to undo the changes, I tried the one that was there but it didn't do anything maybe because I couldn't find the right device.

I was wondering if I could undo the things that I did, or have I just bricked my keyboard?

pbatard, 2020-09-28:
https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it

Chart111122, 2020-09-28:
Sorry for taking your time.

I happened to find the right drivers to uninstall, it wasn't in the **_Universal Serial Bus devices_** it was in the **_Human Interference Devices_**.

Its one of the **_HID-compliant vendor-defined devices_**.
 
[image]

closed 2020-09-29 by pbatard