pbatard/libwdi #202: HackRF installation successful but driver not working

alexwahl, 2020-11-09:
Hello!
I have a problem with the installation of Zadig for a HackRF one. 
My Setup: Windows 10 Pro (10.0.18363 Build 18363) in a virtualbox on a Arch Linux host (kernel 5.4.75-1-lts. 
The HackRF is attached to the virtual machine and shows up in device manager.
The installation with Zadig reports "successful"

```

 Using inf name: HackRF_One.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\DH2AW\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\DH2AW\usb_driver\HackRF_One.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\DH2AW\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\DH2AW\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'hackrf_one.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\DH2AW\usb_driver\hackrf_one.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\DH2AW\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\DH2AW\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\DH2AW\usb_driver\HackRF_One.cat'
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_1D50&PID_6089 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] Deleted existing certificate 'CN=USB\VID_1D50&PID_6089 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\VID_1D50&PID_6089 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\VID_1D50&PID_6089 (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\VID_1D50&PID_6089 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\DH2AW\usb_driver\HackRF_One.cat'
libwdi:info [SelfSignFile] Successfully deleted private key
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter HackRF_One.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_1D50&PID_6089\0000000000000000A06063C82B11391B'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_1D50&PID_6089&REV_0102'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-1694127139-1974405403-3936319245-1002'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_1D50&PID_6089&REV_0102 - please wait...
libwdi:debug [syslog] ce.inf"
libwdi:debug [syslog] er\installer_x64.exe" "WinUSB_Generic_Device.inf"
libwdi:debug [syslog] UpdateDriverForPlugAndPlayDevices) - USB\VID_1D50&PID_6089&REV_0102]
libwdi:debug [syslog] >>>  Section start 2020/11/09 20:02:58.154
libwdi:debug [syslog]       cmd: "C:\Users\DH2AW\usb_driver\installer_x64.exe" "HackRF_One.inf"
libwdi:debug [syslog] sers\DH2AW\usb_driver\installer_x64.exe" "HackRF_One.inf"
libwdi:debug [syslog] /09 19:56:12.140
libwdi:debug [syslog]     sto:                     {Core Driver Package Delete: exit(0x00000000)} 19:43:15.902
libwdi:debug [syslog]      sto:                {Unstage Driver Package: exit(0x00000000)} 19:43:15.902
libwdi:debug [syslog]      dvs:           {Driver Setup Delete Driver Package - exit (0x00000000)} 19:43:15.902
libwdi:debug [syslog]      dvs:      {DrvSetupUninstallDriver - exit(00000000)}
libwdi:debug [syslog]      inf: {SetupUninstallOEMInf exit (0x00000000)}
libwdi:debug [syslog] <<<  Section end 2020/11/09 19:43:15.920
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
libwdi:debug [syslog] [Boot Session: 2020/11/09 19:53:52.500]
libwdi:debug [syslog] >>>  [SetupCopyOEMInf - C:\Users\DH2AW\usb_driver\WinUSB_Generic_Device.inf]
libwdi:debug [syslog] >>>  Section start 2020/11/09 19:55:31.843
libwdi:debug [syslog]       cmd: "C:\Users\DH2AW\usb_driver\installer_x64.exe" "WinUSB_Generic_Device.inf"
libwdi:debug [syslog]  C:\Users\DH2AW\usb_driver\WinUSB_Generic_Device.inf]
libwdi:debug [syslog] >>>  Section start 2020/11/09 19:55:31.843
libwdi:debug [syslog]       cmd: "C:\Users\DH2AW\usb_driver\installer_x64.exe" "WinUSB_Generic_Device.inf"
libwdi:debug [syslog] er_x64.exe" "WinUSB_Generic_Device.inf"
libwdi:debug [syslog] 2\DriverStore\FileRepository\hackrf_one.inf_amd64_6c10ffe7d1addae6} 19:43:15.897
[... 198 lines cut ...]
```

 but after the installation the driver is not working properly.
The Device Manager shows 

```
This device cannot start. (Code 10)

An invalid parameter was passed to a service or function.
```
Problem Status : 

```
An invalid parameter was passed to a service or function.

C000000D
```

Problem Code : `0000000A`

I tried reinstalling, rebooting, device removal and reinstall many times but nothing worked.
If i use the hackrf under linux everything works fine, so it seems like there is no problem on the hackrf itself.

Thanks in advance
Alex

alexwahl, 2020-11-09:
VirtualBox Version is 6.1.16 r140961 with extension pack

alexwahl, 2020-11-09:
HackRF Firmware is Version 2018.01.1, Board ID 2

pbatard, 2020-11-09:
I'm going to be harsh here, but the WinUSB driver was installed properly, and this is not a HackRF support forum.

If the driver was installed successfully, which it was, then you will have to seek support elsewhere, because the only part I can ever support is the driver installation itself. Everything else, i.e. whether the driver can actually work with your device or the software you use is entirely out of scope for this project.

As such, I will close this issue.

closed 2020-11-09 by pbatard