pbatard/libwdi #204: Zadig coming up blank.

Ek17s, 2020-12-14:
Hi there, I have been using a Mayflash adaptor to connect a gamecube controller to my pc. It's been working for months but today it stopped working. I tried to re-install the driver with Zadig but all the text boxes that usually auto fill with the correct information are blank and I can't select anything. I clicked advanced mode and this is what I got.

Thank you, for any help.

Zadig 2.5.739
Windows 10 64-bit (Build 19041)
ini file 'zadig.ini' not found in 'C:\Users\conno\Downloads' - default parameters will be used
default driver set to 'WinUSB'
IMPORTANT: 04F3
IMPORTANT: 04F2
IMPORTANT: 8087
IMPORTANT: 0000
0 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F3&PID_0C03&REV_0143
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 3.6.11001.10301
libwdi:debug [wdi_create_list] WUDFRd USB device (1): USB\VID_04F3&PID_0C03\5&3B902538&0&3
libwdi:debug [wdi_create_list] Device description: 'ELAN:Fingerprint'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F2&PID_B64F&REV_9649&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.488
libwdi:debug [wdi_create_list] usbvideo USB device (2): USB\VID_04F2&PID_B64F&MI_00\6&147B4C&0&0000
libwdi:debug [wdi_create_list] Device description: 'HD User Facing (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8087&PID_0026&REV_0002
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_E0&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 22.0.0.2
libwdi:debug [wdi_create_list] BTHUSB USB device (3): USB\VID_8087&PID_0026\5&3B902538&0&10
libwdi:debug [wdi_create_list] Device description: 'Intel(R) Wireless Bluetooth(R)'
libwdi:debug [wdi_create_list] Hardware ID: USB\DEVICE_DESCRIPTOR_FAILURE
libwdi:debug [wdi_create_list] Compatible ID: USB\DEVICE_DESCRIPTOR_FAILURE
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.488
libwdi:debug [wdi_create_list] Driverless USB device (4): USB\VID_0000&PID_0002\5&3B902538&0&4
libwdi:debug [wdi_create_list] Device description: 'Unknown USB Device (Device Descriptor Request Failed)'
Using inf name: WinUSB_Generic_Device.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] Successfully extracted driver files to 'C:\Users\conno\usb_driver'
libwdi:info [wdi_prepare_driver] Successfully created 'C:\Users\conno\usb_driver\WinUSB_Generic_Device.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\conno\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\conno\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'winusb_generic_device.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\conno\usb_driver\winusb_generic_device.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\conno\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\conno\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\conno\usb_driver\WinUSB_Generic_Device.cat'
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] Created new key container
libwdi:debug [CreateSelfSignedCert] Generated new keypair...
libwdi:info [CreateSelfSignedCert] Created new self-signed certificate 'CN=USB\MS_COMP_WINUSB (libwdi autogenerated)'
libwdi:debug [SelfSignFile] Successfully created certificate 'CN=USB\MS_COMP_WINUSB (libwdi autogenerated)'
libwdi:info [SelfSignFile] Added certificate 'CN=USB\MS_COMP_WINUSB (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] Successfully signed file 'C:\Users\conno\usb_driver\WinUSB_Generic_Device.cat'
libwdi:info [SelfSignFile] Successfully deleted private key
libwdi:debug [wdi_install_driver] Using progress bar mode
libwdi:debug [installer process] got parameter WinUSB_Generic_Device.inf
libwdi:debug [process_message] Got request for device_id
libwdi:debug [process_message] No device_id - sending empty string
libwdi:debug [installer process] got device_id: ''
libwdi:debug [process_message] Got request for hardware_id
libwdi:debug [process_message] No hardware_id - sending empty string
libwdi:debug [installer process] got hardware_id: ''
libwdi:debug [installer process] got user_sid: 'S-1-5-21-396309596-3714560801-2474913749-1001'
libwdi:debug [installer process] using syslog 'C:\Windows\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
[... 94 lines cut ...]
IMPORTANT: 04F3
IMPORTANT: 04F2
IMPORTANT: 8087
IMPORTANT: 0000
0 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F3&PID_0C03&REV_0143
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 3.6.11001.10301
libwdi:debug [wdi_create_list] WUDFRd USB device (1): USB\VID_04F3&PID_0C03\5&3B902538&0&3
libwdi:debug [wdi_create_list] Device description: 'ELAN:Fingerprint'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F2&PID_B64F&REV_9649&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.488
libwdi:debug [wdi_create_list] usbvideo USB device (2): USB\VID_04F2&PID_B64F&MI_00\6&147B4C&0&0000
libwdi:debug [wdi_create_list] Device description: 'HD User Facing (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8087&PID_0026&REV_0002
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_E0&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 22.0.0.2
libwdi:debug [wdi_create_list] BTHUSB USB device (3): USB\VID_8087&PID_0026\5&3B902538&0&10
libwdi:debug [wdi_create_list] Device description: 'Intel(R) Wireless Bluetooth(R)'
libwdi:debug [wdi_create_list] Hardware ID: USB\DEVICE_DESCRIPTOR_FAILURE
libwdi:debug [wdi_create_list] Compatible ID: USB\DEVICE_DESCRIPTOR_FAILURE
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.488
libwdi:debug [wdi_create_list] Driverless USB device (4): USB\VID_0000&PID_0002\5&3B902538&0&4
libwdi:debug [wdi_create_list] Device description: 'Unknown USB Device (Device Descriptor Request Failed)'

Ek17s, 2020-12-14:
Tried turning off fast boot, restarting and having another go. Now when I click list all devices I get an option called "Unknown USB Device (Device decriptor request failed), clicked install driver. Didn't work but I got this log.

Zadig 2.5.730
Windows 10 64-bit (Build 19041)
ini file 'zadig.ini' not found in 'C:\Users\conno\Downloads' - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
4 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F3&PID_0C03&REV_0143
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 3.6.11001.10301
libwdi:debug [wdi_create_list] WUDFRd USB device (1): USB\VID_04F3&PID_0C03\5&3B902538&0&3
libwdi:debug [wdi_create_list] Device description: 'ELAN:Fingerprint'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F2&PID_B64F&REV_9649&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.488
libwdi:debug [wdi_create_list] usbvideo USB device (2): USB\VID_04F2&PID_B64F&MI_00\6&147B4C&0&0000
libwdi:debug [wdi_create_list] Device description: 'HD User Facing (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8087&PID_0026&REV_0002
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_E0&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 22.0.0.2
libwdi:debug [wdi_create_list] BTHUSB USB device (3): USB\VID_8087&PID_0026\5&3B902538&0&10
libwdi:debug [wdi_create_list] Device description: 'Intel(R) Wireless Bluetooth(R)'
libwdi:debug [wdi_create_list] Hardware ID: USB\DEVICE_DESCRIPTOR_FAILURE
libwdi:debug [wdi_create_list] Compatible ID: USB\DEVICE_DESCRIPTOR_FAILURE
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.488
libwdi:debug [wdi_create_list] Driverless USB device (4): USB\VID_0000&PID_0002\5&3B902538&0&4
libwdi:debug [wdi_create_list] Device description: 'Unknown USB Device (Device Descriptor Request Failed)'
3 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F3&PID_0C03&REV_0143
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 3.6.11001.10301
libwdi:debug [wdi_create_list] WUDFRd USB device (1): USB\VID_04F3&PID_0C03\5&3B902538&0&3
libwdi:debug [wdi_create_list] Device description: 'ELAN:Fingerprint'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F2&PID_B64F&REV_9649&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.488
libwdi:debug [wdi_create_list] usbvideo USB device (2): USB\VID_04F2&PID_B64F&MI_00\6&147B4C&0&0000
libwdi:debug [wdi_create_list] Device description: 'HD User Facing (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8087&PID_0026&REV_0002
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_E0&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 22.0.0.2
libwdi:debug [wdi_create_list] BTHUSB USB device (3): USB\VID_8087&PID_0026\5&3B902538&0&10
libwdi:debug [wdi_create_list] Device description: 'Intel(R) Wireless Bluetooth(R)'
4 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F3&PID_0C03&REV_0143
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 3.6.11001.10301
libwdi:debug [wdi_create_list] WUDFRd USB device (1): USB\VID_04F3&PID_0C03\5&3B902538&0&3
libwdi:debug [wdi_create_list] Device description: 'ELAN:Fingerprint'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04F2&PID_B64F&REV_9649&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.488
libwdi:debug [wdi_create_list] usbvideo USB device (2): USB\VID_04F2&PID_B64F&MI_00\6&147B4C&0&0000
libwdi:debug [wdi_create_list] Device description: 'HD User Facing (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8087&PID_0026&REV_0002
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_E0&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 22.0.0.2
libwdi:debug [wdi_create_list] BTHUSB USB device (3): USB\VID_8087&PID_0026\5&3B902538&0&10
libwdi:debug [wdi_create_list] Device description: 'Intel(R) Wireless Bluetooth(R)'
libwdi:debug [wdi_create_list] Hardware ID: USB\DEVICE_DESCRIPTOR_FAILURE
libwdi:debug [wdi_create_list] Compatible ID: USB\DEVICE_DESCRIPTOR_FAILURE
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.488
libwdi:debug [wdi_create_list] Driverless USB device (4): USB\VID_0000&PID_0002\5&3B902538&0&4
libwdi:debug [wdi_create_list] Device description: 'Unknown USB Device (Device Descriptor Request Failed)'
Using inf name: Unknown_USB_Device_(Device_Descriptor_Request_Failed).inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\conno\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\conno\usb_driver\Unknown_USB_Device_(Device_Descriptor_Request_Failed).inf'
[... 254 lines cut ...]
libwdi:debug [process_message] switching timeout back to finite
libwdi:debug [installer process] A more recent driver was found (force option required).
libwdi:debug [process_message] installer process completed
Driver Installation: FAILED (Resource already exists)
4 devices found.
[... cut ...]
libwdi:debug [wdi_create_list] BTHUSB USB device (3): USB\VID_8087&PID_0026\5&3B902538&0&10
libwdi:debug [wdi_create_list] Device description: 'Intel(R) Wireless Bluetooth(R)'
libwdi:debug [wdi_create_list] Hardware ID: USB\DEVICE_DESCRIPTOR_FAILURE
libwdi:debug [wdi_create_list] Compatible ID: USB\DEVICE_DESCRIPTOR_FAILURE
libwdi:debug [wdi_create_list] Driver version: 10.0.19041.488
libwdi:debug [wdi_create_list] Driverless USB device (4): USB\VID_0000&PID_0002\5&3B902538&0&4
libwdi:debug [wdi_create_list] Device description: 'Unknown USB Device (Device Descriptor Request Failed)'

pbatard, 2020-12-16:
1. In your initial run, the use of `USB\MS_COMP_WINUSB` tells me that you were installing the [WCID](https://github.com/pbatard/libwdi/wiki/WCID-Devices) version of the driver, which will not work unless your device has a WCID descriptor, which I'm pretty sure it doesn't. So it's no surprise it didn't work.
2. In your second run, you are clearly trying to install the driver for a device that Windows sees as defective (`USB\DEVICE_DESCRIPTOR_FAILURE`), so that won't work either. As far as Windows is concerned, your hardware device has a problem, because it does not provide the basic USB device descriptors Windows (and Zadig) needs to be able to install a driver for it. So you will need to fix this issue, because there's nothing Zadig can do to make a device work if Windows reports that it can't perform basic USB tasks such as providing USB descriptors.

&rarr; Closing, as this is clearly not a Zadig problem.

closed 2020-12-16 by pbatard