pbatard/libwdi #206: Unable to get wdi-simple.exe to associate WinUSB drivers with an STM32 USB micro controller

lukebayes, 2020-12-31:
I have an electronic product that uses an STM32 micro controller and requires USB device firmware updates (DFU).

I'm having a lot of trouble getting the WinUSB drivers to install and associate without having the device(s) connected and/or active.

First, I live on Linux and haven't really dealt with Windows computers since the late nineties, so it's highly likely I'm doing something obvious and wrong, please don't hesitate to tell me something that should be obvious to a Windows person. :-)

Essentially, we have an HID device that we communicate with to put the STM32 into bootloader / DFU mode. We need to install the WinUSB driver for both our HID device and the STM32 bootloader, but the STM32 bootloader isn't available on the bus, unless we flip the state of the hardware.

The workflow I'm hoping for is something like:

```bash
.\zadic.exe --vid 0x0483 --pid 0xdf11 --create "STM32 BOOTLOADER"
.\zadic.exe --vid 0x335e --pid 0x8a01 --create "Maple Bootloader"
# Execute script to run a DFU update
```
Or possibly:

```bash
wdi-simple.exe -v 0x0483 -p 0xdf11 -b -i 0
wdi-simple.exe -v 0x335e -p 0x8a01 -b -i 0
# Execute script to run a DFU update
```

Unfortunately, the only flow I've been able to get (sort of) working is more like:

```bash
# Plug in the USB cable
wdi-simple.exe -v 0x335e -p 0x8a01 -b -i 0
# Execute script to flip the device state to STM32 Bootloader
.\zadic.exe --vid 0x0483 --pid 0xdf11 --create "STM32 BOOTLOADER"
# Unplug and re-plug the USB cable to reset the device to primary mode
# Execute the script to run a DFU update
```

Unfortunately, testing this installation process is excruciatingly slow and unreliable and I've managed to get my computer into a state where I am no longer able to uninstall these drivers.

Any tips, pointers or help would be appreciated!

lukebayes, 2020-12-31:
Just updating here, I was able to find [this very helpful post](http://mirobetm.blogspot.com/2014/08/removing-libwdi-drivers.html), which indicates the following:

> 1. List all 3rd party inf packages by:
> # pnputil.exe -e > somefile.txt
> 
> 2. Then look for "libwdi" string in somefile.txt
> Example:
> 
> Published name :            oem134.inf
> Driver package provider :   libusbx.org
> Class :                     Urządzenia uniwersalnej magistrali szeregowej
> Driver date and version :   06/02/2012 6.1.7600.16385
> Signer name :               USB\VID_046D&PID_C019 (libwdi autogenerated)
> 
> Now all you need is the command below for reach entry found:
> 
> # pnptuil.exe -d oem134.inf

Using this technique, I was able to delete about 50 libwdi inf files.

Unfortunately, there are 2 inf that refuse to be removed with the following message:
```powershell
PS C:\Users\lukeb\Desktop\pnputil> pnputil.exe -d oem142.inf
Microsoft PnP Utility

Deleting the driver package failed : One or more devices are presently installed using the specified INF.
PS C:\Users\lukeb\Desktop\pnputil>
```
I have just rebooted the computer and there are no (external) USB devices connected to this computer.

The devices in question have the following details from the `pnputil.exe -e` output:
(Both are associated with the STM32 Bootloader, which is certainly not connected)
```
Published name :            oem83.inf
Driver package provider :   libwdi
Class :                     Universal Serial Bus devices
Driver date and version :   06/23/2020 6.1.7600.16385
Signer name :               USB\VID_0483&PID_DF11 (libwdi autogenerated)

Published name :            oem143.inf
Driver package provider :   libwdi
Class :                     Universal Serial Bus devices
Driver date and version :   06/23/2020 6.1.7600.16385
Signer name :               USB\VID_0483&PID_DF11&MI_00 (libwdi autogenerated)

```

lukebayes, 2020-12-31:
Was able to successfully delete these 2 remaining drivers with the `pnputil.exe /force` parameter as follows:
```powershell
pnputil.exe -d oem83.inf /force
pnputil.exe -d oem143.inf /force
```

lukebayes, 2020-12-31:
A quick note:

Even after uninstalling all listed libwdi drivers (by running `pnputil.exe -e > .\somefile.txt` and then searching somefile.txt for "libwdi" and successfully removing them all), I still still have a mysterious libwdi driver on this computer.

Steps:
* Delete all discovered inf drivers
* Plug in the USB device
* Run: `.\wdi-simple.exe -v 0x335e -p 0x8a01 -b -i 0 -n "Maple Bootloader" -d "maple_driver"`
* Execute my script to put the device into STM32 Bootloader mode
* The STM32 device now has an unsigned libwdi driver associated with it (even though it has a different VID/PID and name than what the driver was intended to connect to).

The Device manager Driver Tab for the device indicates that it's libwdi, 6/23/2020, version: 6.1.7600.16385 and "Not digitally signed."

Clicking "Uninstall this device" seems to remove this ghost driver and put the device back in the Unknown device warning state.

For whatever it's worth, the following command will successfully install the libwdi driver for the STM32 BOOTLOADER without the device connected:

```powershell
.\zadic.exe --vid 0x0483 --pid 0xdf11 --create "STM32 BOOTLOADER"
```

But, for some reason, I still need to have my device connected to the bus for the similar command to work for our main device.

lukebayes, 2020-12-31:
I'm starting to suspect there may be an issue with associated the driver against a Composite USB Device?

Because it's a Composite USB device, the VID/PID values are identical for this entire collection of devices.

Here are some screen grabs of the Device Manager details for my container and concrete device:

[image]

[image]

Any tips appreciated!

lukebayes, 2020-12-31:
Well, I found [this line in wdi-simple.c](https://github.com/pbatard/libwdi/blob/master/examples/wdi-simple.c#L229), which experimentally appears to be the crux of what forces the device & driver association to work for this device and machine.

I went ahead and hacked a patch to hard code my `hardware_id`, but alas, it seems the `device_id` is also needed for this device.

lukebayes, 2020-12-31:
For what it's worth, I went ahead and integrated the wdi-simple.exe driver installation process inside our DFU update process. It's not ideal, but at least it seems to work (so far).

Thanks for all your work to provide these command line interfaces to the Zadig tool, it made a huge difference for us!

Please let me know if you see something obvious that I'm doing wrong in here.

pbatard, 2021-02-01:
Since you appear to have solved your problem, I will close this issue.

closed 2021-02-01 by pbatard