pbatard/libwdi #208: Driver installation cancelled by user

JoernMueller, 2021-02-01:
#### Steps performed
1. Replaced a custom driver with WinUSB using Zadig 2.5.730.
2. Removed WinUSB for the particular device again using the Windows Hardware Manager.
3. Reinstalled the original custom driver again using the `.inf` file.
4. Tried to replace the custom driver again as done already in step 1.
5. Zadig reports: *Driver installation cancelled by user.* which did not happen.

The log seems to come closer to the real cause of the issue:
```
libwdi:warning [CreateSelfSignedCert] Could not obtain a key container: The requested key container is corrupted.
libwdi:warning [wdi_prepare_driver] could not sign cat file
```
But lacks any information which certificate exactly is corrupted and required for the signing process.

Checking the related issues:
- [Zadig: Driver Installation: Installtion Cancel by User](https://github.com/pbatard/libwdi/issues/20)
- [Error: Could not obtain a key container: Unknown PKI error 0x8009001A](https://github.com/pbatard/libwdi/issues/168)

they both state in a rather general way that *something* must have been broken on the system.
Checking the files in `C:\Users\<userID>\AppData\Roaming\Microsoft\Crypto\RSA\` I see they have not been touched for more than a year while the steps describe above have been performed yesterday and no other changes on the system that possibly might affect the integrity of any certificates have been done in between. So if the signing certificate is corrupted it must have been broken by any of the steps 1 - 5. 

#### Expected Behaviour
The error reporting should be precise and specific enough to allow the user to locate and fix the problem.

pbatard, 2021-02-01:
> I see they have not been touched for more than a year

A timestamp is not a guarantee that a file has not been modified.
I could take an old floppy disk, that I haven't touched since 1990, and according to you, declare that _"the files on this floppy can't possibly be corrupted since the file explorer shows that their timestamp is from 199x"_. But actually trying to read the files, from a media that has never exactly be known for its reliability, especially on a time range of 30 years, might tell a very different story.

Disk corruption occurs. Bad RAM occurs. Cosmic rays occur. There are literally zillions of ways your files in `Crypto\RSA\` could have been corrupted without showing an altered timestamps, and that's not even talking about the Windows internals where this or that update may trigger errors. And yes, [even Microsoft sometimes introduces major corruption on data that is meant to be reliable and that you would expect to be impervious to these kind of blunders, especially when coming from a megacorporation that's supposed to have ample budget to test its stuff before deployment to the masses](https://pete.akeo.ie/2020/06/et-tu-microsoft.html).

> it must have been broken by any of the steps 1 - 5.

Then you should have little difficulty proving it by replicating your steps on a clean machine.

> The error reporting should be precise and specific enough to allow the user to locate and fix the problem.

And you are free to submit a patch for that, especially as doing so would make you realize that providing comprehensive error details that can help users pinpoint where PKI issue likes is really Microsoft's job and not Zadig's, as the only insight Zadig has into what happened is the error it reported.

Zadig reports exactly what Windows reports, and we can't magically make new error details materialize out of thin air when Microsoft doesn't provided that data.

Since, no matter how much you might think otherwise, I don't really see anything more Zadig could do here, and especially, I see no hard evidence that Zadig is responsible for your environmental issue, I will close it.

closed 2021-02-01 by pbatard

JoernMueller, 2021-02-01:
Well, I see, probably a cosmic ray has hit exactly the file cluster of the cert exactly in the approximately two hours between step 1 and 4. A conclusion as sharp as Okham's razor.
And nope, I will not set up a clean machine to prove you anything about your code. I already took some time to report the issue to you as also two other users did in issues #168 and #200 before me. It´s up to you whether you want to take this hints as a valueable input on where some improvement might be required (at least the obviously misleading popup message should be quite easy to fix) or if you prefer to keep blaming cosmic rays and the systems environment.

pbatard, 2021-02-01:
Again, you are free to invest your own time to improve this Open Source project.

Or you can keep deluding yourself into believing that the maintainers of a project such as this one have the resources of a Microsoft support department and that we can magically make money appear out of thin air, from the exact $0 you paid to use this product, in order to support an extensive investigation into what nobody has ever provided any hint as being anything but a pure Microsoft Windows issue.

Ceedjee, 2021-02-18:
Hi,

First of all **thank you** for your code @pbatard 
Just adding my 2 cents on this (it seems) recurring topic:

Zadig was working fine on my computer but not on my colleague's, so I tried to investigate a little bit this issue and I noticed 2 things:
- when everything works fine, libwdi is supposed to be removed from RSA folder; for some reason it's not always the case and it can lead to same error with subsequent driver installations (note that we still get a "Successfully deleted private key")
- the flag "CRYPT_MACHINE_KEYSET" was not appreciated on some hosts, so what I did was to remove it after first unsuccessful CryptAcquireContextW

I figured this notes could help on that particular topic (even though I'm not yet able to really determine the root cause of the mentionned points, I guess it can be rights limitations)

I'm not a developer myself and definitely not comfortable with Windows APIs; so I'm not sure a PR from me would be really helpful but you tell me if you want one

pbatard, 2021-02-18:
Thanks for the info. As pointed out above I already added a workaround that will hopefully address the problem of the key removal on reluctant machines. But it's not in a release yet. If you are interested, I can try to upload a BETA somewhere, to confirm that it fixes the issue you saw, but it may be a while before I am able to do that.

Also, FIY, we are already in an `out:` section when the `Successfully deleted private key` message is issued, per [this section if the code](https://github.com/pbatard/libwdi/blob/master/libwdi/pki.c#L1028-L1036):
```C
	// Clean up
out:
	/*
	 * Because we installed our certificate as a Root CA as well as a Trusted Publisher
	 * we *MUST* ensure that the private key is destroyed, so that it cannot be reused
	 * by an attacker to self sign a malicious applications.
	 */
	if ((pCertContext != NULL) && (DeletePrivateKey(pCertContext))) {
		wdi_info("Successfully deleted private key");
	}
```

In other words, the removal of the private key is not tied to anything (apart from having a CertContext, but if you don't have that, you wouldn't have a key to delete and, as can be seen from `CreateSelfSignedCert(...)` you also wouldn't have had a "silent" error, as **every** single error path that leads to the failure of the creation of the keypair or certificate will produce an explicit warning. And no such warnings have been present in the user reports I have seen. Especially, if a host rejects `CRYPT_MACHINE_KEYSET` you will get a `Could not obtain a key container:` warning. And for the record, `CRYPT_MACHINE_KEYSET` is there for a reason, because we want the certificate to be stored at the machine level and not at the user level, so that the driver can work with all users.

So, this means that, when this issue happens, `CreateSelfSignedCert()` has been successful which in turn means that, as far as Windows is concerned, Windows is perfectly happy with how the keys were generated and it should be able to delete them.

Now, my expectation is that people who do encounter this issue have actually gotten a `Failed to delete private key` beforehand, on a successful driver install that they previously carried out. But of course, since the install was successful, they didn't pick on this (because, obviously, we don't treat this as a fatal error and people are unlikely to look at their log on success to see if there were any warnings).

But there again, I see no programmatic reason, that I could fix in libwdi, to make that earlier key deletion work if it produced a warning, because there really isn't much extra I can tell Windows to do when issuing `CRYPT_DELETEKEYSET` to its API fails, and it's clearly not due to the application somehow passing wrong parameters (since the call would consistently fail otherwise).

Right now, my best guest is that some folks are configuring their platform in a manner that makes __silent__ deletion of keysets fail (since we also pass `CRYPT_SILENT` along `CRYPT_DELETEKEYSET`). But I'm pretty sure that means that non standard system policies have been applied, and there's only so much I am willing to invest in supporting nonstandard Windows installations.

At any rate, I do expect the workaround that I have applied to be able to sidestep that issue, so I don't think there's need for any more work, unless I get some clear reports that the workaround does not work.

Ceedjee, 2021-02-18:
I am sure the issue is not coming from libwdi and I figured why the machine level installation was there, just not necessary in my particular case and maybe that could help some other folks in the future :)

Regarding your workaround, I rebuilt zadig from the master branch so I guess I do have it on my side but it does not work either. Might be due to the state my faulty host is in already.

As for the (maybe not) missing "goto out" I was more refering about the fact that DeletePrivateKey() returns True in [that case](https://github.com/pbatard/libwdi/blob/master/libwdi/pki.c#L879-L881)

Anyway, thanks for your time

pbatard, 2021-02-18:
> I was more refering to the fact that DeletePrivateKey() returns True in that case

You have a point, though my assumption there was that, regardless of whether we managed to delete the key or not (which we don't want to treat as a fatal error) you want to reimport the certificate to reflect the changes applied by Windows with regards to the key.

I guess, if the other workaround doesn't work, I can try to skip the reimport.

But I wouldn't mind getting some confirmation that users who do experience the issue do get the `Failed to delete private key` during an earlier install, to validate that Windows is indeed screwing the reimport of a certificate if `CRYPT_DELETEKEYSET` failed...

Ceedjee, 2021-02-18:
All I can say is that I faced the issue when trying to reinstall on my machine, and I never looked at the logs time because everything was working fine. But my colleague, who shared his problem with me, never heard of Zadig/Libwdi before trying to install WinUSB. 
I guess I could try to redo everything on some "blank" windows hosts and share the logs if I'm "lucky' enough to reproduce it.

But in the end, I'm pretty sure it's due to some limitations in our Windows setup