pbatard/libwdi #209: Driver Installation Cancelled by User

jrhug, 2021-02-02:
I originally installed this driver and then had to remove it.  Now, I am trying to reinstall it and I get the warning:
**libwdi:warning [CreateSelfSignedCert] Could not obtain a key container: The requested key container is corrupted.**  

Where do I find this "corrupted key container"?  I was successful in loading the driver once.  Do I need to take extra actions when uninstalling?  The system security settings did not change between the original installation and the attempt to reinstall. 

**Log**
Using inf name: Flyswatter2_(Interface_1).inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\hughjame\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\hughjame\usb_driver\Flyswatter2_(Interface_1).inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\hughjame\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\hughjame\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'flyswatter2_(interface_1).inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\hughjame\usb_driver\flyswatter2_(interface_1).inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\hughjame\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\hughjame\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] Successfully created file 'C:\Users\hughjame\usb_driver\Flyswatter2_(Interface_1).cat'
libwdi:debug [CreateSelfSignedCert] Set Enhanced Key Usage, URL and CPS
libwdi:warning [CreateSelfSignedCert] Could not obtain a key container: The requested key container is corrupted.
libwdi:warning [wdi_prepare_driver] could not sign cat file
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter Flyswatter2_(Interface_1).inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0403&PID_6010&MI_01\7&1CE5AEFC&0&0001'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_0403&PID_6010&REV_0700&MI_01'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-839522115-1383384898-515967899-5572085'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_0403&PID_6010&REV_0700&MI_01 - please wait...
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_0403&PID_6010&REV_0700&MI_01]
libwdi:debug [syslog] >>>  Section start 2021/02/02 17:50:03.697
libwdi:debug [syslog]       cmd: "C:\Users\hughjame\usb_driver\installer_x64.exe" "Flyswatter2_(Interface_1).inf"
libwdi:debug [syslog]      ndv: INF path: C:\Users\hughjame\usb_driver\Flyswatter2_(Interface_1).inf
libwdi:debug [syslog]      ndv: Install flags: 0x00000001
libwdi:debug [syslog]      ndv: {Update Device Driver - USB\VID_0403&PID_6010&MI_01\7&1CE5AEFC&0&0001}
libwdi:debug [syslog]      ndv:      Search options: 0x00000080
libwdi:debug [syslog]      ndv:      Searching single INF 'C:\Users\hughjame\usb_driver\Flyswatter2_(Interface_1).inf'
libwdi:debug [syslog]      dvi:      {Build Driver List} 17:50:03.719
libwdi:debug [syslog]      dvi:           Searching for hardware ID(s):
libwdi:debug [syslog]      dvi:                usb\vid_0403&pid_6010&rev_0700&mi_01
libwdi:debug [syslog]      dvi:                usb\vid_0403&pid_6010&mi_01
libwdi:debug [syslog]      dvi:           Searching for compatible ID(s):
libwdi:debug [syslog]      dvi:                usb\class_ff&subclass_ff&prot_ff
libwdi:debug [syslog]      dvi:                usb\class_ff&subclass_ff
libwdi:debug [syslog]      dvi:                usb\class_ff
libwdi:debug [syslog]      sig:           {_VERIFY_FILE_SIGNATURE} 17:50:04.116
libwdi:debug [syslog]      sig:                Key      = flyswatter2_(interface_1).inf
libwdi:debug [syslog]      sig:                FilePath = c:\users\hughjame\usb_driver\flyswatter2_(interface_1).inf
libwdi:debug [syslog]      sig:                Catalog  = c:\users\hughjame\usb_driver\Flyswatter2_(Interface_1).cat
libwdi:debug [syslog] !    sig:                Verifying file against specific (valid) catalog failed.
libwdi:debug [syslog] !    sig:                Error 0x800b0100: No signature was present in the subject.
libwdi:debug [syslog]      sig:           {_VERIFY_FILE_SIGNATURE exit(0x800b0100)} 17:50:04.217
libwdi:debug [syslog]      sig:           {_VERIFY_FILE_SIGNATURE} 17:50:04.220
libwdi:debug [syslog]      sig:                Key      = flyswatter2_(interface_1).inf
libwdi:debug [syslog]      sig:                FilePath = c:\users\hughjame\usb_driver\flyswatter2_(interface_1).inf
libwdi:debug [syslog]      sig:                Catalog  = c:\users\hughjame\usb_driver\Flyswatter2_(Interface_1).cat
libwdi:debug [syslog] !    sig:                Verifying file against specific Authenticode(tm) catalog failed.
libwdi:debug [syslog] !    sig:                Error 0x800b0100: No signature was present in the subject.
libwdi:debug [syslog]      sig:           {_VERIFY_FILE_SIGNATURE exit(0x800b0100)} 17:50:04.251
libwdi:debug [syslog]      dvi:           Created Driver Node:
[... 82 lines cut ...]
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6010&REV_0700&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 2.12.28.0
libwdi:debug [wdi_create_list] FTDIBUS USB device (2): USB\VID_0403&PID_6010&MI_00\7&1CE5AEFC&0&0000
libwdi:debug [wdi_create_list] Device description: 'Flyswatter2 (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_05C8&PID_0369&REV_0108&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 3.5.8.14
libwdi:debug [wdi_create_list] SPUVCbv USB device (3): USB\VID_05C8&PID_0369&MI_00\7&5B0FA56&0&0000
libwdi:debug [wdi_create_list] Device description: 'HP HD Webcam (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0403&PID_6010&REV_0700&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 2.12.28.0
libwdi:debug [wdi_create_list] FTDIBUS USB device (5): USB\VID_0403&PID_6010&MI_01\7&1CE5AEFC&0&0001
libwdi:debug [wdi_create_list] Device description: 'Flyswatter2 (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8087&PID_07DA&REV_7869
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_E0&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 18.1.1525.1445
libwdi:debug [wdi_create_list] BTHUSB USB device (6): USB\VID_8087&PID_07DA\6&A2946AD&0&6
libwdi:debug [wdi_create_list] Device description: 'Intel(R) Wireless Bluetooth(R)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_138A&PID_003F&REV_0104
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 4.5.338.0
libwdi:debug [wdi_create_list] WUDFRd USB device (12): USB\VID_138A&PID_003F\00303448B894
libwdi:debug [wdi_create_list] Device description: 'Synaptics FP Sensors (WBF) (PID=003f)'

pbatard, 2021-02-03:
Can you please provide details on how exactly you uninstalled the driver? Did you use Device Manager? Something else? Did you select "Remove the driver files"? Did you reboot your machine before trying to install the driver again?

I am getting sporadic reports for "The requested key container is corrupted", for which I have recently applied a workaround (but that workaround is not yet in a Zadig release), but, even as I tried to replicate the issue, I have never been able to see it on my machine, no matter how many uninstall/reinstall of a driver I go through. I am currently suspecting that there's a component to this that has to do with devices that get a third party driver automatically installed by Windows, which I assume the Flyswatter is (I'm afraid I don't have such a device to test with).

As to your question, I'm afraid that, since I have never experienced this issue, I have no idea how one can manually remove/reset the "corrupted" key container that Windows is complaining about. You may try to use `certmgr`, go to _Trusted Publisher_ and try to remove the `(libwdi autogenerated)` certificates you find there to see if that helps.

If not, I'll see if I can produce a pre-release of Zadig, with the workaround mentioned above, as I'm actually curious to find out if it can provide a solution for this issue.

jrhug, 2021-02-03:
Thank You for the prompt response.  
When the driver was removed, I used device manager and selected the checkbox to "Delete the driver software for this device".   The system has been rebooted several times.  The system is running Windows 10.  The certificates were removed and the issue is still present.

General information about the Flyswatter2, which may or may not be useful.  The external device has two USB ports.  I am only trying to update one of the ports.  Could this be a problem? 

[image]

[image]

jrhug, 2021-02-03:
UPDATE: As workaround, I 'Disable Driver Signature Enforcement'.  This allows for the driver to be installed.  If I learn anymore information about my issue, I will update this thread.  
Thank you again for the help.

closed 2021-02-04 by jrhug