pbatard/libwdi #211: Gamecube controller adaptater

Haypzeh, 2021-02-06:


Haypzeh, 2021-02-06:
Hello there, I got an issue with Zadig and my (unofficial) adapter.
I don't have the WUP-028, but Zadig detects it as "Gamecube Controller Adapter" with usbs id 0079 and 1846.
Otherwise when I install the driver it doesn't work on my Dolphin Emulator (the adapter is not detected).
How can I fix it please?

Have a nice day,
Haypzeh

Haypzeh, 2021-02-06:
[image]
[image]

pbatard, 2021-02-07:
> How can I fix it please?

Zadig shows that the driver was successfully installed, and you need to understand that Zadig only automates what you could also do manually, by picking up the WinUSB driver files published by Microsoft and crafting your own driver installation package.

Zadig is a **generic** driver installer, which means that, as long as the driver was installed successfully, which it was, it has absolutely nothing to do with whether a driver will work with a specific application or not, and you will have to see with the makers of the application if you have issues.

I will therefore encourage you to thake this issue with the Dolphin support user forums because I'm afraid it has nothing to do with libwdi/Zadig.

closed 2021-02-07 by pbatard

Haypzeh, 2021-02-07:
Okay thank you very much!