pbatard/libwdi #212: Zadig hangs almost forever

kunosee, 2021-03-11:
When starting Zadig over an SMB shared drive, the installation hangs almost forever. Giving nice comments about the log time, while finally reporting a broken install and the UI hangs and must be killed from taskmanager.
Once the exe-file is copied to a local drive, everything runs smooth.

pbatard, 2021-03-12:
Thanks for the report.

As much as I would like to, I don't think I'm going to look into this (too busy + libwdi is now a low priority project for me). But if someone wants to do that and send a patch, I'll be happy to apply it.

At least this issue report should help people who run into this situation.

closed 2021-03-12 by pbatard

diabolo38, 2021-03-13:
My 2ct if that can help . Depending window setting (uac requested?) or maybe user right/permission many things/app wan't work at alll on some shared or network drive (setup msi etc for instance wan't work on a vm shared folder)
theirs sometime clear message about it sometime not :(

kunosee, 2021-03-21:
I am almost shure, that it must be related, that zadig.exe tries to open itself after elevation. As the elevated (UAC) account, has no drive mapping (or no user rights) to the samba-share, some file operation may hang.
So either zadig must extract everything it needs from its own binary before elevation, or there should be a warning, when started from a network share.