pbatard/libwdi #213: Would like to see generic HID driver added to list of Zadig drivers

Videogamer555, 2021-04-26:
Some devices are designed to be operated at a slightly higher level than raw USB, these operate at the raw HID level. Windows has generic HID drivers (hidclass.sys hidparse.sys and hidusb.sys), in the same way it has generic USB drivers (WinUSB). However, for an HID device that doesn't directly report that it is an HID device, then a program like Zadig could generate a proper INF file and then install that device as a generic HID device. This would allow it (assuming that it actually is an HID device) to be controlled via Windows's HID API, including reading input reports via the WM_INPUT window message (after the device is registered to the given hWnd with the RegisterRawInputDevices function).

One of the most infamous USB HID devices that doesn't actually report itself as an HID device (nor does it have a proper USB cable, but it is a USB device and can be plugged in to a USB port with a hacked USB cable), is the original XBox controller. For Windows XP you can download a hacker-written driver that lets the controller work, but that doesn't work on any newer version of Windows. And even though later versions of Windows did include a driver for the XBox360 controller (and later the XBox One controller), Microsoft never added one for the XBox original controller. However given that it is a game controller, I highly suspect that the type of USB protocol it's using is in fact the HID protocol, even though it doesn't report itself as an HID device when it's plugged into the computer. If Zadig allowed installing the generic HID drier for any device, then this XBox original controller issue (and possibly other similar issues with other proprietary HID devices) could be solved.

pbatard, 2021-04-26:
I'm afraid that that's a bit too specific a request for me to want to invest any time on it. As you may understand, the amount of people besides yourself this would actually benefit is very limited. I may accept a patch if someone wants to send one (but even then, if the only benefit is for folks using the now very old original XBox controller, I'm doubtful I would want to include it in Zadig), but that's about it.

&rarr; Rejected.

closed 2021-04-26 by pbatard

mcuee, 2021-09-05:
Moreover, if you really want to use the thing, you can try the following. 
https://www.s-config.com/xbcd-original-xbox-controllers-win10/

yoiang, 2024-11-17:
@Videogamer555 Did you ever find a generic HID driver builder or equivalent?