pbatard/libwdi #214: How to run the exe file with cmd

rodchen-king, 2021-04-27:


rodchen-king, 2021-04-27:
I want to run the exe with cmd. How to do it?
Is there some option parameters? I can run the exe with list all device and choose one device and then install it.

pbatard, 2021-04-27:
`installer_###.exe` is not designed to run from the commandline. It can only be used when invoked from a libwdi application.

If you want a commandline executable you can run to install a driver, you need to look at the [examples](https://github.com/pbatard/libwdi/tree/master/examples), and especially `wdi-simple`:
```
D:\libwdi\Win32\Debug\examples>wdi-simple -h

-n, --name <name>          set the device name
-f, --inf <name>           set the inf name
-m, --manufacturer <name>  set the manufacturer name
-v, --vid <id>             set the vendor ID (VID, use 0x prefix for hex)
-p, --pid <id>             set the product ID (PID, use 0x prefix for hex)
-i, --iid <id>             set the interface ID (MI)
-t, --type <driver_type>   set the driver to install
                           (0=WinUSB, 1=libusb-win32, 2=libusbK, 3=usbser, 4=custom)
-w, --wcid                 use a WCID driver instead of a device-specific
                           one (WinUSB, libusb-win32 or libusbK only)
    --filter               use the libusb-win32 filter driver (requires -t1)
-d, --dest <dir>           set the extraction directory
-x, --extract              extract files only (don't install)
-c, --cert <certname>      install certificate <certname> from the
                           embedded user files as a trusted publisher
    --stealth-cert         installs certificate above without prompting
-s, --silent               silent mode
-b, --progressbar=[HWND]   display a progress bar during install
                           an optional HWND can be specified
-o, --timeout              set a timeout (in ms) to wait for any pending installations
-l, --log                  set log level (0=debug, 4=none)
-h, --help                 display usage
```

You will however need to compile `wdi-simple` yourself as I am not providing any releases besides Zadig.

[This](https://github.com/pbatard/libwdi/wiki/Compiling-and-debugging-libwdi-or-Zadig) may help you do so.

closed 2021-04-27 by pbatard