pbatard/libwdi #215: The following problem occurs when I replace  libusb-win32-bin-1.2.6.0   with   libusb-win32-bin-1.2.7.1

zhanghao1111, 2021-05-14:
Windows cannot verify the digital signature of the driver required for this device. Some software or hardware has been changed recently. A file with an incorrect signature or a damaged file may be installed, or the installed file may be malicious software of unknown origin. (Code 52)
[image]

pbatard, 2021-06-04:
Sorry but I can only provide help for the driver that was installed with Zadig, which is still 1.2.6.0 for the current version.

If you replace the driver files, then of course the signatures won't match and you will need to sort it out yourself.

The last official release of libusb-win32 is 1.2.6.0 (per https://sourceforge.net/projects/libusb-win32/files/libusb-win32-releases/), and until there is an official release of 1.2.7.x, I am not planning to update the files used by Zadig/libwdi.

As such, I will close this issue.

closed 2021-06-04 by pbatard

mcuee, 2021-09-05:
FYI libusb-win32 1.2.7.1 is a unsigned snapshot release and should not be used by end user. 
https://sourceforge.net/projects/libusb-win32/files/libusb-win32-snapshots/20190918/